reserve85/HoymilesZeroExport #192: Victron GX native support

Hofyyy, 2024-05-03:
Hello,

first thank you for the great project.
I use 3 hoymiles devices + a seperate Victron ESS without solar.
So I would like to use the Victron Grid meter for your script too.

I saw that you have some py for victron already, but I am not sure if that would work for the official Victron GX device.
Which is more or less the victron controler. Do you know if that will also work with the GX device?

Links:
https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/victron-energy.yaml

Second question, it the modbus dependency included in the docker image,
so that I could use that in docker directly?

Sorry for the basic questions.

best regards
Sebastian

smaicloud, 2024-05-03:
Would be great, I want to use a Cerbo soon too.

reserve85, 2026-11-11:
Native support for the Victron GX is not planned. Two options that should work without it, although I have not tested them with a GX device: the script powermeter (see scripts/GetPowerFromVictronMultiplus.sh as a starting point) or the MQTT powermeter reading the grid value the GX publishes via MQTT. Closing; if someone gets it running, a short write-up would be welcome.

closed 2026-11-11 by reserve85