reserve85/HoymilesZeroExport #194: Maybe a great script extension?

MavenCH, 2024-05-04:
Hello everyone

First of all a big thank you to the creators of this script. It works great :)
I put it into operation yesterday and even if the weather is not too good right now I could already see it a few times at work :)

Why I am opening this here is, I wanted to ask if there is a possibility that the script sends me data via MQTT or otherwise that I can display in Home Assistant, so I can see when the script and how much it has made changes to the power limit?
Actually like the command in the script "sudo journalctl -u HoymilesZeroExport.service -n 20000 -e -f" only that I have the values in HA and can see them as a graphic.
Like the graph when and how much electricity my PV system generates but here how much it is throttled.

I would be very grateful if there would be a possibility here :)
Because then I could create a script or an automation that then for example my car starts to charge or something else starts to draw electricity or starts to draw more. (For example, I can set my car via Home Assistant with how much it charges and in combination with that I could set it so that it adapts to the script :) )
I think that would certainly be a great extension for the script.

Many thanks and have a nice weekend.
Greetings from Switzerland

irrwitzer42, 2024-05-04:
Hi Maven,

wouldn’t it be much better to fetch those values directly from ahoyDTU / openDTU?
Those do have MQTT support and you get the realtime values, without the dependency to hoymileszeroexport. I’m using ahoy’s prometheus endpoints to fetch those metrics and put them in a prometheus DB.

I think, reserve85’s script should focus on the limitting, not the reporting ;-)

BR,
  Irrwitzer

tomquist, 2024-05-04:
I implemented the current MQTT integration and might be able to work on this. Which data would you like to get, that you can’t get via your DTU already?

reserve85, 2024-05-04:
I fetch those values directly from AHOY / Tasmota:

[image]

MavenCH, 2024-05-04:
Hello everyone
Thanks for the quick responses :)

@irrwitzer42 Thank you very much for your answer.
Theoretically this would be a possibility, but I see it as inconvenient if I have to create an additional DB just to get this one value.
If Ahoy would already provide this via MQTT, the whole thing would already be perfect because I could get the data without detours.

@tomquist thank you very much for your answer.
I have already seen the MQTT interface from the script and initially understood that this already provides the data, but after a lot of tinkering and not understanding why my MQTT server terminates the connection with the field: ‘Client HoymilesZeroExport disconnected, not authorised.’ I understood that this is not for sending but for receiving?
I had already made a request because of this, but I closed it again when I understood that this interface is intended for something else.  (mqtt not authorised error... #193)
If it would be possible for you to add this this would of course be fantastic and I would be extremely grateful :)
The data I would be interested in is as follows;
- What is adjusted to. 1500W to 1100W or 320W to 1500W
Just like @reserve85 has done. But I would only need the point ‘Limit’. Since the rest of Ahoy already comes via MQTT. 
@reserve85 already has it exactly as I imagine it. But I would prefer it if it wouldn't need more tools to get the data so I would prefer if you could extend it in the script :)
I want to be able to create an overview like this via MQTT and Home Assistant :)

@reserve85 thank you very much for your quick feedback :)
How did you manage that? Did you install something additional on the ESP or how did you do it?
Because that's exactly the kind of overview I'm trying to get. Simply in Home Assistant and if possible via MQTT, but of course I am also open to alternatives, even if I prefer compact solutions to x different tools. 

Thank you already for your support and I hope we can find a solution :)

Have a nice day / a good one if I live in Europe.
Greetings from Switzerland

reserve85, 2024-05-04:
Hey,

it is home assistant, and the current limit does already come from AHOY, you just need to add it manually...

configuration.yaml:

```
mqtt:
  - sensor:
      - name: "HM_1500_Limit_in_Percent"
        state_topic: "Hoymiles/HM-1500/ch0/active_PowerLimit"
        unit_of_measurement: "%"
```

and if you want it in "W" you need to calculate it:

```
template:
  - sensor:
      - name: "HM_1500_Limit"
        unit_of_measurement: "W"
        device_class: energy
        state: >-
          {{ states('sensor.hm_1500_limit_in_percent') | float(default=100) | multiply(15) | float }}
```

MavenCH, 2024-05-04:
Hello reserve85

That's a bad joke now, isn't it?
Ahoy has already provided this data, just not categorised...
I have tried it and it works great. I can't believe it was that easy... 
Now I just have to find out how I can set the script so that it doesn't regulate at +- 0W but that it regulates around -400W to -500W.

As far as I've seen, it's not enough just to adjust the value for ‘POWERMETER_TARGET_POINT = -75’ to -500, for example

In any case, thank you, irrwitzer42 and tomquist for your support.

I wish you all a pleasant weekend.
Greetings from Switzerland

Translated with DeepL.com (free version)

MavenCH, 2024-05-04:
Hello reserve85

That's a bad joke now, isn't it?
Ahoy has already provided this data, just not categorised...
I have tried it and it works great. I can't believe it was that easy... 
Now I just have to find out how I can set the script so that it doesn't regulate at +- 0W but that it regulates around -400W to -500W.

As far as I've seen, it's not enough just to adjust the value for ‘POWERMETER_TARGET_POINT = -75’ to -500, for example

In any case, thank you, irrwitzer42 and tomquist for your support.

I wish you all a pleasant weekend.
Greetings from Switzerland

reserve85, 2024-05-04:
I would never have thought that this script would be so well received here. It was actually intended for my private use… thanks at all!!

MavenCH, 2024-05-04:
Hello reserve85

We have to thank you. You have created a super easy to use tool / script that works great.
I even think that this script should be integrated into Ahoy or OpenDTU.

Thank you for your work and your great support.

Thanks and greetings from Switzerland

crisi-solar, 2024-06-15:
@MavenCH Installiere den MQTT Explorer in Homeassistant, da siehst du alles was über MQTT reinkommt.
[image]

reserve85, 2026-11-11:
The script now publishes its state via MQTT (limit per inverter and total, config values), see the MQTT_CONFIG section in the config file. Closing.

closed 2026-11-11 by reserve85