reserve85/HoymilesZeroExport #197: SELECT_INTERMEDIATE_METER to be a script ?

Thierry43, 2024-05-11:
Hello,
Would it be possible to add the option to use a script as an intermediate meter ?
I currently use a script as a power-meter to read power value from a MSunPV power router (https://ard-tek.com/). 
I would also like to use the PV power value read from the same device. 

Thanks in advance !

Thierry

Thierry43, 2024-05-12:
Great ! I will test it tomorrow. 
Just one more question: Must the intermediate meter value be positive or negative ? 

If you want to add my scripts to get data from a MsunPV Solar Power Diverter from https://ard-tek.com/ you can find them below:

-----------
#! /bin/bash
# Script to read powermeter values from a MSunPV (ard-tek.com)
IFS=";"
#Grid power
AC_GRID=$(curl -s http://MSunPV_IP/status.xml | tail -n 3 | (read var1 var2 var3; echo ${var1:7:20}))
echo ${AC_GRID%%,*}
------------------------------
#! /bin/bash
# Script to read PV Solar power values from a MSunPV (ard-tek.com)
IFS=";"
#PV Solar power
AC_PV=$(curl -s http://MSunPV_IP/status.xml | tail -n 3 | (read var1 var2 var3; echo ${var2}))
echo ${AC_PV%%,*}
--------------------------------

reserve85, 2024-05-12:
> Great ! I will test it tomorrow. Just one more question: Must the intermediate meter value be positive or negative ?

Doesn’t care if pos or neg, it’s converted to an absolute value.

Thierry43, 2024-05-13:
> > Great ! I will test it tomorrow. Just one more question: Must the intermediate meter value be positive or negative ?
> 
> Doesn’t care if pos or neg, it’s converted to an absolute value.

Perfect ! Thanks a million for your great work ;-)

reserve85, 2024-05-13:
Does everything work for you? Then I would add the scripts to the repository…

Thierry43, 2024-05-13:
Everything is working fine !

closed 2024-05-17 by reserve85