reserve85/HoymilesZeroExport #200: opposite to ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT

kthemall, 2024-05-27:
Hi,

mir scheint dass sich das zurückfahren der leistung meiner Config langsamer abspielt als das hochsetzen. ich habe momentan den Parameter: ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 0 was ja anscheinend bei batteriebetriebenen hoymiles empfohlen wird. das funktioniert auch ganz passabel (zeitverzögerung hat man ja immer wegen der übertragung auf die opendtu und von dort auf die WR. aber iwie kommt es mir vor dass das zurückfahren der leistung noch länger dauert. gibt´s da ein setting quasi als opposite zu ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 100?

danke tom

kthemall, 2024-05-27:
sorry, meine config die ich override steuere, der rest ist standard.

[COMMON]
INVERTER_COUNT = 3
SLOW_APPROX_LIMIT_IN_PERCENT = 20
SLOW_APPROX_FACTOR_IN_PERCENT = 20
LOOP_INTERVAL_IN_SECONDS = 2
SET_LIMIT_TIMEOUT_SECONDS = 5
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 0
SET_INVERTER_TO_MIN_ON_POWERMETER_ERROR = true

travisgsmith, 2024-05-28:
Yes! I would **love** to have this feature as well. I was thinking about modifying the code myself, but perhaps reserve85 could do it more quickly and easily.

I have a 11kW setup here in Canada with 8 Hoymiles HMS-1600 inverters and my system is oversized by 25% more than what my utility will allow me to feed onto the grid. It would be great if this wonderful HoymilesZeroExport script had two additional variables like:

POWERMETER_MIN_POINT
ON_FEEDING_TO_GRID_JUMP_TO_LIMIT_PERCENT

This would allow users to avoid over feeding onto the grid (for anyone who has to worry about such issues).

Hinweis: reserve85, dieses Projekt ist erstaunlich. Vielen Dank für alles, was du tust!

reserve85, 2024-05-28:
Hi,
have a look here: https://github.com/reserve85/HoymilesZeroExport/tree/dev_min_point

untested yet.

Ollipop030, 2024-05-28:
Wäre es nicht besser, wenn bei POWERMETER_MIN_POINT = -600 direkt auf [POWERMETER_TARGET_POINT + POWERMETER_TOLERANCE] geregelt wird? Den richtigen Prozentwert der Leistung kenne ich ja nicht, dieser ist ja aber auch variabel, je nachdem ob gerade 5000 Watt benötigt werden, und 600 Watt eingespeist  (WR Leistung dann bei 5600W) oder aber nur 500 Watt Bedarf und 600 Watt Einspeisung (WR Leistung 1100W).

Ollipop030, 2024-05-28:
Ich würde es vielleicht so machen:

In der ini:
POWERMETER_MIN_POINT = -600
FAST_LIMIT_DECREASE = true

Wenn false, alles wie gehabt.
Wenn true, und MIN_POINT = -600, dann direkt auf POWERMETER_TARGET_POINT + POWERMETER_TOLERANCE

Also POWERMETER_MIN_POINT - (POWERMETER_MIN_POINT - POWERMETER_TARGET_POINT + POWERMETER_TOLERANCE)

reserve85, 2024-05-28:
hab das mal so implementiert, keine ahnung ob es klappt :-)

closed 2024-05-30 by reserve85