reserve85/HoymilesZeroExport #202: Bitte um Hilfe für Update und MQTT-Einrichtung

crisi-solar, 2024-05-31:
Ich habe VERSION = 1.89 am laufen und möchte auf die aktuelle Version updaten.
Leider finde ich keine deutsche Beschreibung dazu wie das funktioniert und abläuft.
Muss man vorher das laufende Script stoppen?

Möchte auch MQTT aktivieren um die Daten in den HomeAssistant zu bekommen.
Ist es dazu notwendig in der HoymilesZeroExport_Config_Override.ini den Abschnitt 
[HOMEASSISTANT]
# --- defines for HOME ASSISTANT ---
zu konfigurieren oder reicht es beim Abschnitt
# [MQTT_CONFIG]
bei jeder Zeile die Raute am Beginn zu entfernen inkl. Zeilen 251 und 252?

Vorab schon mal vielen Dank für die Hilfen!

reserve85, 2024-05-31:
updaten: sudo ./update.sh 
du musst nichts stoppen etc.

zu MQTT:  genau wie du es beschrieben hast.

crisi-solar, 2024-05-31:
Ist es korrekt dass nach dem Update noch immer die alte Versionnummer im Override-File steht?

crisi-solar, 2024-06-01:
Gehört der Abschnitt "# Uncomment the following section if you want to use MQTT to dynamically reconfigure some settings while the script is running" zum Bereich [INTERMEDIATE_AMIS_READER]?

Oder ist es doch ein eigener Abschnitt um die Steuerung von HoymilesZeroExport von HomeAssistant zu ermöglichen?

crisi-solar, 2024-06-01:
Jetzt habe ich zumindest mal eine Verbindung
2024-06-01 04:14:27: New connection from 192.168.68.55:58611 on port 1883.
2024-06-01 04:14:27: New client connected from 192.168.68.55:58611 as HoymilesZeroExport (p2, c1, k60, u'christian').

Jetzt noch irgendwie in die Integration MQTT einbinden oder gibt es eine eigene Integration oder Add-On?

crisi-solar, 2024-06-01:
Kann wer mal bitte über diese Datei anschauen ob ein Fehler drinnen ist damit ich es als Fehlerquelle ausschließen kann?

```
# HoymilesZeroExport - https://github.com/reserve85/HoymilesZeroExport
# Copyright (C) 2023, Tobias Kraft

# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.

# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.

# You should have received a copy of the GNU General Public License
# along with this program.  If not, see <http://www.gnu.org/licenses/>.

# ---------------------------------------------------------------------
# --- DO NOT DELETE ANY ENTRIES HERE - else your script won't start ---
# ---------------------------------------------------------------------

[VERSION]
VERSION = 1.94
[SELECT_DTU]
# --- define your DTU (only one) ---
USE_AHOY = false
USE_OPENDTU = true

[SELECT_POWERMETER]
# --- define your Powermeter (only one) ---
USE_TASMOTA = false
USE_SHELLY_EM = false
USE_SHELLY_3EM = false
USE_SHELLY_3EM_PRO = true
USE_SHRDZM = false
USE_EMLOG = false
USE_IOBROKER = false
USE_HOMEASSISTANT = false
USE_VZLOGGER = false
USE_SCRIPT = false
USE_AMIS_READER = false

[AHOY_DTU]
# --- defines for AHOY-DTU ---
# in settings/inverter set interval to 6 seconds!
AHOY_IP = xxx.xxx.xxx.xxx
AHOY_PASS =

[OPEN_DTU]
# --- defines for OPEN-DTU ---
OPENDTU_IP = 192.168.68.54
OPENDTU_USER = admin
OPENDTU_PASS = openDTU42

[TASMOTA]
# --- defines for Tasmota Smartmeter Modul---
TASMOTA_IP = xxx.xxx.xxx.xxx
# if you restricted the web login enter your username and password
TASMOTA_USER = 
TASMOTA_PASS = 
# the following three constants describes how to navigate through the Tasmota-JSON. 
# if you do not know the format of your Tasmota reader, open a browser and put in the following string replacing xxx with the IP address of your Tasmota device: http://xxx.xxx.xxx.xxx/cm?cmnd=status%2010
# e.g. JSON_Result = {"StatusSNS":{"Time":"2023-02-28T12:49:49","SML":{"total_kwh":15011.575,"curr_w":-71}}}
TASMOTA_JSON_STATUS = StatusSNS
# Prefix for Web UI and MQTT JSON payload
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = SML
# Power-MQTT label (the current power in Watt, positive (import more energy than export) or negative (export more energy than import)).
TASMOTA_JSON_POWER_MQTT_LABEL = curr_w
[... 927 lines cut ...]

# grid power
#    ...
#     |
#     | -------> if you defined ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT > 0, then the limit will jump to the defined percent of HOY_MAX_WATT when reaching this point. else: increasing limit <-------
#     |
#   [0W]      [POWERMETER_MAX_POINT]
#     |
#     | -------> increasing limit <-------
#     |
#  [-50W]     [POWERMETER_TARGET_POINT + POWERMETER_TOLERANCE]
#     |
#     | -------> no limit change between -100W ... -50W <-------
#     |
#  [-75W]     [POWERMETER_TARGET_POINT]
#     |
#     | -------> no limit change between -100W ... -50W <-------
#     |
#  [-100W]    [POWERMETER_TARGET_POINT - POWERMETER_TOLERANCE]
#     |
#     | -------> decreasing limit <-------
#     |
#    ...

```

crisi-solar, 2024-06-01:
Schade dass es noch keiner geschafft hat HoymilesZeroExport mittels MQTT erfolgreich an Homeassistant anzubinden und mir deshalb zu helfen.

lullatscho, 2024-06-01:
- Mosquitto broker add on in HA installieren
- dann unter Personen in HA an besten eine extra User für mqtt einrichten (einloggen erlauben und lokales netzwerk)
- opendtu mqtt einrichten mit Hostname (IP von HA) und mqtt user Zugangsdaten

Bilder vermutlich nicht in der korrekten reihenfolge

[image]
[image]
[image]

tomquist, 2024-06-01:
Die MQTT Anbindung ist nur zur Steuerung einzelner Parameter der Nulleinspeisung gedacht. Es können darüber keine Daten ausgelesen werden. Zum Auslesen der Daten, am Besten den Wechselrichter anbinden.

Ansonsten musst du eine Anbindung an HomeAssistant selbst umsetzen, das Script unterstützt aktuell keine Auto-Discovery. Kenne mich mit HomeAssistant auch nicht so gut aus, daher kann ich leider beim Anlegen der Entitäten nur wenig helfen. Am Besten du schaust dir [diese Dokumentation](https://www.home-assistant.io/integrations/number.mqtt/) mal an und nutzt die Topics von [hier](https://github.com/reserve85/HoymilesZeroExport/blob/6b2c95575dfe1012906596f439cf5c6f9c32ab23/HoymilesZeroExport_Config.ini#L243-L250). Hier mal eine Beispiel-Yaml:
```
mqtt:
  - number:
      name: "Powermeter Target Point"
      command_topic: "zeropower/set/powermeter_target_point"
      mode: box
      min: -2000
      max: 2000
      unit_of_measurement: W
  - number:
      name: "Powermeter Max Point"
      command_topic: "zeropower/set/powermeter_max_point"
      mode: box
      min: 0
      max: 2000
      unit_of_measurement: W
  - number:
      name: "Powermeter Tolerance"
      command_topic: "zeropower/set/powermeter_tolerance"
      mode: box
      min: -200
      max: 200
      unit_of_measurement: W
  - number:
      name: "On Grid Usage - Jump to limit"
      command_topic: "zeropower/set/on_grid_usage_jump_to_limit_percent"
      min: 0
      max: 100
      unit_of_measurement: %
  - number:
      name: "Inverter 1 - Min Watt"
      command_topic: "zeropower/set/inverter/0/min_watt_in_percent"
      min: 0
      max: 100
      unit_of_measurement: %
  - number:
      name: "Inverter 1 - Normal Watt"
      command_topic: "zeropower/set/inverter/0/normal_watt"
      min: 0
      max: 2000
      unit_of_measurement: W
  - number:
      name: "Inverter 1 - Reduce Watt"
      command_topic: "zeropower/set/inverter/0/reduce_watt"
      min: 0
      max: 2000
      unit_of_measurement: W
  - number:
      name: "Inverter 1 - Battery Priority"
      command_topic: "zeropower/set/inverter/0/battery_priority"
      min: 1
      max: 5
```

crisi-solar, 2024-06-02:
@tomquist Läuft dieses MQTT nur auf dem Raspy wo auch HoymilesZeroExport läuft weil in der
MQTT_Config
# MQTT_BROKER = localhost
steht.
Das obige Mqtt habe ich schon in Homeassistant in die /homeassistant/configuration.yaml eingebunden, mußte bei 2x
unit_of_measurement: %
die % unter "%" Anführungszeichen setzen.

tomquist, 2024-06-02:
Bei korrekter config taucht da nichts auf, weil HoymilesZeroExport auf den MQTT nur lesend zugreift, d.h. es schreibt selbst dort nichts rein. Die taucht nur etwas auf, wenn du die HomeAssistant Konfiguration korrekt durchgeführt hast und eine der Werte über HomeAssistant änderst. Die IP-Adresse des HomeAssistant einzutragen ist korrekt, da dort auch der MQTT Broker läuft. Wenn korrekt eingerichtet ist, müsstest du das in den Logs von HoymilesZeroExport sehen, sobald du einen Wert in HoneAssistant veränderst.

crisi-solar, 2024-06-02:
Vielen, vielen Dank für die supertolle Hilfe, es funktioniert!!!

Nur noch 2 Fragen:
1. Wie kann ich über Homeassistant ein Reset machen auf Originalzustand?
Überall das Wort set auf reset ändern?
zeropower/set/powermeter_target_point
zeropower/reset/powermeter_target_point ?

2. Muss ich hier beim Abschnitt Homeassistant nicht meinen Sensor vom ShellyPro 3EM hinterlegen statt sensor.dtz541_sml_curr_w?
Original ist hier
HA_CURRENT_POWER_ENTITY = sensor.dtz541_sml_curr_w
eingetragen.

So schaut es jetzt in Homeassistant aus:
[image]

tomquist, 2024-06-02:
> 1. Wie kann ich über Homeassistant ein Reset machen auf Originalzustand?
>    Überall das Wort set auf reset ändern?
>    zeropower/set/powermeter_target_point
>    zeropower/reset/powermeter_target_point ?

Für die Resets solltest du besser Buttons benutzen, z.B.:
```
  - button:
      command_topic: "zeropower/reset/powermeter_target_point"
```

> 2. Muss ich hier beim Abschnitt Homeassistant nicht meinen Sensor vom ShellyPro 3EM hinterlegen statt sensor.dtz541_sml_curr_w?
>    Original ist hier
>    HA_CURRENT_POWER_ENTITY = sensor.dtz541_sml_curr_w
>    eingetragen.

Den ShellyPro 3EM kannst du auch direkt an das Script anbinden, ohne dafür über HomeAssistant zu gehen, indem du statt `USE_SHELLY_3EM` anstatt `USE_HOMEASSISTANT` benutzt und die config `[SHELLY]` konfigurierst:
```
[SHELLY]
SHELLY_IP = xxx.xxx.xxx.xxx
# if you restricted the web login enter your username and password, for Shelly "Generation 2" devices the username is always "admin" (e.g. Shelly pro 3EM)
SHELLY_USER =
SHELLY_PASS =
```

crisi-solar, 2024-06-02:
Shelly war schon eingebunden und läuft perfekt.
Buttons habe ich gesetzt, funktioniert.

Vielen Dank nochmals!

tomquist, 2024-06-03:
Die Version 1.95 gibt jetzt auch alle Config-Werte, die aktuell gesetzten Limits und einen online/offline Status an MQTT raus.

crisi-solar, 2024-06-04:
Habe heute sudo ./update.sh gemacht, jedoch ist 
HoymilesZeroExport_Config_Override.ini noch immer VERSION = 1.94
Die HoymilesZeroExport_Config.ini ist jedoch upgedatet auf VERSION = 1.95
Und eine HoymilesZeroExport_Config_Override.ini.save mit VERSION = 1.94 habe ich auch noch.

Ist das so ok?

reserve85, 2024-06-04:
Ja, die override wird nicht angefasst, das ist ja der Zweck der Datei. Dort solltest du nur deine individuelle Konfiguration haben

crisi-solar, 2024-06-04:
> Ja, die override wird nicht angefasst, das ist ja der Zweck der Datei. Dort solltest du nur deine individuelle Konfiguration haben

In der HoymilesZeroExport_Config.ini VERSION = 1.95 ist nach dem heutigen Update diese Zeile die ich in der override nicht drinnen habe.
# MQTT_LOG_LEVEL = INFO

Ausschnitt nach dem heutigen Update aus HoymilesZeroExport_Config.ini VERSION = 1.95
[image]

Ausschnitt nach dem heutigen Update aus HoymilesZeroExport_Config_Override.ini wo VERSION = 1.94 angezeigt wird:
[image]

Wenn ich in der override keine Zeilen einfügen darf, wie bekomme ich dann die neue Zeile
MQTT_LOG_LEVEL = INFO hinzu?
Oder muss ich doch noch in der config.ini
# MQTT_LOG_LEVEL = INFO zu
MQTT_LOG_LEVEL = INFO ändern?

reserve85, 2024-06-04:
Du kannst natürlich in der override die Zeile ergänzen. Die Datei „überschreibt“ die Einstellungen aus der normalen config. Alles sowas in der override nicht definiert ist, wird aus der normalen config geladen. Dadurch ist es auch übersichtlicher wenn du nur deine abweichenden Einstellungen darin speicherst. Du brauchst keine komplette „Kopie“ der config darin abzubilden.

crisi-solar, 2024-06-06:
Kann wer mir bitte von den untenstehenden Zeilen die Enitäten bzw. Topics für Homeassistant mitteilen?
Habe es selbst probiert, leider kein Erfolg. Ergebnis siehe oben.

zeropower/set/inverter/<n>/*: To change the settings of the (n+1)th inverter

zeropower/state/inverter/<n>/*: The current settings of the (n+1)th inverter

zeropower/set/on_grid_feed_fast_limit_decrease: To enable or disable the on grid feed fast limit decrease

topic zeropower/log - zeigt mir wie oben ersichtlich zwar an, wechselt jedoch viel zu schnell die Anzeige dass ich mit dem lesen nicht mitkomme.

Hat es nicht hier irgendwann / irgendwo eine deutsche Übersetzung / Erklärung der HoymilesZeroExport_Config.ini gegeben?
Finde diese leider nicht mehr.

Danke!

crisi-solar, 2024-06-08:
Welche Settings können hiermit geändert werden?
zeropower/set/inverter/<n>/*: To change the settings of the (n+1)th inverter

crisi-solar, 2024-06-17:
Es wäre toll wenn man Zero Power Log mittels MQTT ein- und ausschalten kann.
Oder habe ich diese Möglichkeit übersehen?

sibelle-labs, 2024-08-22:
Hi,
@crisi-solar Es wäre klasse wenn du die config für haos / mqtt teilen könntest oder gar github/PR ?, sowie es aussieht ist bei dir schon fertig ? Danke :-)

crisi-solar, 2024-08-23:
> Hi, @crisi-solar Es wäre klasse wenn du die config für haos / mqtt teilen könntest oder gar github/PR ?, sowie es aussieht ist bei dir schon fertig ? Danke :-)

Hallo, ich habe dann gar nicht mehr weiter gemacht da ich bei einigen Sachen nicht mehr weiterkam.

Bei manchen Punkten weiß ich gar nicht was die bewirken soll. 
So schaut es unvollendet aus:
[image]
[image]

Wenn du möchtest schicke ich dir meine komplette configuration.yaml, gib mir bitte deine Emailadresse.
LG

Bend3smo, 2024-10-08:
Moin, ich bräuchte auch mal eine funktionierende configuration.yaml Konfiguration.

crisi-solar, 2024-10-17:
@Bend3smo 
Hast du nicht diese /homeassistant/zeroexport.yaml -Datei wenn du den Fileeditor im HA startest?
[zeroexport.txt](https://github.com/user-attachments/files/17422613/zeroexport.txt)

Bend3smo, 2024-10-17:
Danke!

Nein die datei gibt es bei mir nicht. Wüsste auch nicht woher die kommen soll. Zeroexport läuft bei mir als container über Portainer.

reserve85, 2026-11-11:
Hier gab es seit 2024 keine Aktivität mehr, und es ist kein Fehler im Skript offen. Ich räume die Issue-Liste auf und schließe das Thema. Erfahrungen zu Einstellungen sammeln wir weiter in #28.

closed 2026-11-11 by reserve85