reserve85/HoymilesZeroExport #205: Nach Update kein autom. Ein bei steigender Akkuspannung

robson1311, 2024-06-05:
Hallo Leute,
wie im Titel schon geschrieben will nach dem Update (openDTU und Zeroexport) auf die neueste Version das automatische Einschalten nicht mehr funktionieren.
Habe die Einstellungen einer vorigen Version übernommen, doch da gibt es wohl irgend ein Problem.
Habe heute früh, da die Sonne fein werkelte, die Ein- u. Ausschalt-Spannungen etwas angepasst, um sehen was im Protokoll steht. 
Ausschalten bzw. reduzieren funktioniert bei 42 Volt, und bei 42,6 Volt sollte die Regelung greifen. Im Screenshot gut zu sehen das die Spannung zuerst mit einstelliger Dezimalzahl steigt, dann je näher es dem Wert kommt plötzlich neunstellig und mehr.
Die 42,6 werden aber nicht erreicht, obwohl die DTU sagt es stehen bereits 42,7 Volt an.
Wo liegt denn hier der Hund begraben?
Hänge ein paar Bilder der Situation mit an.

Vielen Dank einstweilen!
[image]
[image]
[image]

reserve85, 2024-06-05:
Von welcher Version hast du geupdated?

Aber ich habe auf die schnelle einen Fehler in der Average-berechnung gefunden:
HOY_BATTERY_AVERAGE_CNT wurde einfach ignoriert, ich hatte immer den Durchschnittswert der letzten 5 Werte berechnet. Hoffe das war das Problem, du kannst jetzt nochmal updaten.

robson1311, 2024-06-06:
Ich hab von der 1.85er Version geupdatet.
Leider hat das erneute Update nichts gebracht, die Spannungswerte steigen zwar, doch die Regelung will ab dem eingestellten Punkt nicht regeln. Es bleibt beim Inverterlimit, also den 10% - 120Watt.
Wenn ich neustarte, die Einschaltspannung ist ja bereits überschritten, geht es zuerst zurück auf 5% Minimum und regelt dann alles wie immer.
Doch wenn über Nacht das reduce limit erreicht wird, bleibt es wieder dort hängen.
LG

reserve85, 2024-06-06:
Kannst du mir das log zukommen lassen? Ggf. mit einem Bild von der openDTU?

robson1311, 2024-06-06:
Habe das Log deaktiviert, werde es einschalten. Wie kann ich es dann abrufen?
Sorry, mit reduce limit meinte ich HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V =

reserve85, 2024-06-06:
Ja nur zum Verständnis: wenn der Wert unter HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V fällt, dann wird gedrosselt auf 120W. Fällt er unter 39V dann wird der inverter ausgeschaltet und erst ab 42,6V wieder eingeschaltet. 

Die reduzierten 120W werden laut deiner ini erst ab 49,2V wieder aufgehoben! Denke das ist dein Problem?

das Log kannst du im nachhinein noch exportieren:

sudo journalctl -u HoymilesZeroExport.service -n 10000 -e > logfile.log
Falls die Zeilen nicht ausreichen kannst du die 10000 erhöhen.

robson1311, 2024-06-06:
Ohh, dann hat es wohl vor dem Bugfix deshalb bei mir funktioniert. Dachte nämlich die HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V wären die max. also "normale" Batteriespannung, wie eben bei meinem System die 49,2 Volt.
Habs jetzt angepasst und hoffe es läuft. Bin jetzt 10 Tage auf Urlaub und kann leider erst danach berichten.

Hab dir ein Bier spendiert!
**Danke für deinen Support!**

robson1311, 2024-06-25:
Schaut gut aus!
Vielen Dank!

closed 2024-06-25 by robson1311