reserve85/HoymilesZeroExport #206: POWERMETER_TARGET_POINT dynamisch anpassen

echo1483, 2024-06-06:
Hallo, bei mir läuft alles soweit prima. Ich habe jetzt eine Batterie mit AC-Ladegerät. Zum Laden werden min. 250 Watt benötigt. 
Deshalb habe ich die Einstellung „POWERMETER_TARGET_POINT = -250“ vorgenommen. 
Wenn die Batterie aufgeladen ist geht die Leistung auf 2 Watt zurück. Jetzt ist der Wert -250 viel zu hoch, eigentlich würden -75 ausreichen. Meine Batterie wird mit einem Shelly Uni überwacht, kann man die Ladespannung nutzen um den POWERMETER_TARGET_POINT zu ändern?? Wenn ja was muss man einstellen, ich selber bin nicht der Programmierer.

LG Paul

reserve85, 2024-06-14:
Den Targetpoint könntest du über die MQTT Anbindung variabel setzen.

tomquist, 2024-06-14:
Ich steuere abhängig vom stündlichen Tibber Strompreis und vom SoC meiner Batterie den target point per MQTT. Wenn der Strompreis günstig ist, setze ich ihn auf 10kW, so dass die Batterie geladen wird und das Haus aus dem Netz versorgt wird. Bei SoC > 99 regle ich den target point auf einen negativen Wert, ansonsten auf einen leicht positiven Wert (damit die Batterie nicht ins Netz entlädt).

echo1483, 2024-06-14:
Könntet ihr mal ein Beispiel einstellen. Ich habe keine Ahnung wie man das
umsetzen muss. Danke

tomquist, 2024-06-14:
Das müsstest du extern außerhalb des Scripts steuern. Z.B. über HomeAssistant oder ein anderes Heimautomatisierungssystem.

echo1483, 2024-06-14:
Bei mir läuft alles über ein Raspberry Pi. Da ist auch mein IOBroker
installiert.

tomquist, 2024-06-14:
Der ioBroker kann sicherlich auch MQTT. Ich weiß nicht ob du schon einen MQTT Broker irgendwo laufen hast. Gibt für den ioBroker auch ein paket, das einen MQTT broker enthält: https://www.iobroker.net/#en/adapters/adapterref/iobroker.mqtt/README.md

echo1483, 2024-06-14:
MQTT ist installiert, damit werden diverse Geräte gesteuert geht auch alles
ohne Probleme. Dafür benutze ich Blockly die Scripte habe ich meist im Netz
gefunden. Leider kriege ich das alleine nicht hin.

trombose009, 2024-08-26:
> Hallo, bei mir läuft alles soweit prima. Ich habe jetzt eine Batterie mit AC-Ladegerät. Zum Laden werden min. 250 Watt benötigt. Deshalb habe ich die Einstellung „POWERMETER_TARGET_POINT = -250“ [...]

hi paul, wie kann man mit dir kontakt aufnehmen? würde mich für/über ein paar infos bzgl. des AC ladegeräts und den aufbau interessieren.
sweetie.gwendoline@gmail.com

reserve85, 2026-11-11:
Der Zielwert lässt sich inzwischen zur Laufzeit per MQTT ändern (powermeter_target_point, siehe Abschnitt MQTT_CONFIG in der Config). Damit kann man ihn z. B. abhängig vom Ladezustand der Batterie setzen. Ich schließe das Issue.

closed 2026-11-11 by reserve85