reserve85/HoymilesZeroExport #208: nullanpassung beschleunigen

kthemall, 2024-06-13:
hi, könnte man die 0W anpassung beschleunigen indem man zwischen dem setzen der limits der (in meinem fall 3) Wechselrichter eine erneute aktuelle verbrauchsabfrage startet? wie man sieht vergehen zwischen dem abfragen der leistung und dem setzen aller 3 wechselrichter beinahe 10 sek. das ist zu lang für zb. ein ceranfeld welches beim kochen ja immer pulst. oder gibt es so ein setting vielleicht schon irgendwo in der config datei und ich finde sie nicht? ich speise somit eigentlich fast 10 sekunden mit 1800w ein obwohl der ofen gar nicht mehr aktiv ist und die leistung nicht mehr benötigt wird. 

danke tom

Jun 13 10:46:43 iobroker python3[23125]: 2024-06-13 **10:46:43** INFO     setting new limit to 1835 Watt
Jun 13 10:46:43 iobroker python3[23125]: 2024-06-13 10:46:43 INFO     OpenDTU: Inverter "HM-800-Phase1": setting new limit from 90 Watt to 611 Watt
Jun 13 10:46:46 iobroker python3[23125]: 2024-06-13 10:46:46 INFO     OpenDTU: Inverter "HM-800-Phase1": Limit timeout!
Jun 13 10:46:46 iobroker python3[23125]: 2024-06-13 10:46:46 INFO     OpenDTU: Inverter "HM-800-Phase2": setting new limit from 90 Watt to 611 Watt
Jun 13 10:46:49 iobroker python3[23125]: 2024-06-13 10:46:49 INFO     OpenDTU: Inverter "HM-800-Phase2": Limit acknowledged
Jun 13 10:46:49 iobroker python3[23125]: 2024-06-13 10:46:49 INFO     OpenDTU: Inverter "HM-800-Phase3": setting new limit from 90 Watt to 611 Watt
Jun 13 10:46:52 iobroker python3[23125]: 2024-06-13 10:46:52 INFO     OpenDTU: Inverter "HM-800-Phase3": Limit acknowledged
Jun 13 10:46:52 iobroker python3[23125]: 2024-06-13 10:46:52 INFO     OpenDTU: Inverter "HM-800-Phase1" reachable: True
Jun 13 10:46:52 iobroker python3[23125]: 2024-06-13 10:46:52 INFO     OpenDTU: Inverter "HM-800-Phase2" reachable: True
Jun 13 10:46:52 iobroker python3[23125]: 2024-06-13 10:46:52 INFO     OpenDTU: Inverter "HM-800-Phase3" reachable: True
Jun 13 10:46:52 iobroker python3[23125]: 2024-06-13 **10:46:52** INFO     powermeter AmisReader: -1243 Watt

reserve85, 2024-06-14:
Schwierig, in dem gesamten System ist überall Verzögerung drin. 
Die Regelung teilt die Leistung auch immer auf alle Inverter auf...
Selbst nach dem Anpassen dauert es u.U. ein paar Sek. bis der Hoymiles die Regelung übernimmt und du das dann am Powermeter siehst. Die DTU aktualisiert ja auch nur alle x sekunden ihren Wert und beim Limit-Setzen wird das auch noch priorisiert.
Ein Ceranfeld ist dann auch immer suboptimal, da es ständig kurz ein und wieder ausschaltet.

kthemall, 2024-06-14:
alles klar, dann werden wir wohl damit leben müssen. aber danke für deine super arbeit in dem projekt. LG

kthemall, 2024-06-14:
kann zu

closed 2024-06-14 by kthemall