reserve85/HoymilesZeroExport #209: Einspeisung nach reglung im Monat

fatzo, 2024-06-13:
Hallo zusammen
Was bleibt den so übrig nach eurer Reglung?
Was speißt Ihr so ins Netz ein im Durschschnitt bei 600/800/etc Watt.

gruß Fatzo

kthemall, 2024-06-14:
bei mir ung. 0,8kWh 

LG

tomquist, 2024-06-14:
Bei mir waten es letzten Monat ca. 0,06kWh pro Tag. Habe den Target Point aber auch in der Regel auf einem Wert leicht über 0, wenn die Batterie nicht voll ist, so dass durch leichte Schwankungen des Verbrauchs nicht überproduziert wird. Hat natürlich den Preis, dass dann ein bisschen mehr als nötig von Netz bezogen wird.

Ollipop030, 2024-06-14:
1,8KW Anlage, hab im Mai 21,37kWh verschenkt. Liegt aber am Stromzähler, der mir nur alle 10 Sekunden einen aktuellen Wert schickt, daher dauert das Runterregeln immer etwas länger.

kthemall, 2024-06-14:
> > bei mir ung. 0,8kWh
> > LG
> 
> am Tag?

jupp: 
[image]

reserve85, 2024-06-14:
da spielt aber auch mit wie viele Wechselrichter etc.
Die Regelung wird mit jedem Wechselrichter der dazu kommt auch langsamer...

kthemall, 2024-06-14:
> da spielt aber auch mit wie viele Wechselrichter etc. Die Regelung wird mit jedem Wechselrichter der dazu kommt auch langsamer...

ja klar, ich hab 3x HM800 und pulsende Geräte wie Ceranfelder beim kochen usw. das spielt da ne große Rolle. prinz. ist mir das ja egal da ich die Einspeisung vergütet bekomme und technisch ist das ja auch fast nicht mehr zu optimieren (ich meine scripttechnisch). da sind die Hoymiles + opendtu oder ahoy einfach zu träge. Nichts desto trotz ein super Projekt und man spart dadurch einiges an Euros. Danke dafür nochmal @reserve85

Ollipop030, 2024-06-14:
Man kann auch versuchen den Eigenverbrauch zu erhöhen. Ich habe diese Woche eine Brauchwasserwärmepumpe bestellt. Die verbraucht im Betrieb knapp 600W, bei meinen 1,8KW sollte das locker drin sein (zumindest bei Sonnenschein).

fatzo, 2024-06-15:
> Man kann auch versuchen den Eigenverbrauch zu erhöhen. Ich habe diese Woche eine Brauchwasserwärmepumpe bestellt. Die verbraucht im Betrieb knapp 600W, bei meinen 1,8KW sollte das locker drin sein (zumindest bei Sonnenschein).

Ich bin am überlegen mir noch einen Heizstab in den Warmwassertank zu instalieren, bin mir aber noch nicht schlüssig wie ich das umsetze. Vor allem welcher Heizstab !

fatzo, 2024-06-15:
ich habe jetzt in 15 Tagen 10 kw eingespeißt, das heißt durchschnitt 0,666 kw am Tag

robson1311, 2024-08-26:
Also ich habe gerade nachgesehen was bei meiner Anlage so eingespeist wird. Gar nicht mal so wenig, denn im Juli waren es rund 40kwh und bis jetzt im August 35kwh. 
Ist natürlich wertvolle Energie die bei Speicherbetrieb und einem HM-1200 besser genutzt werden könnte.

Muss nachsehen, aber ich glaube mein Target Point ist auf -55w.
Wenn ich den weiter runter setze, sollte weniger eingespeist, aber dafür etwas mehr bezogen werden. 
Sehe ich das richtig, so wie @tomquist schreibt, könnte man den Target sogar Richtung 0 setzen?

tomquist, 2024-08-26:
Als generelle Richtlinie würde ich sagen, wenn der Speicher in der Regel nicht leer wird, dann setzte den Target Point auf einen negativen Wert. Wenn der Speicher am Ende des Tages immer leer ist (so wie bei mir), dann setze ihn auf einen leicht positiven Wert. Dadurch wird die Eigennutzungsrate maximiert.

robson1311, 2024-08-26:
> Als allgemeine Richtlinie würde ich sagen, wenn der Speicher in der Regel nicht leer wird, dann setzt den Target Point auf einen negativen Wert. Wenn der Speicher am Ende des Tages immer leer ist (so wie bei mir), dann setze ihn auf einen leicht positiven Wert. Dadurch wird die Eigennutzungsrate maximiert.

Danke dir, das werde ich testen!

robson1311, 2024-08-30:
Hab den Targetpoint jetzt auf 0 gestellt, und siehe da, die Einspeisung hat sich seitdem in etwa halbiert.
Läuft soweit gut, vielen Dank!

reserve85, 2026-11-11:
Hier gab es seit 2024 keine Aktivität mehr, und es ist kein Fehler im Skript offen. Ich räume die Issue-Liste auf und schließe das Thema. Erfahrungen zu Einstellungen sammeln wir weiter in #28.

closed 2026-11-11 by reserve85