reserve85/HoymilesZeroExport #211: Problem HoymilesZeroExport mit openDTU - hängt in der Früh

crisi-solar, 2024-06-18:
openDTU v.24.6.10
HoymilesZeroExport V1.95 auf Raspi pi zero w neu installiert - MQTT ist NICHT aktiviert

Seit längerem hat die openDTU in der Früh keine Verbindung zu den zwei WR mehr, in der Ereignisanzeige der DTU wird noch die Startzeit der WR angezeigt, danach ist funkstille. Auch wenn es am Abend finster wird wird der Balken nicht rot sondern bleibt bis in der Früh gelb.
Egal wie oft jetzt die WR oder DTU neu gestartet oder stromlos gemacht und danach bis 15 Minuten gewartet wird, der Balken bleibt immer gelb und es passiert nichts.
Meistens half es nur das Netzwerk komplett zu starten, dann lief alles einwandfrei bis zum nächsten Morgen.

Durch Zufall bin ich aber draufgekommen dass es am einfachsten ist in der Früh ZeroExport runterzufahren, dann bei der DTU einen Neustart machen und nach einigen Minuten wechselt der Balken von gelb auf rot und nach weiteren Minuten auf blau und es läuft. Danach ZeroExport wieder starten und es läuft den ganzen Tag ohne Probleme bis nächsten Morgen.

Das Problem kann meiner Meinung nach nur an einem Bug in der Software liegen weil ich letzte Woche einige Tage statt ZeroExport den SMS verwendet habe und da lief mit der gleichen DTU alles astrein: 
am Abend kam der rote Balken, in der Früh wurde der Balken gelb und dann blau ohne einmal ein Gerät neu starten zu müssen.

LG

crisi-solar, 2024-06-19:
So sah die DTU-Anzeige heute Früh aus, die WR produzieren schon:
[image]

Ausschnitt von der Konsole:
[image]

reserve85, 2024-06-19:
Interessant wäre das Log vom Script,

Ollipop030, 2024-06-19:
> Das Problem kann meiner Meinung nach nur an einem Bug in der Software liegen

Wie schon mein Vorredner: Das Script sendet ja nur alle paar Sekunden ein Limit, sonst wird ja nichts übertragen. Ich nutze Ahoy, und ab einem gewissen Punkt muss ich auch das neu starten, weil es einfach hängt (passiert ca. nach einer Woche, manchmal nach drei Tagen). Ich denke mal, OpenDTU und Ahoy haben viel damit zu tun, alle paar Sekunden Limits zu setzen, und irgendwann ist der Speicher voll oder sowas. Sind ja alles nur Hobbyentwickler hier.
Aber wie @reserve85 schon anmerkte, vielleicht ist dein Timing zu schnell? Oder Limit zu niedrig? Die Hoymiles mögen keine Limits <5%.
Poste mal deine Config.

crisi-solar, 2024-06-19:
Wie kann ich dir das Logfile privat zukommen lassen, hat ca. 7 MB?

Das wäre Zufall dass meine zwei DTUs das selbe Problem haben sollten.
Mit dem SMS funktionieren beide DTUs ohne Probleme mit den beiden WR. 
Aber alles ist möglich.

Falls das Timing zu schnell sein sollte (auf der DTU sind 5 Sekunden Abfrageintervall eingestellt) sollte ein eventuelles Problem auch Untertags auftreten und nicht nur wenn sich beide WR am Abend abschalten.

reserve85, 2024-06-19:
kannst du hier als *.txt einfach anhängen

reserve85, 2024-06-19:
schau mal hier, evtl das gleiche Problem:
https://github.com/tbnobody/OpenDTU/issues/2040

smaicloud, 2024-06-19:
Ich hatte auch mal das Problem mit der openDTU. Anfangs half immer neu starten. Irgendwann ging überhaupt keine Verbindung mehr. Hatte mehrfach die Software aktualisiert, von mal zu mal wurde es gefühlt schlimmer. Habe dann Ahoy installiert und seit dem läuft es problemlos. Seit Wochen keinen Neustart mehr und alles funktioniert. Das zero Export Script kann ja nicht Schuld sein, es steuert ja nur die DTU an.  Vielleicht mal die Frequenz wechseln?

crisi-solar, 2024-06-19:
[logfile.txt](https://github.com/user-attachments/files/15899713/logfile.txt)

@smaicloud Das alle zwei verschiedene DTUs mit verschiedener Software gemeinsam spinnen wäre purer Zufall.
Wie schon geschrieben funktionieren beide DTUs mit SmartMicroSolar und den beiden WR hervorragend.

reserve85, 2024-06-19:
```
Jun 19 02:49:49 raspberrypi python3[506]: 2024-06-19 02:49:49 INFO     OpenDTU: Inverter "HMS-1600-4T": setting new limit from 1600 Watt to 1600 Watt
Jun 19 02:49:55 raspberrypi python3[506]: 2024-06-19 02:49:55 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit timeout!
Jun 19 02:49:55 raspberrypi python3[506]: 2024-06-19 02:49:55 INFO     OpenDTU: Inverter "HMS-2000-4T": setting new limit from 2000 Watt to 2000 Watt
Jun 19 02:50:01 raspberrypi python3[506]: 2024-06-19 02:50:01 INFO     OpenDTU: Inverter "HMS-2000-4T": Limit timeout!
Jun 19 02:50:06 raspberrypi python3[506]: 2024-06-19 02:50:06 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Jun 19 02:50:06 raspberrypi python3[506]: 2024-06-19 02:50:06 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: True
```

Konnte gerade nur durchscrollen, aber sehr auffällig, dass "reachable: true" die ganze nacht bei beiden Invertern auf "true" steht. 
Hängen die an einer Batterie? Denn wenn das auf "True" steht, dann wird natürlich auch versucht zu regeln.

Auch hier beim vermutlichen "Ausschalten weil keine Sonne mehr da":

```
Jun 18 21:19:16 raspberrypi python3[506]: 2024-06-18 21:19:16 INFO     powermeter Shelly3EMPro: 332 Watt
Jun 18 21:19:16 raspberrypi python3[506]: 2024-06-18 21:19:16 INFO     Inverterlimit was already accepted at 1600 Watt
Jun 18 21:19:23 raspberrypi python3[506]: 2024-06-18 21:19:23 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Jun 18 21:19:24 raspberrypi python3[506]: 2024-06-18 21:19:24 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: False
Jun 18 21:19:24 raspberrypi python3[506]: 2024-06-18 21:19:24 INFO     powermeter Shelly3EMPro: 328 Watt
Jun 18 21:19:24 raspberrypi python3[506]: 2024-06-18 21:19:24 INFO     Inverterlimit was already accepted at 1600 Watt
Jun 18 21:19:29 raspberrypi python3[506]: 2024-06-18 21:19:29 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Jun 18 21:19:29 raspberrypi python3[506]: 2024-06-18 21:19:29 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: False
Jun 18 21:19:29 raspberrypi python3[506]: 2024-06-18 21:19:29 INFO     powermeter Shelly3EMPro: 329 Watt
Jun 18 21:19:29 raspberrypi python3[506]: 2024-06-18 21:19:29 INFO     Inverterlimit was already accepted at 1600 Watt
Jun 18 21:19:34 raspberrypi python3[506]: 2024-06-18 21:19:34 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: False
Jun 18 21:19:35 raspberrypi python3[506]: 2024-06-18 21:19:35 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: False
Jun 18 21:19:40 raspberrypi python3[506]: 2024-06-18 21:19:40 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: False
Jun 18 21:19:40 raspberrypi python3[506]: 2024-06-18 21:19:40 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: True
Jun 18 21:19:40 raspberrypi python3[506]: 2024-06-18 21:19:40 INFO     OpenDTU: Inverter "HMS-2000-4T" / serial number "116491439579" / temperature 26.3 degC
Jun 18 21:19:40 raspberrypi python3[506]: 2024-06-18 21:19:40 INFO     powermeter Shelly3EMPro: 333 Watt
Jun 18 21:19:40 raspberrypi python3[506]: 2024-06-18 21:19:40 INFO     setting new limit to 1933 Watt
Jun 18 21:19:40 raspberrypi python3[506]: 2024-06-18 21:19:40 INFO     OpenDTU: Inverter "HMS-2000-4T": setting new limit from -1 Watt to 1933 Watt
Jun 18 21:19:46 raspberrypi python3[506]: 2024-06-18 21:19:46 INFO     OpenDTU: Inverter "HMS-2000-4T": Limit timeout!
Jun 18 21:19:51 raspberrypi python3[506]: 2024-06-18 21:19:51 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Jun 18 21:19:51 raspberrypi python3[506]: 2024-06-18 21:19:51 INFO     OpenDTU: Inverter "HMS-1600-4T" / serial number "116491626836" / temperature 27.0 degC
Jun 18 21:19:51 raspberrypi python3[506]: 2024-06-18 21:19:51 INFO     OpenDTU: Inverter "HMS-2000-4T" / serial number "116491439579" / temperature 26.3 degC
Jun 18 21:19:51 raspberrypi python3[506]: 2024-06-18 21:19:51 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: True
Jun 18 21:19:52 raspberrypi python3[506]: 2024-06-18 21:19:52 INFO     powermeter Shelly3EMPro: 324 Watt
Jun 18 21:19:52 raspberrypi python3[506]: 2024-06-18 21:19:52 INFO     setting new limit to 2257 Watt
Jun 18 21:19:52 raspberrypi python3[506]: 2024-06-18 21:19:52 INFO     OpenDTU: Inverter "HMS-1600-4T": setting new limit from -1 Watt to 1003 Watt
Jun 18 21:19:58 raspberrypi python3[506]: 2024-06-18 21:19:58 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit timeout!
Jun 18 21:19:58 raspberrypi python3[506]: 2024-06-18 21:19:58 INFO     OpenDTU: Inverter "HMS-2000-4T": setting new limit from 1933 Watt to 1253 Watt
Jun 18 21:20:04 raspberrypi python3[506]: 2024-06-18 21:20:04 INFO     OpenDTU: Inverter "HMS-2000-4T": Limit timeout!
Jun 18 21:20:10 raspberrypi python3[506]: 2024-06-18 21:20:10 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Jun 18 21:20:10 raspberrypi python3[506]: 2024-06-18 21:20:10 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: True
Jun 18 21:20:10 raspberrypi python3[506]: 2024-06-18 21:20:10 INFO     powermeter Shelly3EMPro: 328 Watt
Jun 18 21:20:10 raspberrypi python3[506]: 2024-06-18 21:20:10 INFO     setting new limit to 2585 Watt
Jun 18 21:20:10 raspberrypi python3[506]: 2024-06-18 21:20:10 INFO     OpenDTU: Inverter "HMS-1600-4T": setting new limit from 1003 Watt to 1148 Watt
Jun 18 21:20:16 raspberrypi python3[506]: 2024-06-18 21:20:16 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit timeout!
Jun 18 21:20:16 raspberrypi python3[506]: 2024-06-18 21:20:16 INFO     OpenDTU: Inverter "HMS-2000-4T": setting new limit from 1253 Watt to 1436 Watt
Jun 18 21:20:22 raspberrypi python3[506]: 2024-06-18 21:20:22 INFO     OpenDTU: Inverter "HMS-2000-4T": Limit timeout!
```

Da werden die Inverter ganz kurz "inaktiv" und im nächsten Durchgang wieder "aktiv".
Sieht für mich so aus, als würde in der OpenDTU Firmware was nicht passen. Evtl nutzt SmartMicroSolar eine andere Schnittstelle, z.B. MQTT o.Ä.?

crisi-solar, 2024-06-19:
Akku habe ich keinen.
Vielleicht sind diese Einstellungen in der DTU falsch und gehören alle ausgeschaltet?
[image]

SMS hat meines Wissens kein MQTT da ich den sonst schon in HomeAssistant eingebunden hätte.

reserve85, 2024-06-19:
ich habe keine Ahnung wie die Abfrage der Inverter bei SMS funktioniert, vielleicht wird da einfach nur geschaut ob "0W" erzeugt wird. Ist nicht open source, daher kann ich auch nicht nachschauen.

Jedenfalls sehe ich im Log erstmal keinen Fehler, für mich ist eindeutig der Rückgabewert der DTU falsch.

Die andere Frage ist auch noch, ob das auch wirklich das verursachende Problem des "DTU Absturzes" ist.

Morgens sind laut im Log die Inverter übrigens auch wieder kurz "inactive" und werden dann im nächsten Schritt "Active":

```
Jun 19 04:46:03 raspberrypi python3[506]: 2024-06-19 04:46:03 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit timeout!
Jun 19 04:46:03 raspberrypi python3[506]: 2024-06-19 04:46:03 INFO     OpenDTU: Inverter "HMS-2000-4T": setting new limit from 2000 Watt to 2000 Watt
Jun 19 04:46:09 raspberrypi python3[506]: 2024-06-19 04:46:09 INFO     OpenDTU: Inverter "HMS-2000-4T": Limit timeout!
Jun 19 04:46:14 raspberrypi python3[506]: 2024-06-19 04:46:14 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Jun 19 04:46:14 raspberrypi python3[506]: 2024-06-19 04:46:14 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: True
Jun 19 04:46:14 raspberrypi python3[506]: 2024-06-19 04:46:14 INFO     powermeter Shelly3EMPro: 213 Watt
Jun 19 04:46:14 raspberrypi python3[506]: 2024-06-19 04:46:14 INFO     Inverterlimit 3600 Watt was previously not accepted by at least one inverter, trying again...
Jun 19 04:46:14 raspberrypi python3[506]: 2024-06-19 04:46:14 INFO     setting new limit to 3600 Watt
Jun 19 04:46:14 raspberrypi python3[506]: 2024-06-19 04:46:14 INFO     OpenDTU: Inverter "HMS-1600-4T": setting new limit from 1600 Watt to 1600 Watt
Jun 19 04:46:20 raspberrypi python3[506]: 2024-06-19 04:46:20 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit timeout!
Jun 19 04:46:20 raspberrypi python3[506]: 2024-06-19 04:46:20 INFO     OpenDTU: Inverter "HMS-2000-4T": setting new limit from 2000 Watt to 2000 Watt
Jun 19 04:46:25 raspberrypi python3[506]: 2024-06-19 04:46:25 INFO     OpenDTU: Inverter "HMS-2000-4T": Limit timeout!
Jun 19 04:46:30 raspberrypi python3[506]: 2024-06-19 04:46:30 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Jun 19 04:46:30 raspberrypi python3[506]: 2024-06-19 04:46:30 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: True
Jun 19 04:46:31 raspberrypi python3[506]: 2024-06-19 04:46:31 INFO     powermeter Shelly3EMPro: 218 Watt
Jun 19 04:46:31 raspberrypi python3[506]: 2024-06-19 04:46:31 INFO     Inverterlimit 3600 Watt was previously not accepted by at least one inverter, trying again...
Jun 19 04:46:31 raspberrypi python3[506]: 2024-06-19 04:46:31 INFO     setting new limit to 3600 Watt
Jun 19 04:46:31 raspberrypi python3[506]: 2024-06-19 04:46:31 INFO     OpenDTU: Inverter "HMS-1600-4T": setting new limit from 1600 Watt to 1600 Watt
Jun 19 04:46:36 raspberrypi python3[506]: 2024-06-19 04:46:36 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit timeout!
Jun 19 04:46:36 raspberrypi python3[506]: 2024-06-19 04:46:36 INFO     OpenDTU: Inverter "HMS-2000-4T": setting new limit from 2000 Watt to 2000 Watt
Jun 19 04:46:47 raspberrypi python3[506]: 2024-06-19 04:46:47 WARNING  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPConnectionPool(host='192.168.68.54', port=80): Read timed out. (read timeout=10)")': /api/limit/status
Jun 19 04:46:49 raspberrypi python3[506]: 2024-06-19 04:46:49 INFO     OpenDTU: Inverter "HMS-2000-4T": Limit acknowledged
Jun 19 04:46:55 raspberrypi python3[506]: 2024-06-19 04:46:55 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Jun 19 04:46:55 raspberrypi python3[506]: 2024-06-19 04:46:55 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: True
Jun 19 04:46:55 raspberrypi python3[506]: 2024-06-19 04:46:55 INFO     powermeter Shelly3EMPro: 213 Watt
Jun 19 04:46:55 raspberrypi python3[506]: 2024-06-19 04:46:55 INFO     Inverterlimit 3600 Watt was previously not accepted by at least one inverter, trying again...
Jun 19 04:46:55 raspberrypi python3[506]: 2024-06-19 04:46:55 INFO     setting new limit to 3600 Watt
Jun 19 04:46:55 raspberrypi python3[506]: 2024-06-19 04:46:55 INFO     OpenDTU: Inverter "HMS-1600-4T": setting new limit from 1600 Watt to 1600 Watt
Jun 19 04:47:04 raspberrypi python3[506]: 2024-06-19 04:47:04 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit timeout!
Jun 19 04:47:10 raspberrypi python3[506]: 2024-06-19 04:47:10 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Jun 19 04:47:10 raspberrypi python3[506]: 2024-06-19 04:47:10 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: True
Jun 19 04:47:10 raspberrypi python3[506]: 2024-06-19 04:47:10 INFO     powermeter Shelly3EMPro: 216 Watt
Jun 19 04:47:10 raspberrypi python3[506]: 2024-06-19 04:47:10 INFO     Inverterlimit 3600 Watt was previously not accepted by at least one inverter, trying again...
Jun 19 04:47:10 raspberrypi python3[506]: 2024-06-19 04:47:10 INFO     setting new limit to 3600 Watt
Jun 19 04:47:10 raspberrypi python3[506]: 2024-06-19 04:47:10 INFO     OpenDTU: Inverter "HMS-1600-4T": setting new limit from 1600 Watt to 1600 Watt
Jun 19 04:47:17 raspberrypi python3[506]: 2024-06-19 04:47:17 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit timeout!
Jun 19 04:47:23 raspberrypi python3[506]: 2024-06-19 04:47:23 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
[... cut ...]
Jun 19 04:49:05 raspberrypi python3[506]: 2024-06-19 04:49:05 INFO     OpenDTU: Inverter "HMS-1600-4T": setting new limit from 1001 Watt to 1095 Watt
Jun 19 04:49:08 raspberrypi python3[506]: 2024-06-19 04:49:08 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit acknowledged
Jun 19 04:49:08 raspberrypi python3[506]: 2024-06-19 04:49:08 INFO     OpenDTU: Inverter "HMS-2000-4T": setting new limit from 1251 Watt to 1369 Watt
Jun 19 04:49:10 raspberrypi python3[506]: 2024-06-19 04:49:10 INFO     OpenDTU: Inverter "HMS-2000-4T": Limit acknowledged
```

Die DTU akzeptiert da auch das Limit. Sieht für mich alles OK aus bis auf das "Active" nachts...

crisi-solar, 2024-06-19:
Danke für die bisherige Info und Mühe!!

Das in der Früh mit Active und Inactive habe ich bei SMS auch wenn die WR gerade an der Schwelle beim Mindestwert für die Einschaltung sind. Da geht es um 1 Watt bei einer von den 4 Modulen das noch "unrund" ist.

Ich habe bei beiden WR in der DTU Befehle Nachts senden und Daten Nachts abrufen deaktiviert.

Wenn ich mich noch richtig erinnere fing das Problem an als ich den zweiten WR dazukaufte, bin mir aber nicht so sicher.

crisi-solar, 2024-06-20:
Heute Früh waren die Balken beider WR in der DTU noch immer gelb und keine Verbindung zu den WR und die beiden WR produzieren trotzdem schon Strom.
Also brachte die Ausschaltung der Befehle "Nachts senden" und "Nachts abrufen" bei beiden WR keine Ändereung.

Ich aktivierte also diese beiden Befehle wieder, speicherte diese, änderte zum Spaß die Reihenfolge der Wechselrichter und siehe da, nach einigen Minuten wurden die Balken rot, dann gelb und anschließend blau.

Werde heute Nachmittag wieder mal den SMS anhängen und schauen ob sich die DTU am Abend abschaltet.

reserve85, 2024-06-20:
Ok, zeige mir bitte noch das Log von heute Nacht

crisi-solar, 2024-06-20:
Woher kommt das Limit -1 wenn es nirgends gesetzt wurde?
Jun 19 21:20:40 raspberrypi python3[2739]: 2024-06-19 21:20:40 INFO     OpenDTU: Inverter "HMS-2000-4T": setting new limit from -1 Watt to 1130 Watt

[logfile_20.6.txt](https://github.com/user-attachments/files/15910270/logfile_20.6.txt)

reserve85, 2024-06-20:
ganz normal, sobald der inverter als "not reachable" gemeldet wird, wird intern das limit gelöscht (auf -1). 

aber schau mal hier:

```
Jun 20 03:02:52 raspberrypi python3[2739]: 2024-06-20 03:02:52 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: False
Jun 20 03:02:53 raspberrypi python3[2739]: 2024-06-20 03:02:53 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: False
Jun 20 03:02:58 raspberrypi python3[2739]: 2024-06-20 03:02:58 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: False
Jun 20 03:02:58 raspberrypi python3[2739]: 2024-06-20 03:02:58 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: False
Jun 20 03:03:03 raspberrypi python3[2739]: 2024-06-20 03:03:03 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: False
Jun 20 03:03:03 raspberrypi python3[2739]: 2024-06-20 03:03:03 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: False
Jun 20 03:03:09 raspberrypi python3[2739]: 2024-06-20 03:03:09 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: False
Jun 20 03:03:09 raspberrypi python3[2739]: 2024-06-20 03:03:09 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: False
Jun 20 03:03:14 raspberrypi python3[2739]: 2024-06-20 03:03:14 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: False
Jun 20 03:03:14 raspberrypi python3[2739]: 2024-06-20 03:03:14 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: False
Jun 20 03:03:19 raspberrypi python3[2739]: 2024-06-20 03:03:19 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: False
Jun 20 03:03:20 raspberrypi python3[2739]: 2024-06-20 03:03:20 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: False
Jun 20 03:03:25 raspberrypi python3[2739]: 2024-06-20 03:03:25 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: False
Jun 20 03:03:25 raspberrypi python3[2739]: 2024-06-20 03:03:25 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: False
Jun 20 03:03:30 raspberrypi python3[2739]: 2024-06-20 03:03:30 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: False
Jun 20 03:03:30 raspberrypi python3[2739]: 2024-06-20 03:03:30 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: False
Jun 20 03:03:36 raspberrypi python3[2739]: 2024-06-20 03:03:36 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: False
Jun 20 03:03:36 raspberrypi python3[2739]: 2024-06-20 03:03:36 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: False
Jun 20 03:03:41 raspberrypi python3[2739]: 2024-06-20 03:03:41 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Jun 20 03:03:41 raspberrypi python3[2739]: 2024-06-20 03:03:41 INFO     OpenDTU: Inverter "HMS-1600-4T" / serial number "116491626836" / temperature 30.3 degC
Jun 20 03:03:41 raspberrypi python3[2739]: 2024-06-20 03:03:41 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: True
Jun 20 03:03:41 raspberrypi python3[2739]: 2024-06-20 03:03:41 INFO     OpenDTU: Inverter "HMS-1600-4T" / serial number "116491626836" / temperature 30.3 degC
Jun 20 03:03:41 raspberrypi python3[2739]: 2024-06-20 03:03:41 INFO     OpenDTU: Inverter "HMS-2000-4T" / serial number "116491439579" / temperature 29.4 degC
Jun 20 03:03:42 raspberrypi python3[2739]: 2024-06-20 03:03:42 INFO     powermeter Shelly3EMPro: 197 Watt
Jun 20 03:03:42 raspberrypi python3[2739]: 2024-06-20 03:03:42 INFO     setting new limit to 3600 Watt
Jun 20 03:03:42 raspberrypi python3[2739]: 2024-06-20 03:03:42 INFO     OpenDTU: Inverter "HMS-1600-4T": setting new limit from -1 Watt to 1600 Watt
Jun 20 03:03:47 raspberrypi python3[2739]: 2024-06-20 03:03:47 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit timeout!
Jun 20 03:03:47 raspberrypi python3[2739]: 2024-06-20 03:03:47 INFO     OpenDTU: Inverter "HMS-2000-4T": setting new limit from -1 Watt to 2000 Watt
Jun 20 03:03:53 raspberrypi python3[2739]: 2024-06-20 03:03:53 INFO     OpenDTU: Inverter "HMS-2000-4T": Limit timeout!
Jun 20 03:03:58 raspberrypi python3[2739]: 2024-06-20 03:03:58 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Jun 20 03:03:58 raspberrypi python3[2739]: 2024-06-20 03:03:58 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: True
Jun 20 03:03:59 raspberrypi python3[2739]: 2024-06-20 03:03:59 INFO     powermeter Shelly3EMPro: 195 Watt
Jun 20 03:03:59 raspberrypi python3[2739]: 2024-06-20 03:03:59 INFO     Inverterlimit 3600 Watt was previously not accepted by at least one inverter, trying again...
Jun 20 03:03:59 raspberrypi python3[2739]: 2024-06-20 03:03:59 INFO     setting new limit to 3600 Watt
Jun 20 03:03:59 raspberrypi python3[2739]: 2024-06-20 03:03:59 INFO     OpenDTU: Inverter "HMS-1600-4T": setting new limit from 1600 Watt to 1600 Watt
Jun 20 03:04:04 raspberrypi python3[2739]: 2024-06-20 03:04:04 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit timeout!
Jun 20 03:04:04 raspberrypi python3[2739]: 2024-06-20 03:04:04 INFO     OpenDTU: Inverter "HMS-2000-4T": setting new limit from 2000 Watt to 2000 Watt
Jun 20 03:04:09 raspberrypi python3[2739]: 2024-06-20 03:04:09 INFO     OpenDTU: Inverter "HMS-2000-4T": Limit timeout!
Jun 20 03:04:14 raspberrypi python3[2739]: 2024-06-20 03:04:14 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Jun 20 03:04:14 raspberrypi python3[2739]: 2024-06-20 03:04:14 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: True
```

alles so wie es sein soll (reachable: false) und um 3:04 erwachen deine inverter um zu produzieren. das kann niemals stimmen, da ist es doch noch stockdunkel..

crisi-solar, 2024-06-20:
Das mit der Uhrzeit ist wirklich rätselhaft.
Habe in der Logdatei von Homeassistant in dieser Zeit keinen Eintrag von der OpenDTU.

Werde mal in dem beschriebenen Link einen Verweis auf hierher verweisen.

Gibt es eine Möglichkeit irgendwie eine Logdatei aufzuzeichnen zwischen openDTU mit SMS wenn ich diesen am Nachmittag anstecke?

Nochmals vielen Dank für die Hilfestellung und deine geopferte Zeit!

crisi-solar, 2024-06-20:
Bei der Verwendung von SMS schalten sich beide WR wie es sein soll ab.
Werden sich morgen vermutlich auch ohne Probleme wieder einschalten.

Irgendwie verstehe ich das ganze wirklich nicht mehr.

[image]

reserve85, 2024-06-20:
MMn liegt das daran, dass die inverter als „available“ über die API ausgegeben werden und dann eben versucht wird zu regeln. Aber das ist ins Blaue geraten. Ich kann dir da auch nicht weiterhelfen, bzw. ich wüsste nicht wie.

crisi-solar, 2024-06-20:
Einen neuen Raspy zulegen und alles neu installieren wird mir ja vermutlich nichts bringen.

Vielleicht werde ich demnächst einen Versuch starten nur einen WR mit ZeroExport zu verwenden da es ja am Anfang Jänner 2024 funktioniert hat. Da muss ich ja nur in der Override den Invertercount auf 1 und beim zweiten Inverter auf false setzen.
Habe gerade in meiner Override.ini gesehen dass alle 16 Inverter auf Enabled=true stehen. So ok?

Ein anderer Veresuch wäre die DTU und ZeroExport auf die jeweilige Version von Stand Jännere 2024 downzugraden.

Ich könnte die DTU auch auf einen SmartPlug-Stecker verwenden und über HA z.B. um 4 Uhr aus- und einschalten und so einen Neustart auslösen.

Werde mich mal auch in AHOY einlesen, fürchte mich jetzt schon davor HA damit neu einzurichten.

crisi-solar, 2024-06-21:
DTU hat sich mit SMS in der Früh problemlos eingeschaltet, Balken sind bei beiden WR blau.

reserve85, 2024-06-21:
> Habe gerade in meiner Override.ini gesehen dass alle 16 Inverter auf Enabled=true stehen. So ok?
ich würde die einfach rauslöschen, lasse da doch nur die Werte drin die du geändert hast. Ist sonst total unübersichtlich.

Jedenfalls kann ich dir da echt nicht helfen, ich setzte nur Limits wenn die Inverter von der DTU als "erreichbar" gemeldet werden. Wenn die DTU nachts "erreichbar" meldet und dann daraufhin abstürzt bzw. die Kommunikation einstellt läuft an dieser Stelle was falsch.

crisi-solar, 2024-06-22:
Mit Ahoy 0.8.84 funktioniert es einwandfrei!
Die WR haben sich gestern Abend abgeschaltet und heute in der Früh wieder eingeschaltet.

Nochmals vielen Dank an @reserve85 für die großartige Unterstützung!

reserve85, 2024-06-22:
Top,
Dann bitte das Thema bei openDTU eintragen, da kommen auch mehr Leute mit ähnlichen Problemen: https://github.com/tbnobody/OpenDTU/issues/2082

stefan123t, 2024-09-18:
@lumapu als Ahoy und  @tbnobody als OpenDTU main developer sind prädestiniert das zu wissen, die Details und Unterschiede kenne ich leider nicht.
Meines Wissens hat Lukas das im letzten Jahr bei Ahoy auch etwas (bzw. sogar stark?) angepaßt. 

Vielleicht können die beiden ja hier Ihre Implementierung und das Zusammenspiel der REST API und das Senden des ActivePowerLimit einmal kurz skizzieren ?

reserve85, 2026-11-11:
Die Ursache lag bei OpenDTU (der Wechselrichter wurde weiter als erreichbar gemeldet), beim Melder läuft es mit Ahoy. Im Skript selbst gibt es dazu nichts zu ändern, deshalb schließe ich hier. Das Thema gehört zu https://github.com/tbnobody/OpenDTU/issues/2082.

closed 2026-11-11 by reserve85