reserve85/HoymilesZeroExport #214: [Question] - How to setup up the script in a docker container?

Falerion-git, 2024-06-22:
I ran the script previously fine together with iobroker on a rasp pi 4.
Now I wanted to try out the homeassistant OS and running the script in a docker container.

I can't get it to create the container and run the script with the provided docker run command. 

I have all files from the latest release in one folder, created a HymilesZeroExport_Config_Override.ini file. (from my previously working instllation)
I also have the Dockerfile and the compose.yaml in that directory.
Then running the provided command

`docker run -d --name hoymileszeroexport \
    -v ${PWD}/HoymilesZeroExport_Config_Override.ini:/app/HoymilesZeroExport_Config_Override.ini \
    ghcr.io/reserve85/hoymileszeroexport:main -c ./HoymilesZeroExport_Config_Override.ini`

I then get the error message 

`docker: Error response from daemon: error while creating mount source path '/root/HoymilesZeroExport/HoymilesZeroExport_Config_Override.ini': mkdir /root/HoymilesZeroExport: read-only file system.`

I guess I'm missing something here?
Could someone provide a short step by step description how to setup such a container? Or point out which step I'm missing?
Thanks in advance!

-Falerion

tomquist, 2024-06-22:
Looks like you’re mounting the config file from a read only file system. You could try mounting as read-only:
```
docker run -d --name hoymileszeroexport \ -v source=${PWD}/HoymilesZeroExport_Config_Override.ini,destination=/app/HoymilesZeroExport_Config_Override.ini,readonly \ ghcr.io/reserve85/hoymileszeroexport:main -c ./HoymilesZeroExport_Config_Override.ini
```

reserve85, 2026-11-11:
This question has been answered above, so I'm closing the issue. Feel free to open a new one if something is still unclear.

closed 2026-11-11 by reserve85