reserve85/HoymilesZeroExport #219: Feature Request "Waschmaschinen-Modus"

Ollipop030, 2024-06-23:
Ich nenne das einfach mal Waschmaschinen Modus. Die Idee kam mir heute als die Waschmaschine mal wieder lief.

Könnte man nicht bei jedem Durchlauf den Schaltzustand eines Shellys abfragen? Hintergrund ist Folgender:
Immer wenn die Waschmaschine läuft wird rauf und runtergeregelt (Motor an/Motor aus). Wenn ich mir aber den Stromverbrauch nach dem Waschen ansehe sehe ich immer wieder Netzbezug, anschließend eine sehr hohe Einspeisung. Dann wieder Netzbezug und so weiter. Gegen die kurzfristige zu hohe Einspeisung habe ich nichts, aber den Netzbezug könnte man noch bremsen. Jetzt meine Idee:
Die Waschmaschine hängt an einem Shelly Plug. Das Script fragt pro Durchlauf einmal den in der *.ini definierten Shelly ab, ob er "on" ist. Wenn Shelly on, dann gilt nicht POWERMETER_TARGET_POINT, sondern POWERMETER_TARGET_POINT_SHELLY_ACTIVE. Dieser Wert ist dann aber bei 500 Watt (je nach dem wie viel die Waschmaschine so braucht). Also wird einfach einfach eine höhere Einspeisung zugelassen, ohne ständig zu regeln. Im Shelly selbst kann man dann noch definieren, wenn Verbrauch <10 Watt für 5 Minuten, dann "off". Das bekommt dann auch das Script mit, und alles ist wieder wie gewohnt. Das würde dann natürlich auch mit anderen Geräten funktionieren, die solche Schwankungen im Verbrauch haben, beim 3D Drucker habe ich das auch bemerkt.

Was haltet ihr davon? Wäre das sinnvoll? Integrieren kann ich das leider nicht, spreche leider kein Python. Hier müsste Tobias aushelfen.

reserve85, 2024-06-24:
MMn. fällt die Anwendung eher in die Fernsteuerung via MQTT (Home Assistant etc). Als Integration würde ich das nicht aufnehmen.

Ollipop030, 2024-06-24:
Schade, dachte das könnte man auf dem kurzem Dienstweg einbauen. MQTT habe ich nicht, werde ich aber auch nicht extra einrichten dafür.

reserve85, 2024-08-19:
weil ich hier grad dein Name in der E-Mail-Benachrichtigung gesehen hab: @BlackcatSandy haben wir mal an der h_da zusammen studiert (M.Sc)?

reserve85, 2026-11-11:
Der Zielwert lässt sich inzwischen zur Laufzeit per MQTT ändern (powermeter_target_point, siehe Abschnitt MQTT_CONFIG in der Config). Damit kann eine Automation, z. B. in Home Assistant oder ioBroker, den Zielwert umstellen, solange die Waschmaschine läuft. Eine direkte Shelly-Abfrage im Skript ist nicht geplant. Kurze Lastspitzen sind ein eigenes Thema und werden in #225 behandelt. Ich schließe das Issue.

closed 2026-11-11 by reserve85