reserve85/HoymilesZeroExport #220: Trucki2meanwellgateway T2MG direkt im Script einbinden?

tclv2000, 2024-06-24:
Ich lade einen Powerqueen Akku mit einem Menawell Ladegerät. Dieses Ladegerät wird über den T2MG Stick und den Shelly 3 EM geregelt.  Außerdem läuft Dein Script zur Nulleinspeisung i.  V. m. mit ein Paar Regelungen über Openhab via MQTT.

Ist eine direkte Anbindung des T2MG an das Script möglich?

Solange die maximale Ladeleistung des Meanwell-Ladegerätes nicht erreicht ist, wird dann nicht runtergeregelt? Quasi als fiktives Plus auf den Targetpoint bis das Ladegerät die maximale Leistung hat?

Regele das aktuell zwar lauffähig über MQTT/Openhab, jedoch ist der zeitliche Versatz entsprechen lang.

reserve85, 2024-06-26:
Hi,

out of the Box würde ich sagen dass der Anwendungsfall eher etwas für HomeAssistant o.Ä ist. 
Dort kannst du den Targetpoint des Scripts anpassen, solange dein Ladegerät noch lädt. 
Du machst das schon via Openhab? Was meinst du mit zeitlicher Versatz?

tclv2000, 2024-06-26:
Mit dem zeitlichen Versatz meinte ich die Verabeitungszeit bis sich meine 4 HM600 wieder auf den Targetpoint eingependelt haben. Außerdem standen sich die Regelung des Ladegrätes und die Regelung der Inverter immer irgendwie im Weg.

Mein Gedanke war, dass die Regelung des Scriptes erst beginnt, wenn das Ladegerät seine maximale Leistung liefert.

Die Lösung war nicht die Anpassung des Targetpoints über MQTT, sondern die Anpassung der minimalen Inverterleistung. Solange das Ladegerät seine maximale Leistung nicht erreicht, setze ich die minimale Inverterleistung auf 100% und das T2MG übernimmt die Regelung auf den dort gesetzten Targetpoint. Bei erreichen der maximalen Leistung des Ladegerätes wird der Wert wieder auf 5% gesetzt und die Regelung auf den Targetpoint des Scripts setzt ein.

closed 2024-06-26 by tclv2000