reserve85/HoymilesZeroExport #222: Frage zu Victron Multiplus II + HoymilesZeroExport

crisi-solar, 2024-07-12:
Möchte meine Anlage mit einem ca. 7 bis 10kW-Speicher AC-seitig aufrüsten und denke da an Victron Multiplus II.
Wer hat diesen Victron mit mehreren Hoymiles WR und HoymilesZeroExport am laufen und kann darüber berichten wie gut dieses Zusammenspiel funktioniert?
Welcher Akku wäre von Vorteil für HoymilesZeroExport? Eventuell 2x Pylyontech US5000?
Toll wäre ev. ein Link zu einem Shop der das alles in einem Komplettpaket verkauft und Ahnung von Nulleinspeisung  mit HoymilesZeroExport hat.

Danke mal schon vorab für eure Hilfe!

crisi-solar, 2024-07-13:
Hat hier keiner den Victron Multiplus II mit HoymilesZeroExport am laufen da keine Antworten bzw. Tipps kommen?

crisi-solar, 2024-07-16:
Da mir hier keiner eine Infos wegen dem Victron Multiplus geben kann, welche Kombination gibt es um 2x Pytes V5 in HoymilesZeroExport zu integrieren? Der Output vom WR soll über 3kWh liegen.
Welche Akkukombination habt ihr mit HoymilesZeroExport installiert?

smaicloud, 2024-07-16:
Hi @crisi-solar,

ich rüste aktuell auch auf einen MP2 5000 und 3 Pylontech 5000 Akkus um. Meine 3 Hoymiles WR werde ich behalten, hier habe ich gute AC Leistung bei geringem Verlust.

Es wird keinen Shop geben der dich bei Bastellösungen untersützt. Jeder muss Garantie geben und das kann kein Shop bzw. Firma. Da musst du was fertiges kaufen. Oder eben selbst fähig sein so etwas zu bauen.

Das Script hier ist die falsche Lösung dafür. Es gibt aber dazu etwas: https://github.com/henne49/dbus-opendtu
Genau das schafft die Einbindung von externen WR welche nicht Victron zertifiziert sind (Fronius out of the box).

crisi-solar, 2024-07-16:
@smaicloud Danke für die Info!

Dachte nicht dass das Ganze so kompliziert wird da ja im Script hier bei den Invertern auch die Möglichkeit gibt eine Batterie zu konfigurieren. 
So sollte es doch möglich sein von einer Batterie auf alle 4 Anschlüsse von einem HMS-2000-4T zu gehen und so 2000W zu laden und einzuspeisen. Für was sollten diese Batterieeinstellmöglichkeiten sonst da sein?

Bin eigentlich mit der Softwar HoymilesZeroExport vollstens zufrieden und möchte eigentlich gar nicht wechseln und wieder von vorne alles neu einrichten.

Angeblich sollen die Pytes V5 besser sein als die Pylontech 5000, gibt es einen Grund warum du dich trotzdem für diese entschieden hast?
Werde mir mal deinen Link anschauen, vielen Dank dafür! 
Hast du vielleicht auch einen Link zu einem youtube-Video wo eine fertige Konfiguration wie wir sie brauchen zu sehen ist?

Vielleicht können wir uns privat weiter austauschen da das dann nichts mehr mit HoymilesZeroExport zu tun hat.

smaicloud, 2024-07-16:
@crisi-solar Passt weniger hier rein, besser privat. Per mail zum Handynummeraustausch?!

crisi-solar, 2024-07-16:
Gerne, schreib mir auf hoymiles@gmx.at

crisi-solar, 2024-07-17:
> @crisi-solar Passt weniger hier rein, besser privat. Per mail zum Handynummeraustausch?!

> @smaicloud Gerne, schreib mir auf [hoymiles@gmx.at](mailto:hoymiles@gmx.at)

Leider noch keine Email erhalten.

bootix, 2024-08-30:
> Hat hier keiner den Victron Multiplus II mit HoymilesZeroExport am laufen da keine Antworten bzw. Tipps kommen?

Ich habe HoymilesZeroExport seit über 1 Jahr mit Victron Multiplus-II im Einsatz. Anfangs mit einem EM3 als Energiezähler. Seit dem Erscheinen des "shell script based interface" konnte ich den EM2 ausbauen und lese direkt aus dem Multiplus aus.

Der Aufbau sieht so aus: ET112 als Energiezähler am Multiplus-II 48/5000 GX. Zwei Pylontech US5000 am MP-II. Parallel am AC-IN hängen drei HM-1500 mit OpenDTU. 

Dabei steht "POWERMETER_TARGET_POINT = -250" und "POWERMETER_TOLERANCE =  100".

Läuft alles einwandfrei!

crisi-solar, 2024-08-30:
@bootix Danke für deine Info und für die Konfigdatei!

Ich habe seit einigen Tagen meine Victronanlage laufen.
1x MP 2 48/5000 mit Cerbo GX
3x Pytes V5a
1x EM540

Meine zwei HMS-2000 sind am AC Out1, der HMS-1600 am AC In mit Ahoy die ich auch in der Victron Geräteliste inkludiert habe.

Dank HomeAssistant geht das ausprobieren ganz easy und schnell.
[image]

LG

crisi-solar, 2024-08-31:
@bootix Hast du bei deinen Hoymiles auch die Daten für das Frequenzshift für einen Inselbetrieb geändert weil die Hoymiles zu hart reduzieren für den Multiplus 2.

bootix, 2024-09-01:
Nein, da die Hoymiles vor dem MP-II am AC-IN hängen und somit kein Inselbetrieb besteht. Der Frequenzshift ist, wie du auch schreibst, nur im Inselbetrieb möglich. AC-IN wäre also vom Netz getrennt und die Hoymiles an AC-OUT.

In dem Fall würde man das HoymilesZeroExport Script auch nicht benötigen.

crisi-solar, 2024-09-04:
Hat wer schon probiert HoymilesZeroExport in den Cerbo GX zu integrieren?

crisi-solar, 2024-09-11:
@bootix Ich habe deine Einstellungen wie oben beschrieben "POWERMETER_TARGET_POINT = -250" und "POWERMETER_TOLERANCE = 100" ausprobiert, damit speist er bei mir zuviel ins Netz.

Jetzt suche ich eine Möglichkeit HoymilesZeroExport bis zu einer Batterieladung von ca. 80-85% auszuschalten und erst dann wieder einzuschalten. 
Derzeit mache ich das händisch indem ich über die Konsole den Befehl "systemctl stop HoymilesZeroExport.service" setze und bei ca. 80% dann das Script mit "sudo ./restart.sh" wieder starte.
Leider gibt es keinen MQTT-Befehl das Script so aus- und einzuschalten.

reserve85, 2026-11-11:
Hier gab es seit 2024 keine Aktivität mehr, und es ist kein Fehler im Skript offen. Ich räume die Issue-Liste auf und schließe das Thema. Erfahrungen zu Einstellungen sammeln wir weiter in #28.

closed 2026-11-11 by reserve85