reserve85/HoymilesZeroExport #223: Zero export HM.

rradkov, 2024-07-31:
Hello everyone. 
If the script is running and you decided to manually reduce the power output from the DTU, the script is saying that the invertor was already accepted the limit at XXXX Watt

[image]

reserve85, 2024-08-18:
@rradkov Can you test the dev if it is working with openDTU? I can confirm AHOY is working.

rradkov, 2024-08-18:
Hey @reserve85 
I've just tested it and it works like a charm ! Thanks !
If I set manually the limit from the DTU, the script automatically override it as it is supposed to do it ! 

Quick question.
I have problem with DTU connection to the Invertor early in the morning and/or after power outage.

So I was wondering about set Persistent limit and Non-Persistent limit. As far I know Persistent limit it saved in the memory of the invertor and if the inverter is rebooted or in event of power outage it will be limited to the Persistent applied value. Lets say 50%. Is it possible the script to automatically set the Non-persistent limit only and in that case when there is power outages or DTU lost connection with Inverter or scripts doesn't apply the settings to the inverter this will be some sort of a fail-safe zero export limitation and it will not return energy to the grid. 
Is there a way that you can implement this settings in the configuration:
If there is no connection to the invertor, DTU or whatever this will be the safe limit - Persistent
IF everything is working as it is supposed, then the script will set only non-persistent limit?

reserve85, 2024-08-18:
well, the script only sets always non-persistent limits. I know there are a couple of users with connection problems on openDTU, see here: https://github.com/reserve85/HoymilesZeroExport/issues/211 (translator helps)

perhaps you can give a try to AHOY, afaik there are no connection problems....

closed 2024-08-18 by reserve85