reserve85/HoymilesZeroExport #224: Frage zu Inverter ausschalten/standby

tester277, 2024-08-02:
Hallo zusammen,
Danke für das gut strukturierte Script mit vielen Möglichkeiten. Gibt es auch eine Möglichkeit einen Inverter (zuständig für die Einspeisung in der Nacht) in den Standby bzw. Aus Zustand zu versetzen und auch wieder einzuschalten?
Ich würde das gern über das Script steuern.

RealNBB, 2024-08-14:
Oh ja…dem Vorschlag schließe ich mich an, das wäre wirklich klasse, vor allen für die User, die eine Batterie ihr Eigen nennen!

Weiß nicht,,, die Option könnte man ja iwie so gestalten, dass nur die Inverter neugestartet werden, die „true“ bei „HOY_BATTERY_MODE“ stehen haben.

Dann – nach erfolgreichem time sync – 2 Minuten vor Mitternacht die Inverter neu starten lassen (-> „HOY_BATTERY_MODE_RESTART_INVERTER_BEFORE_MIDNIGHT = true/false“.

Ganz cool wäre es dann, wenn man abhängig von die Inverter nicht gelichzeitig, sondern nacheinander starten lässt.

Z.B. bei drei Inverter im Battery Mode mit folgenden IDs: #1, #4, #5

>  Gesamtanzahl Inverter in Battery Mode = 3 (müsste man intern summieren lassen)

>  Restart #1: 24Uhr – 1 Minute (feste Größe) – SUMME(Gesamtanzahl Inverter in Battery Mode+1) = 23:57Uhr Neustart

> Restart #4: 24Uhr – 1 Minute (feste Größe) – SUMME(Gesamtanzahl Inverter in Battery Mode+2) = 23:58Uhr Neustart

> Restart #5: 24Uhr – 1 Minute (feste Größe) – SUMME(Gesamtanzahl Inverter in Battery Mode+3) = 23:59Uhr Neustart

 

Das hätte zur Folge, dass Yield Day nicht zu 100% stimmt, aber bei mehr als einem Inverter im Battery Mode die Versorgung über die Batterie nie abbricht, da nicht alle Inverter gleichzeitig gestartet werden.

Soll aber nur ein Vorschlag sein! 😊

Generell ein ganz ganz fetten Dank für dieses tolle Skript. Es läuft es wunderbar und macht genau das, was es soll! 😊

Joppo67, 2024-09-09:
Hallo zusammen,
hallo Tobias,

zunächst mal vielen Dank für das Script und der damit verbunden Arbeit die Du Dir gemacht hast! 👍

Ich bin neu hier und habe das Script nun seit Samstag am laufen was auch prima klappt. 

Gesteuert wird bei mir ein HM-600, der an einer 48V Batterie hängt, via Ahoy-DTU.
Die Batterien werden tagsüber von einem Victron Smartsolar, an dem 2 PV Platten hängen, geladen.

Da ich noch eine "große" PV-Anlage habe, brauche ich tagsüber keinen Strom aus den Batterien. 
Ich möchte nur nachts meinen Verbauch reduzieren. 
Darum fände es auch ein super Feature, wenn man den Wechselrichter tagsüber in den Standby versetzen könnte, so dass er nichts mehr aus den Batterien zieht. 

Vorstellen könnte ich mir, dass man die Werte ts_sunrise und ts_sunset aus der Ahoy-DTU für das Ab- bzw wieder Einschalten nutzen könnte?

tohmc, 2024-11-27:
Eine Möglichkeit die inverter zu stoppen oder in standby zu versetzten über mqtt wäre stark.
So könnte man sich im Smart Home auch automatisierungen die das ein und ausschalten bedienen nutzen.

reserve85, 2026-11-11:
Das ist jetzt auf dem dev-Branch umgesetzt (V 1.107): Ein Wechselrichter lässt sich zur Laufzeit per MQTT aus- und wieder einschalten.

- `zeropower/set/inverter/<n>/enabled` mit `false` schaltet den Wechselrichter aus und nimmt ihn aus der Regelung, `true` schaltet ihn wieder ein (`<n>` beginnt bei 0; `on`/`off` und `1`/`0` gehen auch).
- `zeropower/reset/inverter/<n>/enabled` stellt den Wert aus der Config-Datei wieder her.
- Der aktuelle Zustand steht in `zeropower/state/inverter/<n>/enabled`.

Voraussetzung ist der Abschnitt [MQTT_CONFIG] in der Config. Test-Image: ghcr.io/reserve85/hoymileszeroexport:dev

Wer es ausprobiert: Rückmeldung wäre super, bevor ich es nach main übernehme.