reserve85/HoymilesZeroExport #226: Hilfe mit HOY_BATTERY_MODE

sibelle-labs, 2024-08-20:
Hallo @ Freunde der Sonne :-)
Situation: 
2x PV Module je 400w
1x Battery Marstek B2500-D (2.240wh) 
1x Hoymiles HM-600
Grundverbrauch (50Wh min.)

HoymilesZeroExport mit AHOY(Hoymiles), Hichi mit Tasmota SML & Home Assistant.
und hier meine Anfänge:

[INVERTER_1]
HOY_INVERTER_WATT = 600
HOY_MAX_WATT = 400

[CONTROL]
POWERMETER_TARGET_POINT = 5
POWERMETER_TOLERANCE = 10
POWERMETER_MAX_POINT = 100

Hat jemand schon eine funktionierende Configuration für solche Konstellationen ?
Vielen Dank im Voraus,

RealNBB, 2024-08-21:
Würde mich schwer wundern, wenn du jemand mit GENAU dieser Config hier findet.

Aber anders herum gefragt: Was genau ist dein problem? Bzw. hast du überhaupt ein Problem? :D

sibelle-labs, 2024-08-21:
Hallöchen !
> Würde mich schwer wundern, wenn du jemand mit GENAU dieser Config hier findest.
Das ist Gold richtig :-)
> Aber anders herum gefragt: Was genau ist dein problem? Bzw. hast du überhaupt ein Problem? :D
Hmm... Ich habe es seit 2 Tage am Laufen und schaut gut aus, und auf guten Glück da ich ein paar Parametern nicht verstehe.
POWERMETER_TARGET_POINT
POWERMETER_MAX_POINT
POWERMETER_MIN_POINT

Statt Paneele habe ich die Batterie am WR angeschlossen, deshalb war meine Frage auch wegen HOY_BATTERY_MODE.
Aktuell habe ich  HOY_BATTERY_MODE = false.
Was bringt HOY_BATTERY_MODE = true ? und was müsste ich hiermit beachten ?

reserve85, 2024-08-21:
Battery Mode ist eher für „ungeregelte“ Akkus direkt am inverter, da kannst du z.B. definieren bei welcher Spannung der Inverter ausgeschaltet bzw. wieder eingeschaltet wird oder seine Leistung reduziert.

closed 2024-09-12 by sibelle-labs