reserve85/HoymilesZeroExport #228: Drosselung bzw. Erhöhung greift zu stark

RealNBB, 2024-08-22:
Hallo,

mir ist heute (die Sonne scheint gerade so schön hier bei uns im Süden :) ) aufgefallen, dass die Regelung zu stark greift - sowohl beim reduzieren als auch beim erhöhen. Ggf. aber  auch nur ein Config-Fehler meinerseits, kann natürlich auch gut sein! 🗡️ 

Kurz zum Setup (sieht man ja aber auch im Prinzip im Log-File):

- 1x HM-600 (nicht im Battery-Mode)
 - 1x HM-800 (im Battery-Mode)
-- 1x HM-1500 (im Battery-Mode)

Ich frag mich halt, weshalb das Skript nur den Endwert von 30W kennt, wo doch ~70W näher am Targetpoint wären?!

Anbei wie gesagt das Log und meine Config
[Log.txt](https://github.com/user-attachments/files/16714929/Log.txt)
[Config.txt](https://github.com/user-attachments/files/16714932/Config.txt)

reserve85, 2024-08-23:
Vermutlich ein Fehler bei der Berechnung im Mixed Mode… kannst du mal testweise die Priorität von den invertern ändern: inv 1 > prio1, inv 2 -> Prio3, inv3 > prio2 ??

RealNBB, 2024-08-24:
wo finde ich denn diese neue dev? :)

reserve85, 2024-08-24:
Hier:

https://github.com/reserve85/HoymilesZeroExport/tree/dev

oder automatisch installieren via:
`sudo ./update.sh dev`

RealNBB, 2024-08-24:
Ich hab [HoymilesZeroExport.py](https://github.com/reserve85/HoymilesZeroExport/blob/dev/HoymilesZeroExport.py)  ausgetauscht, also bestehende Datei in 1.98 gelöscht und diese dann reinkopiert. ich schätze mal, dass das passt.

Fehler trat aber leider wieder auf. So bei Zeile 118.

Irgendwelche Leistungssprünge (nicht im Haushalt und Sonne scheint super konstant gerade ohne irgendwelche Wolken) gab es keine.
[neu 5.txt](https://github.com/user-attachments/files/16737010/neu.5.txt)

reserve85, 2024-08-24:
Ich schau es nochmal an, kann aber montag werden.

reserve85, 2024-08-26:
Hey @RealNBB:

gerade gesehen: 
`2024-08-24 14:47:15 INFO     Log write to file: False
2024-08-24 14:47:15 INFO     Python Version: 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0]
2024-08-24 14:47:15 INFO     Author: Tobias Kraft / Script Version: 1.97`

vermutlich hast du doch noch die falsche Version laufen?

RealNBB, 2024-08-26:
Okay...hätte nicht nur den Container, sondern auch das Image in Portainer löschen müssen.
Die Dev-Version bekomme ich trotzdem nicht installiert.
Egal, was ich mache, er zeigt mir stets ?Author: Tobias Kraft / Script Version: 1.98? an, aber die 1.99 bekomme ich nicht hin. Da fehlen mir offensichtlich einfach die Kenntnisse.

Ich komm hier wohl nur weiter, falls du mir das gewohnte zip-File zur Verfügung stellen könntest.

reserve85, 2024-08-26:
https://github.com/reserve85/HoymilesZeroExport/archive/refs/heads/dev.zip

so?

RealNBB, 2024-08-26:
Tut mir leid, ich muss leider passen. Bekommes einfach nicht installiert. 

Was ich mache: Alle alten Container und Images, die mit dem Skript zu tun hatten, gelöscht, danndie Zip von einem Beitrag darüber entpackt, meine override.config reinkopiert und das Terminal in diesem Ordner gestartet. Eingabe dort dann:

"docker run -d --restart=unless-stopped --name hoymileszeroexport \
    -v ${PWD}/HoymilesZeroExport_Config_Override.ini:/app/HoymilesZeroExport_Config_Override.ini \
    ghcr.io/reserve85/hoymileszeroexport:main -c ./HoymilesZeroExport_Config_Override.ini"

Ich habe es auch mit

"docker run -d --restart=unless-stopped --name hoymileszeroexport \
    -v ${PWD}/HoymilesZeroExport_Config_Override.ini:/app/HoymilesZeroExport_Config_Override.ini \
    ghcr.io/reserve85/hoymileszeroexport:**dev** -c ./HoymilesZeroExport_Config_Override.ini"

versucht, brachte aber auch keinen Erfolg. Immerhin bin ich nun aber auf Version1.98, wo ich eigentlich dachte, dass das schon lange läuft. War wohl an dem Image gelegen in Portainer....schätze ich mal. Kenne mich aber auch Null aus.

tomquist, 2024-08-26:
Mit der ZIP fängst du wenig an wenn du das ganze über Docker laufen lässt, es sei denn du mountest die neue Version des scripts in den Container. Der Grund warum du eine alte Version über den Befehl bekommst ist, dass der `dev` branch nicht automatisch ein neues Docker Image baut wenn darauf gepushed wird. @reserve85 Habe aktuell keinen Zugriff auf einen PC aber das sollte sich durch Anpassen der GitHub Action beheben lassen.

Ansonsten wie gesagt die neue Version in den Container mounten, als Workaround:

```
docker run -d --restart=unless-stopped --name hoymileszeroexport
-v ${PWD}/HoymilesZeroExport_Config_Override.ini:/app/HoymilesZeroExport_Config_Override.ini -v ${PWD}/HoymilesZeroExport.py:/app/HoymilesZeroExport.py
ghcr.io/reserve85/hoymileszeroexport:dev -c ./HoymilesZeroExport_Config_Override.ini
```
(Ungetestet)

RealNBB, 2024-08-26:
Jetzt kommt:

"docker run" requires at least 1 argument.
See 'docker run --help'.

Usage:  docker run [OPTIONS] IMAGE [COMMAND] [ARG...]

Create and run a new container from an image
bash: -v: Kommando nicht gefunden.
bash: ghcr.io/reserve85/hoymileszeroexport:dev: Datei oder Verzeichnis nicht gefunden

tomquist, 2024-08-26:
Ah ja sorry, das meinte ich mit ungetestet und dass ich es gerade keinen Zugriff auf einen PC habe 😅

Eventuell kann jemand anderes oder auch ChatGPT weiterhelfen.

reserve85, 2024-08-26:
Ich habe den workflow gerade erweitert, müsste jetzt gehen?
Bin da leider auch die totale Vollpfeife, habe selbst noch nie was mit Docker gemacht... @tomquist vielleicht kannst du ja mal drüberschauen ob die zwei Zeilen die ganze Hexerei waren? (dev erweitert)

`$ docker pull ghcr.io/reserve85/hoymileszeroexport:dev`

https://github.com/reserve85/HoymilesZeroExport/pkgs/container/hoymileszeroexport/263666775?tag=dev

tomquist, 2024-08-26:
Das sieht gut aus, danke @reserve85 !

@RealNBB du kannst mit dem command das neuste `dev` Image nutzen:

`docker run -d --pull always --restart=unless-stopped --name hoymileszeroexport \
-v ${PWD}/HoymilesZeroExport_Config_Override.ini:/app/HoymilesZeroExport_Config_Override.ini \
ghcr.io/reserve85/hoymileszeroexport:main -c ./HoymilesZeroExport_Config_Override.ini`

RealNBB, 2024-08-27:
Guten Morgen,
 
damit habe ich es nun auch hinbekommen, dass 1.99 bei mir läuft. Danke dafür! :)

Bisher sieht alles gut aus. Aber das Ganze muss ich schon einen längeren Zeitraum beobachten, da es spezielle Konstellationen gab (gefühlt zumindest), wo der Fehler mit der falschen Berechnung vermehrt aufgetretenen ist (v.a. bei wenig Grundlast, aber gleichzeitiger hoher Sonneneinstrahlung).
Also gebt mir mal zwei, drei Wochen, da ich ja auch nicht ständig auf das Skript schaue(n kann).

Anbei nochmal meine Config mit der das Skript nun in Version 1.99 läuft. Ahoy ist 1.29, da alle Versionen darüber bei mir Probleme machen.

`[VERSION]
VERSION = 1.97

[SELECT_DTU]
USE_AHOY = true

[SELECT_POWERMETER]
USE_SHELLY_3EM_PRO = true

[AHOY_DTU]
AHOY_IP = 192.168.0.71
AHOY_PASS = Michael155

[SHELLY]
SHELLY_IP = 192.168.0.70
SHELLY_USER = admin
SHELLY_PASS = Michael155
EMETER_INDEX = 

[COMMON]
INVERTER_COUNT = 3
SLOW_APPROX_LIMIT_IN_PERCENT = 20
SLOW_APPROX_FACTOR_IN_PERCENT = 25
LOOP_INTERVAL_IN_SECONDS = 10
SET_LIMIT_TIMEOUT_SECONDS = 5
POLL_INTERVAL_IN_SECONDS = 1
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 10
ON_GRID_FEED_FAST_LIMIT_DECREASE = true
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 100
ENABLE_LOG_TO_FILE = false
LOG_BACKUP_COUNT = 30
SET_POWERSTATUS_CNT = 5
LOG_TEMPERATURE = false
SET_POWER_STATUS_DELAY_IN_SECONDS = 10
SET_INVERTER_TO_MIN_ON_POWERMETER_ERROR = false
MAX_RETRIES = 3
RETRY_STATUS_CODES = 500,502,503,504
RETRY_BACKOFF_FACTOR = 0.1

[CONTROL]
POWERMETER_TARGET_POINT = -3
POWERMETER_TOLERANCE = 17
POWERMETER_MAX_POINT = 50
POWERMETER_MIN_POINT = -600

[INVERTER_1]
SERIAL_NUMBER = 114184209121
ENABLED = true
HOY_INVERTER_WATT = 600
HOY_MAX_WATT = 600
HOY_MIN_WATT_IN_PERCENT = 5
HOY_COMPENSATE_WATT_FACTOR = 1
HOY_BATTERY_MODE = false
HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V = 48.1
HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V = 49
HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 49.5
HOY_BATTERY_NORMAL_WATT = 600
HOY_BATTERY_REDUCE_WATT = 600
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 50.5
HOY_BATTERY_IGNORE_PANELS = 
HOY_BATTERY_PRIORITY = 1
HOY_BATTERY_AVERAGE_CNT = 1

[INVERTER_2]
SERIAL_NUMBER = 114190934615
ENABLED = true
HOY_INVERTER_WATT = 800
HOY_MAX_WATT = 800
HOY_MIN_WATT_IN_PERCENT = 4
HOY_COMPENSATE_WATT_FACTOR = 1
HOY_BATTERY_MODE = true
HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V = 49
HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V = 50
HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 50.5
HOY_BATTERY_NORMAL_WATT = 800
HOY_BATTERY_REDUCE_WATT = 200
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 52
HOY_BATTERY_IGNORE_PANELS = 
HOY_BATTERY_PRIORITY = 2
HOY_BATTERY_AVERAGE_CNT = 1

[INVERTER_3]
SERIAL_NUMBER = 116190742053
ENABLED = true
HOY_INVERTER_WATT = 1500
HOY_MAX_WATT = 1500
HOY_MIN_WATT_IN_PERCENT = 3
HOY_COMPENSATE_WATT_FACTOR = 1
HOY_BATTERY_MODE = true
HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V = 48
HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V = 49
HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 50
HOY_BATTERY_NORMAL_WATT = 1500
HOY_BATTERY_REDUCE_WATT = 375
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 51
HOY_BATTERY_IGNORE_PANELS = 
HOY_BATTERY_PRIORITY = 1
HOY_BATTERY_AVERAGE_CNT = 1
`

reserve85, 2024-08-27:
SLOW_APPROX_LIMIT_IN_PERCENT gibt an, ab welchem Limit-Sprung (bsp. von 1000W auf 600W -> wären 400W Differenz -> also 40% -> da würde das dann greifen) das Ergebnis um 25% angenähert wird. 
Es wird also nicht direkt auf den neu ausgerechneten Wert gesprungen (600W) sondern erstmal auf 700W und im nächsten Schritt dann weitergeregelt. 

Hängt damit zusammen, wenn du ein Panel teilweise verschattet hast, dann passt die Berechnung nicht zu 100%.

Wegen dem MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER: das kann gut sein, dass es bei mixed-Mode nicht korrekt funktioniert. Würde ich einfach ausschalten. Hintergrund ist, dass ja erstmal der Solarinverter komplett auf 100% laufen muss, bis die Akkus einspringen. MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER limitiert das dann und die Akkus kommen gar nicht mehr dran.

Du könntest dennoch das Log heute abend mal anhängen, ich würde mal drüberschauen.

Gruß Tobi

RealNBB, 2024-08-27:
```
2024-08-27 14:40:20 INFO     Log write to file: False
2024-08-27 14:40:20 INFO     Python Version: 3.11.2 (main, May  2 2024, 11:59:08) [GCC 12.2.0]
2024-08-27 14:40:20 INFO     Author: Tobias Kraft / Script Version: 1.99
2024-08-27 14:40:20 INFO     read config file: /app/HoymilesZeroExport_Config.ini
2024-08-27 14:40:20 INFO     read additional config file: ./HoymilesZeroExport_Config_Override.ini
2024-08-27 14:40:20 INFO     Config file V 1.97
2024-08-27 14:40:20 INFO     ---Init---
2024-08-27 14:40:20 INFO     Ahoy: Current Version: 0.8.140
2024-08-27 14:40:20 INFO     Ahoy: Inverter "yet unknown" Available: True
2024-08-27 14:40:20 INFO     Ahoy: Inverter "HM-600" / serial number "xxx" / temperature 38.3 degC
2024-08-27 14:40:20 INFO     Ahoy: Inverter "yet unknown" Available: True
2024-08-27 14:40:20 INFO     Ahoy: Inverter "HM-600" / serial number "xxx" / temperature 38.3 degC
2024-08-27 14:40:20 INFO     Ahoy: Inverter "HM-800" / serial number "xxx" / temperature 34.2 degC
2024-08-27 14:40:20 INFO     Ahoy: Inverter "yet unknown" Available: True
2024-08-27 14:40:20 INFO     Ahoy: Inverter "HM-600" / serial number "xxx" / temperature 38.3 degC
2024-08-27 14:40:20 INFO     Ahoy: Inverter "HM-800" / serial number "xxx" / temperature 34.2 degC
2024-08-27 14:40:21 INFO     Ahoy: Inverter "HM-1500" / serial number "xxx" / temperature 33.6 degC
2024-08-27 14:40:21 INFO     Ahoy: Inverter "HM-600": Turn on
2024-08-27 14:40:21 INFO     Ahoy: Authenticating...
2024-08-27 14:40:21 INFO     Ahoy: Authenticating successful, received Token: EP9NK1FS23FSUGDN
2024-08-27 14:40:21 INFO     Ahoy: Inverter "HM-600": Turn on
2024-08-27 14:40:31 INFO     Ahoy: Inverter "HM-800": Turn on
2024-08-27 14:40:41 INFO     Ahoy: Inverter "HM-1500": Turn on
2024-08-27 14:40:51 INFO     setting new limit to 107 Watt
2024-08-27 14:40:51 INFO     Ahoy: Inverter "HM-600": setting new limit from -1 Watt to 30 Watt
2024-08-27 14:40:51 INFO     Ahoy: Inverter "HM-600": Limit acknowledged
2024-08-27 14:40:51 INFO     Ahoy: Inverter "HM-1500": setting new limit from -1 Watt to 77 Watt
2024-08-27 14:40:52 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
2024-08-27 14:40:52 INFO     Ahoy: Inverter "HM-800": setting new limit from -1 Watt to 32 Watt
2024-08-27 14:40:53 INFO     Ahoy: Inverter "HM-800": Limit acknowledged
2024-08-27 14:40:53 INFO     intermediate meter AhoyDTU: 227 Watt
2024-08-27 14:40:53 INFO     Lowest panel voltage inverter "HM-800": 54.0 Volt
2024-08-27 14:40:53 INFO     Average min-panel voltage, inverter "HM-800": 54.0 Volt
2024-08-27 14:40:53 INFO     Ahoy: Inverter "HM-800": Turn on
2024-08-27 14:41:03 INFO     Lowest panel voltage inverter "HM-1500": 53.9 Volt
2024-08-27 14:41:03 INFO     Average min-panel voltage, inverter "HM-1500": 53.9 Volt
2024-08-27 14:41:03 INFO     Ahoy: Inverter "HM-1500": Turn on
2024-08-27 14:41:13 INFO     powermeter Shelly3EMPro: 74 Watt
2024-08-27 14:41:13 INFO     ---Start Zero Export---
2024-08-27 14:41:13 INFO     Ahoy: Inverter "HM-600" Available: True
2024-08-27 14:41:13 INFO     Ahoy: Inverter "HM-800" Available: True
2024-08-27 14:41:13 INFO     Ahoy: Inverter "HM-1500" Available: True
2024-08-27 14:41:13 INFO     Lowest panel voltage inverter "HM-800": 54.0 Volt
2024-08-27 14:41:13 INFO     Average min-panel voltage, inverter "HM-800": 54.0 Volt
2024-08-27 14:41:13 INFO     Ahoy: Inverter "HM-800": Turn on
2024-08-27 14:41:23 INFO     Lowest panel voltage inverter "HM-1500": 53.9 Volt
2024-08-27 14:41:23 INFO     Average min-panel voltage, inverter "HM-1500": 53.9 Volt
2024-08-27 14:41:23 INFO     Ahoy: Inverter "HM-1500": Turn on
2024-08-27 14:41:34 INFO     powermeter Shelly3EMPro: 75 Watt
2024-08-27 14:41:34 INFO     setting new limit to 290 Watt
2024-08-27 14:41:34 INFO     Ahoy: Inverter "HM-600": setting new limit from 30 Watt to 213 Watt
2024-08-27 14:41:34 INFO     Ahoy: Inverter "HM-600": Limit acknowledged
2024-08-27 14:41:44 INFO     Ahoy: Inverter "HM-600" Available: True
2024-08-27 14:41:44 INFO     Ahoy: Inverter "HM-800" Available: True
2024-08-27 14:41:44 INFO     Ahoy: Inverter "HM-1500" Available: True
2024-08-27 14:41:44 INFO     Lowest panel voltage inverter "HM-800": 54.0 Volt
2024-08-27 14:41:44 INFO     Average min-panel voltage, inverter "HM-800": 54.0 Volt
2024-08-27 14:41:44 INFO     Ahoy: Inverter "HM-800": Turn on
2024-08-27 14:41:55 INFO     Lowest panel voltage inverter "HM-1500": 53.9 Volt
2024-08-27 14:41:55 INFO     Average min-panel voltage, inverter "HM-1500": 53.9 Volt
2024-08-27 14:41:55 INFO     Ahoy: Inverter "HM-1500": Turn on
2024-08-27 14:42:05 INFO     powermeter Shelly3EMPro: -103 Watt
2024-08-27 14:42:06 INFO     powermeter Shelly3EMPro: -108 Watt
2024-08-27 14:42:07 INFO     powermeter Shelly3EMPro: -108 Watt
2024-08-27 14:42:08 INFO     powermeter Shelly3EMPro: -100 Watt
2024-08-27 14:42:09 INFO     powermeter Shelly3EMPro: -113 Watt
2024-08-27 14:42:10 INFO     powermeter Shelly3EMPro: -107 Watt
2024-08-27 14:42:11 INFO     powermeter Shelly3EMPro: -108 Watt
2024-08-27 14:42:12 INFO     powermeter Shelly3EMPro: -105 Watt
[... 105 lines cut ...]
2024-08-27 14:43:47 INFO     powermeter Shelly3EMPro: -15 Watt
2024-08-27 14:43:49 INFO     powermeter Shelly3EMPro: -17 Watt
2024-08-27 14:43:50 INFO     powermeter Shelly3EMPro: -8 Watt
2024-08-27 14:43:51 INFO     Inverterlimit was already accepted at 189 Watt
2024-08-27 14:43:51 INFO     Ahoy: Inverter "HM-600" Available: True
[... cut ...]
2024-08-27 14:43:53 INFO     powermeter Shelly3EMPro: -14 Watt
2024-08-27 14:43:54 INFO     powermeter Shelly3EMPro: -17 Watt
2024-08-27 14:43:55 INFO     powermeter Shelly3EMPro: -10 Watt
2024-08-27 14:43:56 INFO     powermeter Shelly3EMPro: -18 Watt
2024-08-27 14:43:57 INFO     powermeter Shelly3EMPro: -10 Watt
2024-08-27 14:43:58 INFO     powermeter Shelly3EMPro: -4 Watt
2024-08-27 14:43:59 INFO     powermeter Shelly3EMPro: -4 Watt
2024-08-27 14:44:01 INFO     powermeter Shelly3EMPro: -1 Watt

```

RealNBB, 2024-09-01:
@reserve85 Du hast ja das Label eh schon geändert, aber ich geb doch noch ein Feedback ab! 🗡️ 

Das Problem, dass das Skript wild die Limits geändert hat - insbesondere bei der Konstallation, dass die Sonne stark scheint und der Grundverbrauch dabei sehr niedrig ist - trat nicht mehr auf. Nach Neustart startet das Skript direkt sauber durch, die erforderlichen Limits werden sofor eingestellt und umgesetzt! Ich denk, da kann man tatsächlich ein Haken dran machen. :)

Allerdings habe ich nun ein neues Problemchen:

Skript läuft in V1.101 zusammen mit Ahoy 8.143. Eingestellt ist "HOY_MIN_WATT_IN_PERCENT" = 4% für den HM-1500 (und wie gehabt, Prio 1 für den HM-1500 und Prio 2 für den HM-800). Einstellen tut das Skript folgendes:

[image]

Er regelt den 1500 einfach nicht auf die eingestellten 4% runter, beim 800er passt es mit 5%.

Das Log sagt diesbeüglich nichts aus, ausser dass das Limit bereits akzeptiert wurde (bei keiner geplanten Änderung des Limits.

reserve85, 2024-09-01:
Kannst du mal ein Log anhängen? Sonst kann ich das schlecht nachvollziehen.

RealNBB, 2024-09-02:
Anbei das Log-File sowie die Config nochmal.

[Log.txt](https://github.com/user-attachments/files/16838682/Log.txt)
[Config.txt](https://github.com/user-attachments/files/16838683/Config.txt)

reserve85, 2024-09-02:
Ok, das hilft mir. Schaue ich mir die Tage an…

RealNBB, 2024-09-03:
was mir übrigens auch noch aufgefallen ist: nach einem Neustart kurz (gilt für den 800 und 1500 gleichermaßen) starten die beiden WR  mit den bei "HOY_BATTERY_REDUCE_WATT" eingestellten Werzen und verharren dann auf dieser Leistung. Das Skript schafft es dann nicht mehr, die WR anzusprechen bzw. zu regeln. Evtl aber auch nur ein Ahoy Bug.

reserve85, 2024-09-03:
Hab das heute angefangen die Berechnung zu fixen aber keine Zeit mehr gehabt. Denke aber ich hab den Fehler gefunden, war auf den ersten Blick recht einfach aber am Ende doch nicht so leicht... wie immer halt 😄.  Hoffe morgen kannst du es probieren.

zu dem anderen Problem: kannst du mir das auch im log zeigen?

RealNBB, 2024-09-03:
Nice....king gut!
Stellst du das dann wieder als Dev zur Verfügung? Sonst bekomm ich das doch wieder nicht hin.

Neustart Log mach ich dir morgen.

reserve85, 2024-09-04:
ist online als "dev"

RealNBB, 2024-09-06:
Hey...entschuldige bitte die späte Antwort. Aber momentan komme ich zu wenig.
Das mit min. Prozent sieht gut aus. Zumindest stellt er beim HM-1500 nun wieder die geforderten 4% ein.

Neustart-Log muss ich erst noch machen. Befürchte aber, dass ich am Wochenende eher nicht dazu komme. Bekommst es aber schon noch.

oki...dann mal vorab ein guten start ins Wochenende.

[image]

RealNBB, 2024-09-08:
Was mir jetzt doch aufgefallen ist bei V1.102:

Das Skript stellt den HM-600 nun nicht mehr auf 100%, sondern meist auf nur 83,3%:

[image]

Bei V1.101 war das m.W. nach nicht so, da lief der auf 100%, wenn die Leistung benötigt worden ist.

Log-Auszug im Anhang.
[Log.txt](https://github.com/user-attachments/files/16921879/Log.txt)

reserve85, 2024-09-08:
Schau ich… ;-)

reserve85, 2024-09-08:
habs gerade nochmal aktualisiert, du könntest es nochmal abrufen und einspielen...
Version ist die gleiche

RealNBB, 2024-09-09:
Okay...das mit den "HOY_MIN_WATT_IN_PERCENT" sieht jetzt ganz gut aus: der HM600 läuft nun bei 100%, während die beiden anderen bei ihren fest eingestellten min_prozent_watt rumdümpeln. Aber lass mich das bitte die Tage noch weiter beobachten.

Bezgl. der Neustartproblematik anbei zwei Logs…einmal für den HM800 und einmal für den HM1500. Der HM800 stellt sich nach einem Reboot bei 408W ein, der HM1500 bei 765W. Also NICHT wie ich oben geschrieben habe bei "HOY_BATTERY_REDUCE_WATT" sondern bei ziemlich genau 50% der Leistung des jeweiligen WR.

[Log_Restart_HM800.txt](https://github.com/user-attachments/files/16930305/Log_Restart_HM800.txt)
[Log_Restart_HM1500.txt](https://github.com/user-attachments/files/16930306/Log_Restart_HM1500.txt)

RealNBB, 2024-09-10:
HM-800 nach Reboot:

[image]

HM-1500 nach Reboot:

[image]

reserve85, 2024-09-10:
ja, that´s the problem. ich bekomme "3.9%" und "2.9%" aus AHOY, allerdings produziert der Inverter mit ~50%

RealNBB, 2024-09-10:
Ja genau...das passt nicht so ganz zusammen! 🗡️ 

Das ursprüngliche Problem "HM-600 geht nicht auf 100%" scheint aber mit der neuen V1.102 gelöst zu sein:

[image]

reserve85, 2024-09-10:
Das musst du eigentlich bei AHOY öffnen. Müsste nach einem Reboot-Klick in AHOY zurückgesetzt werden...

reserve85, 2024-09-10:
ja, das ist ein AHOY Problem, der gibt weiterhin das letzte Limit nach außen weiter (und auch in der Weboberfläche) obwohl der WR neugestartet wurde und intern was ganz anderes anregelt.
Nach dem AHOY Neustart fragt er das Limit wieder aktiv ab.

RealNBB, 2024-09-10:
Japp...okay! Dann kann man da wenig bis nichts machen, wenn das ein Thema von Agoy an sich ist.

Zum anderen: Geb mir noch etwas Zeit, so zwei, drei Tage wegen den 100% beim HM-600, aber ich denke, dann kannst du auch daran einen Haken machen.

RealNBB, 2024-09-12:
Okay...das sieht nun etwas komisch aus:

[image]

Sollte nicht eigentlich der HM-1500 bei 3% stehen und der HM-600 hat noch Luft nach oben?

Log dazu im Ahang:

[Log.txt](https://github.com/user-attachments/files/16978630/Log.txt)

RealNBB, 2024-09-12:
Zu spät dafür! ;)

Aber ja...Ahoy ist das Problem, 0.8.143 hat sich mal wieder verschluckt. Ich teste nun mit der 0.8.129, die ist ja nicht so zickig.

reserve85, 2024-09-13:
evtl kannst du _nochmal_ auf die neue dev updaten, ich habe da die ganzen SetLimit Funktionen zu einer zusammengefasst und die Berechnung nochmal besser gemacht... Log Ausgaben für die Inverter sind auch erweitert.

Wäre super

RealNBB, 2024-09-13:
V1.103 läuft!

Die nächsten Tage werden interessant, da ein Mix aus  Wolken und Sonne. Da kann das Skript dann schön regeln! :-D

Ich berichte!

RealNBB, 2024-09-16:
Also V1.103 läuft echt top, völlig anstandslos! 
Konnte auch die letzten vier Tage Dank HomeOffice und Wochenende einige verschiede Settings ausprobieren. Alle liefen mind. 2h ohne irgendwelches komisches Verhalten des Skripts sauber durch. 

[Test.txt](https://github.com/user-attachments/files/17016064/Test.txt)

reserve85, 2024-09-16:
Danke fürs Testen und schön dass bei dir alles funktioniert!

mi-hol, 2024-11-07:
bedeutet das Label "completed" nicht, dass das issue eigentlich "closed" ist?

Die Anzahl "open" issues wir oft zur Bewertung eines Projekt benutzt.
Dabei gilt: je weniger, desdo besser!

closed 2024-11-08 by reserve85