reserve85/HoymilesZeroExport #232: Frage zur Berücksichtigung von Wallboxes

trombose009, 2024-08-28:
Hallo zusammen, 

ich bin begeisterter User. Folgendes: Ich besitze noch einige Elektroautos mit drei wallboxes. Hat jemand eine Idee, wie man berücksichtigen könnte, dass gerade ein Auto lädt? Momentan muss ich da manuell eingreifen, ansonsten schießt mein Speicher über die Script steuerung sukzessive Strom zu, wenn der Strom der PV Anlage vom großen Dach mal nicht ausreicht. das ist nicht gewollt. Die Wallboxen sind über EVCC gesteuert und laden nur, wenn PV Überschuss da ist. Sie können aber nicht unterscheiden, ob der Strom vom Dach oder aus dem Akku vom Hoymiles beigesteuert wird. Im Resultat erhöht das HoymilesZeroExport Script immer weiter den output und das Auto frisst  dann gegen Nachmittag oder Abend alles weg obwohl es das gar nicht unbedingt müsst. 

Kann man verstehen was ich geschrieben habe?  Wenn ja, gibt es eine Idee dazu? Oder gar schon eine Implementierung, wie ich dem script sagen kann, dass es gerade auf keinen Fall was ausgeben soll?  Hätte als Werkzeuge auch noch node-red  im Einsatz. 

Grüße

tomquist, 2024-08-28:
Bekommst du in NodeRED mit wenn das Auto lädt? Wenn ja, könntest du das Script mit einem MQTT Broker verbinden und dann per MQTT den Target Point auf einen hohen positiven Wert verschieben, sobald das Auto lädt.

reserve85, 2024-08-28:
Würde ich auch so vorschlagen, du müsstest in NodeRed dein Auto rausrechnen...

trombose009, 2024-08-28:
[image]

meint ihr das so? ich bekomme vom EVCC alle 20s die info geschickt wieviel Strom in der Wallbox fließt. Habe jetzt eingestellt, dass bei 0 der Wert -75 via mqtt  an zeropower/set/powermeter_target_point gesendet wird. Wenn was fließt der Wert 15000. 

ist das so korrekt?

trombose009, 2024-08-28:
oder ist es gar irgenwie schädlich den Wert alle 20s erneut mit meist dem selben Wert zu überschreiben?

trombose009, 2024-08-28:
ah der flow war noch falsch. der hat nur den ampere-wert weitergegeben. jetzt passt es.
[image]

closed 2024-08-28 by trombose009

tomquist, 2024-08-28:
Alles 20s sollte eigentlich kein Problem darstellen. Die Werte werden lokal gecached und es wird in jeder Schleife des Scripts der zuletzt gesendete Wert genutzt. Ich schreibe den Wert sogar sekündlich in Abhängigkeit von verschiedenen Parametern.

trombose009, 2024-08-28:
danke, ihr sexy menschen für den supergeilen support für so noobs wie mich. Als Endanwender beschäftigt man sich halt immer einmal damit, dann läuft alles und dann hat man halt teilweise ewig nix mehr mit der Materie zu tun. Ich bin da immer sehr dankbar wenn ich nicht gleich runtergeprügelt werde, nur weil ich versehentlich was frage, was in irgendwelche FAQs stehen müsste oder so.
Ich mach bei ein paar Leuten schon fleißig Werbung für das script. :)

reserve85, 2024-08-28:
> Alles 20s sollte eigentlich kein Problem darstellen. Die Werte werden lokal gecached und es wird in jeder Schleife des Scripts der zuletzt gesendete Wert genutzt. Ich schreibe den Wert sogar sekündlich in Abhängigkeit von verschiedenen Parametern.
good to know

@trombose009 die prügel heb ich mir für nächste mal auf ⚔️ 😆