reserve85/HoymilesZeroExport #235: Nach update auf Version V1.100 kommt  "'newLimitSetpoint' is not defined"

XpertXXL, 2024-08-29:
bisher lief das Script absolut stabil und super :-)
Seit dem Update auf Version V1.100 kommt o. a. FM.
Es handelt sich bei mir um eine Installation im Docker Container.
Habe den Container auch schon gelöscht und neu installiert, keine Änderung.
Habe testweise auch mein HoymilesZeroExport_Config_Override.ini File gegen ein minimal Konfiguration File getauscht, auch keine Änderung :-(

Auszug aus Portainer Logfile:
2024-08-29 10:26:48 INFO     Python Version: 3.11.2 (main, May  2 2024, 11:59:08) [GCC 12.2.0]
2024-08-29 10:26:48 INFO     Author: Tobias Kraft / Script Version: 1.100
2024-08-29 10:26:48 INFO     read config file: /app/HoymilesZeroExport_Config.ini
2024-08-29 10:26:48 INFO     read additional config file: ./HoymilesZeroExport_Config_Override.ini
2024-08-29 10:26:48 INFO     Config file V 1.97
2024-08-29 10:26:48 INFO     ---Init---
2024-08-29 10:26:49 INFO     OpenDTU: Current Version: v24.5.27-Database
2024-08-29 10:26:49 ERROR    Invalid version: 'v24.5.27-Database'
2024-08-29 10:27:09 INFO     ---Start Zero Export---
2024-08-29 10:27:09 ERROR    name 'newLimitSetpoint' is not defined
2024-08-29 10:27:29 ERROR    name 'newLimitSetpoint' is not defined

hat jemand dieselben Probleme, oder liegt der Fehler bei mir??

Euer XpertXXL

reserve85, 2024-08-29:
ah ja ich sehs schon, fixe ich nachher...
sorry!!

reserve85, 2024-08-29:
sollte gefixed sein, kannst du es nochmal probieren mit der V1.101?

XpertXXL, 2024-08-29:
suuuper Support, vielen Dank dafür!!!
funktioniert wieder :-)
LG XpertXXL

closed 2024-08-29 by reserve85