reserve85/HoymilesZeroExport #241: 3ter Inverter ignoriert SET_LIMIT_TIMEOUT_SECONDS = 3

martin-st-81, 2024-09-11:
Ich verwende schon seit einiger Zeit dieses hervorragende Script. Seit ich einen 3ten Inverter verwende, läuft dieser auf ein Timeout.
In der Opendtu ist ersichtlich das der Wert nach ca. 2-3 Sekunden auf acknowledged geht. Wie auch bei beiden anderen Invertern.
Laut Log wartet das Script beim dritten Inverter aber immer nur 1 sec und nicht den gesetzten Wert von 3 Sekunden.

Bitte um Unterstützung

Attaching to HoymilesZeroExport
HoymilesZeroExport    | 2024-09-11 09:16:41 INFO     Log write to file: False
HoymilesZeroExport    | 2024-09-11 09:16:41 INFO     Python Version: 3.11.2 (main, Aug 26 2024, 07:20:54) [GCC 12.2.0]
HoymilesZeroExport    | 2024-09-11 09:16:41 INFO     Author: Tobias Kraft / Script Version: 1.101
HoymilesZeroExport    | 2024-09-11 09:16:41 INFO     read config file: /app/HoymilesZeroExport_Config.ini
HoymilesZeroExport    | 2024-09-11 09:16:41 INFO     read additional config file: /app/config.ini
HoymilesZeroExport    | 2024-09-11 09:16:41 INFO     Config file V 1.97
HoymilesZeroExport    | 2024-09-11 09:16:41 INFO     ---Init---
HoymilesZeroExport    | 2024-09-11 09:16:42 INFO     OpenDTU: Current Version: v24.8.5
HoymilesZeroExport    | 2024-09-11 09:16:42 INFO     OpenDTU: Inverter "yet unknown" reachable: True
HoymilesZeroExport    | 2024-09-11 09:16:42 INFO     OpenDTU: Inverter "Hoymiles1" / serial number "1164XXX" / temperature 26.5 degC
HoymilesZeroExport    | 2024-09-11 09:16:42 INFO     OpenDTU: Inverter "yet unknown" reachable: True
HoymilesZeroExport    | 2024-09-11 09:16:42 INFO     OpenDTU: Inverter "Hoymiles1" / serial number "1164XXX" / temperature 26.5 degC
HoymilesZeroExport    | 2024-09-11 09:16:42 INFO     OpenDTU: Inverter "Hoymiles2" / serial number "1164XXX" / temperature 26.3 degC
HoymilesZeroExport    | 2024-09-11 09:16:42 INFO     OpenDTU: Inverter "yet unknown" reachable: True
HoymilesZeroExport    | 2024-09-11 09:16:42 INFO     OpenDTU: Inverter "Hoymiles1" / serial number "1164XXX" / temperature 26.5 degC
HoymilesZeroExport    | 2024-09-11 09:16:42 INFO     OpenDTU: Inverter "Hoymiles2" / serial number "1164XXX" / temperature 26.3 degC
HoymilesZeroExport    | 2024-09-11 09:16:42 INFO     OpenDTU: Inverter "Hoymiles3" / serial number "1164XXX" / temperature 26.9 degC
HoymilesZeroExport    | 2024-09-11 09:16:42 INFO     OpenDTU: Inverter "Hoymiles1": Turn on
HoymilesZeroExport    | 2024-09-11 09:16:53 INFO     OpenDTU: Inverter "Hoymiles2": Turn on
HoymilesZeroExport    | 2024-09-11 09:17:03 INFO     OpenDTU: Inverter "Hoymiles3": Turn on
HoymilesZeroExport    | 2024-09-11 09:17:13 INFO     setting new limit to 720 Watt
HoymilesZeroExport    | 2024-09-11 09:17:13 INFO     OpenDTU: Inverter "Hoymiles1": setting new limit from -1 Watt to 240 Watt
HoymilesZeroExport    | 2024-09-11 09:17:16 INFO     OpenDTU: Inverter "Hoymiles1": Limit acknowledged
HoymilesZeroExport    | 2024-09-11 09:17:16 INFO     OpenDTU: Inverter "Hoymiles2": setting new limit from -1 Watt to 240 Watt
HoymilesZeroExport    | 2024-09-11 09:17:19 INFO     OpenDTU: Inverter "Hoymiles2": Limit acknowledged
HoymilesZeroExport    | 2024-09-11 09:17:19 INFO     OpenDTU: Inverter "Hoymiles3": setting new limit from -1 Watt to 240 Watt
HoymilesZeroExport    | 2024-09-11 09:17:19 INFO     OpenDTU: Inverter "Hoymiles3": Limit timeout!
HoymilesZeroExport    | 2024-09-11 09:17:19 INFO     intermediate meter OpenDTU: 1127 Watt
HoymilesZeroExport    | 2024-09-11 09:17:19 INFO     powermeter Shrdzm: -435 Watt
HoymilesZeroExport    | 2024-09-11 09:17:19 INFO     ---Start Zero Export---
HoymilesZeroExport    | 2024-09-11 09:17:20 INFO     OpenDTU: Inverter "Hoymiles1" reachable: True
HoymilesZeroExport    | 2024-09-11 09:17:20 INFO     OpenDTU: Inverter "Hoymiles2" reachable: True
HoymilesZeroExport    | 2024-09-11 09:17:20 INFO     OpenDTU: Inverter "Hoymiles3" reachable: True
HoymilesZeroExport    | 2024-09-11 09:17:20 INFO     powermeter Shrdzm: -435 Watt
HoymilesZeroExport    | 2024-09-11 09:17:21 INFO     powermeter Shrdzm: -435 Watt

[COMMON]
# Number of Inverters
INVERTER_COUNT = 3
# max difference in percent between SetpointLimit change to approximate the power to new setpoint
SLOW_APPROX_LIMIT_IN_PERCENT = 20
# if slow approximation: additional limit based on the limit-difference to "smoot the curve": newLimitSetpoint = newLimitSetpoint + (LimitDifference * SLOW_APPROX_FACTOR_IN_PERCENT / 100)
SLOW_APPROX_FACTOR_IN_PERCENT = 20
# interval time for setting limit to Hoymiles
LOOP_INTERVAL_IN_SECONDS = 5
# Timeout time to wait for Acknowledge after sending limit to Hoymiles Inverter
SET_LIMIT_TIMEOUT_SECONDS = 3
# polling interval for powermeter (must be <= LOOP_INTERVAL_IN_SECONDS)
POLL_INTERVAL_IN_SECONDS = 1
# if your powermeter exceeds POWERMETER_MAX_POINT: immediatelly set the limit to predefined percent of HOY_MAX_WATT (if you have more than one inverter itÂ´s the sum of all HOY_MAX_WATT)
# value = 0 disables the feature. Values are possible from [0 to 100]
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 100
# if your powermeter falls below POWERMETER_MIN_POINT: immediatelly decrease the limit
ON_GRID_FEED_FAST_LIMIT_DECREASE = false
# max difference between Limit and real output power in % of HOY_MAX_WATT (100 = disabled)
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 100
# enable logging to file
ENABLE_LOG_TO_FILE = false
# how many logfiles you wish to keep
LOG_BACKUP_COUNT = 30
# defines how often the Inverter Power Status will be set, set it to "-1" for disabled (infinite repeat)
SET_POWERSTATUS_CNT = 10
# log the inverter temperature
LOG_TEMPERATURE = false
# delay time after turning the inverter off or on
SET_POWER_STATUS_DELAY_IN_SECONDS = 10
# define if you want to set your inverter to min-limit when your powermeter can't be read out
SET_INVERTER_TO_MIN_ON_POWERMETER_ERROR = true
# Total number of retries to allow.
MAX_RETRIES = 3
# A set of integer HTTP status codes that we should force a retry on. DonÂ´t change unless you know what you are doing
RETRY_STATUS_CODES = 500,502,503,504
# It allows you to change how long the process will sleep between failed requests. The algorithm is as follows: {backoff factor} * (2 ** ({number of total retries} - 1))
RETRY_BACKOFF_FACTOR = 0.1

[CONTROL]
# --- global defines for control behaviour ---
# POWERMETER_TARGET_POINT is the target power for powermeter in watts
POWERMETER_TARGET_POINT = -600
# POWERMETER_TOLERANCE is the tolerance (pos and neg) around the target point. in this range no adjustment will be set
POWERMETER_TOLERANCE = 50
# POWERMETER_MAX_POINT is the maximum power of your powermeter for the normal "regulation loop".
# if your powermeter jumps over this point, the limit will be increased instantly. it is like a "super high priority limit change".
# if you defined ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT > 0, then the limit will jump to the defined percent when reaching this point.
POWERMETER_MAX_POINT = 0
# POWERMETER_MIN_POINT is the minimum power of your powermeter for the normal "regulation loop".
# if your powermeter jumps under this point, the limit will be reduced instantly. it is like a "super high priority limit change".
POWERMETER_MIN_POINT = -800

[INVERTER_1]
SERIAL_NUMBER = 1164XXX
# enable (true) / disable (false) this inverter
ENABLED = true
# manufacturer power rating of your inverter.
HOY_INVERTER_WATT = 2000
# power rating of your inverter
HOY_MAX_WATT = 2000
# minimum limit in percent, e.g. 5%
HOY_MIN_WATT_IN_PERCENT = 12
# factor to multiply before set Limit. Some Inverters have some offsets, with that factor you can compensate it. Default = 1
HOY_COMPENSATE_WATT_FACTOR = 1
# battery powered?
HOY_BATTERY_MODE = false

[INVERTER_2]
SERIAL_NUMBER = 1164XXX
# enable (true) / disable (false) this inverter
ENABLED = true
# manufacturer power rating of your inverter.
HOY_INVERTER_WATT = 2000
# power rating of your inverter
HOY_MAX_WATT = 2000
# minimum limit in percent, e.g. 5%
HOY_MIN_WATT_IN_PERCENT = 12
# factor to multiply before set Limit. Some Inverters have some offsets, with that factor you can compensate it. Default = 1
HOY_COMPENSATE_WATT_FACTOR = 1
# battery powered?
HOY_BATTERY_MODE = false

[INVERTER_3]
SERIAL_NUMBER = 1164XXX
# enable (true) / disable (false) this inverter
ENABLED = true
# manufacturer power rating of your inverter.
HOY_INVERTER_WATT = 2000
# power rating of your inverter
HOY_MAX_WATT = 2000
# minimum limit in percent, e.g. 5%
HOY_MIN_WATT_IN_PERCENT = 12
# factor to multiply before set Limit. Some Inverters have some offsets, with that factor you can compensate it. Default = 1
HOY_COMPENSATE_WATT_FACTOR = 1
# battery powered?
HOY_BATTERY_MODE = false

reserve85, 2024-09-11:
Hi,

da stimmt die Log-Ausgabe vermutlich nicht ganz.
Ich werde das Script dahingehend erweitern, damit die korrekte Fehlermeldung im Log erscheint.

martin-st-81, 2024-09-11:
Der 3te Inverter übernimmt den Wert jedenfalls laut Opendtu, nur das Script schickt den "setting new limit" laufend obwohl dieser bereits gesetzt ist. 

HoymilesZeroExport    | 2024-09-11 09:39:54 INFO     OpenDTU: Inverter "Hoymiles3": setting new limit from 366 Watt to 366 Watt
HoymilesZeroExport    | 2024-09-11 09:39:54 INFO     OpenDTU: Inverter "Hoymiles3": Limit timeout!
HoymilesZeroExport    | 2024-09-11 09:39:55 INFO     OpenDTU: Inverter "Hoymiles1" reachable: True
HoymilesZeroExport    | 2024-09-11 09:39:55 INFO     OpenDTU: Inverter "Hoymiles2" reachable: True
HoymilesZeroExport    | 2024-09-11 09:39:55 INFO     OpenDTU: Inverter "Hoymiles3" reachable: True
HoymilesZeroExport    | 2024-09-11 09:39:55 INFO     powermeter Shrdzm: -574 Watt
HoymilesZeroExport    | 2024-09-11 09:39:56 INFO     powermeter Shrdzm: -574 Watt
HoymilesZeroExport    | 2024-09-11 09:39:58 INFO     powermeter Shrdzm: -594 Watt
HoymilesZeroExport    | 2024-09-11 09:39:59 INFO     powermeter Shrdzm: -594 Watt
HoymilesZeroExport    | 2024-09-11 09:40:00 INFO     powermeter Shrdzm: -594 Watt
HoymilesZeroExport    | 2024-09-11 09:40:01 INFO     Inverterlimit 1099 Watt was previously not accepted by at least one inverter, trying again...
HoymilesZeroExport    | 2024-09-11 09:40:01 INFO     setting new limit to 1099 Watt
HoymilesZeroExport    | 2024-09-11 09:40:01 INFO     OpenDTU: Inverter "Hoymiles3": setting new limit from 366 Watt to 366 Watt
HoymilesZeroExport    | 2024-09-11 09:40:01 INFO     OpenDTU: Inverter "Hoymiles3": Limit timeout!
HoymilesZeroExport    | 2024-09-11 09:40:02 INFO     OpenDTU: Inverter "Hoymiles1" reachable: True
HoymilesZeroExport    | 2024-09-11 09:40:02 INFO     OpenDTU: Inverter "Hoymiles2" reachable: True
HoymilesZeroExport    | 2024-09-11 09:40:02 INFO     OpenDTU: Inverter "Hoymiles3" reachable: True
HoymilesZeroExport    | 2024-09-11 09:40:03 INFO     powermeter Shrdzm: -553 Watt
HoymilesZeroExport    | 2024-09-11 09:40:04 INFO     powermeter Shrdzm: -553 Watt
HoymilesZeroExport    | 2024-09-11 09:40:05 INFO     powermeter Shrdzm: -553 Watt
HoymilesZeroExport    | 2024-09-11 09:40:06 INFO     powermeter Shrdzm: -553 Watt
HoymilesZeroExport    | 2024-09-11 09:40:08 INFO     powermeter Shrdzm: -554 Watt
HoymilesZeroExport    | 2024-09-11 09:40:09 INFO     Inverterlimit 1099 Watt was previously not accepted by at least one inverter, trying again...
HoymilesZeroExport    | 2024-09-11 09:40:09 INFO     setting new limit to 1099 Watt
HoymilesZeroExport    | 2024-09-11 09:40:09 INFO     OpenDTU: Inverter "Hoymiles3": setting new limit from 366 Watt to 366 Watt
HoymilesZeroExport    | 2024-09-11 09:40:10 INFO     OpenDTU: Inverter "Hoymiles3": Limit timeout!
HoymilesZeroExport    | 2024-09-11 09:40:10 INFO     OpenDTU: Inverter "Hoymiles1" reachable: True
HoymilesZeroExport    | 2024-09-11 09:40:10 INFO     OpenDTU: Inverter "Hoymiles2" reachable: True
HoymilesZeroExport    | 2024-09-11 09:40:10 INFO     OpenDTU: Inverter "Hoymiles3" reachable: True

reserve85, 2024-09-11:
ja, da stimmt was nicht mit der Rückgabe oder Auswertung.

kannst du bitte mal schaun was jeweils hier ausgelesen wird (kannst du im Brower eingeben):

http://OPEN-DTU-IP/api/inverter/id/0
http://OPEN-DTU-IP/api/inverter/id/1
http://OPEN-DTU-IP/api/inverter/id/2

OPEN-DTU-IP natürlich mit deiner ersetzen

martin-st-81, 2024-09-11:
die drei Befehle geben leider nichts zurück
http://open-dtu-ip/api/inverter/id/0
http://open-dtu-ip/api/inverter/id/1
http://open-dtu-ip/api/inverter/id/2

[image]

reserve85, 2024-09-11:
ah ich depp... das war für AHOY...

bitte nochmal:
http://192.168.25.32/api/limit/status

martin-st-81, 2024-09-11:
[image]

reserve85, 2024-09-11:
Die Seriennummern passen?

reserve85, 2024-09-11:
ne da kommt vermutlich kein Timeout, die Log Ausgabe passt von mir nicht.

kannst du diese Version hier einspielen und versuchen:
https://github.com/reserve85/HoymilesZeroExport/tree/dev

brauchst vermutlich nur die *.py austauschen

martin-st-81, 2024-09-11:
Was muss ich hier bei docker-compose eintragen ?

version: '3.3'
services:
  hoymileszeroexport:
    container_name: HoymilesZeroExport
    restart: unless-stopped
    image: ghcr.io/reserve85/hoymileszeroexport:main
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./config/HoymilesZeroExport_Config_Override.ini:/app/config.ini
    command: -c /app/config.ini

reserve85, 2024-09-11:
ahso, dann kannst du das so verwenden: https://github.com/reserve85/HoymilesZeroExport/issues/228#issuecomment-2310855595

martin-st-81, 2024-09-11:
Kann es sein, weil beim Inverter 3 auch Buchstaben und nicht nur Zahlen in der Seriennummer sind ?
Meine beiden älteren Hoymiles Wechselrichter haben nur Zahlen in der Serialnummer

[image]

reserve85, 2024-09-11:
ja, irgend so ein schrott wird es sein... :-)

martin-st-81, 2024-09-11:
Ich konnte den Fehler finden :-(
Aufgrund der Buchstaben in der Seriennummer ist es offensichtlich wichtig die Groß und Kleinschreibung zu beachten 
Habe jetzt in deinem Script Kleinbuchstaben verwendet und es funktioniert !

bitte entschuldige die Umstände, dessen war ich mir nicht bewusst 
Danke für die schnelle Hilfe !

reserve85, 2024-09-11:
ja Mega, auch wieder was gelernt.

martin-st-81, 2024-09-11:
Bei Wechselrichtern mit Buchstaben in der Seriennummer, müssen zwingend Kleinbuchstaben verwendet werden

closed 2024-09-11 by martin-st-81