reserve85/HoymilesZeroExport #242: OpenDTU Inverter in der Früh offline - Command Queue voll

martin-st-81, 2024-09-13:
Ich verwende 3 Hoymiles Inverter mit OpenDTU v24.8.5
Opendtu verwendet hier 2 Status (reachable & producing) um die Erreichbarkeit der Wechselrichter zu signalisieren.

reachable -> hier wird aufgrund Zeit und den jeweiligen Dämmerungseinstellungen, versucht die Wechselrichter zu erreichen 
producing -> sobald eine Kommunikation aufgebaut ist geht "producing" auf true

Wenn ich diese tolle Script richtig verstehe, wird auf "reachable" kontrolliert und nicht zusätzlich auf "producing" was dazu führt, dass das Script "ActivePowerLimit" commands an OpenDTU sendet ... auch wenn der Inverter aufgrund "producing" - false nicht erreichbar ist.
Dies führt dazu, dass in der Früh die command Queue von Opendtu voll ist und die Inverter offline bleiben 

Ist es möglich auf beide Faktoren (reachable & producing) zu kontrollieren, bevor das Script ein neues Limit setzen möchte ?
[image]

reserve85, 2024-09-13:
Wozu dann reachable wenn er nicht "erreichbar" ist? Ist das feature neu?

martin-st-81, 2024-09-13:
Für mein Verständnis sollte er aufgrund errechneten Sonnenstand "reachable" sein, und Opendtu versucht diesen an zu funken.
Was aber nicht bedeutet, dass er dann auch immer erreichbar ist. zB bei Schlechtwetter, oder Verschattung. 
Das Funkmodul der Hoymiles werden ja von der DC Seite gespeist. Erst mit "producing" true, liefert der Wechselrichter Leistung und hat gleichzeitig eine aktive Verbindung zu OpenDTU.

Ob das neu ist kA, dieses Verhalten beobachte ich schon des längeren in Früh und muss das Script stoppen & die Opendtu rebooten damit er dann nach 5-10min eine Verbindung aufbaut. Dann starte ich das Script und es läuft wieder wie gewünscht.
Unmittelbar nach dem reboot der OpenDTU steht "reachable" auf TRUE obwohl noch keine Leistungsdaten ausgelesen wurden.
Sobald die ersten Daten angeliefert werden, geht OpenDTU auf "producing" true

reserve85, 2024-09-13:
Ja, schau mal hier: https://github.com/reserve85/HoymilesZeroExport/issues/211
Das ist genau der Fall.

Wäre ja Top, wenn die Lösung so einfach wäre...

martin-st-81, 2024-09-13:
in #211 
scheint es ebenfalls so zu sein, denn um 21:20 "reachable: True" kann ja keine aktive Verbindung zum Wechselrichter (mangels Sonnenschein) sein. Dieses Verhalten ist bei mir jedenfalls das Gleiche.

Sollte dies wirklich an OpenDTU scheitern werden ich wohl auf AhoyDTU wechseln. 

Jun 18 21:19:52 raspberrypi python3[506]: 2024-06-18 21:19:52 INFO     OpenDTU: Inverter "HMS-1600-4T": setting new limit from -1 Watt to 1003 Watt
Jun 18 21:19:58 raspberrypi python3[506]: 2024-06-18 21:19:58 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit timeout!
Jun 18 21:19:58 raspberrypi python3[506]: 2024-06-18 21:19:58 INFO     OpenDTU: Inverter "HMS-2000-4T": setting new limit from 1933 Watt to 1253 Watt
Jun 18 21:20:04 raspberrypi python3[506]: 2024-06-18 21:20:04 INFO     OpenDTU: Inverter "HMS-2000-4T": Limit timeout!
Jun 18 21:20:10 raspberrypi python3[506]: 2024-06-18 21:20:10 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Jun 18 21:20:10 raspberrypi python3[506]: 2024-06-18 21:20:10 INFO     OpenDTU: Inverter "HMS-2000-4T" reachable: True

reserve85, 2024-09-13:
Für mein Verständnis ist das ganz klar falsch, reachable bedeutet für mich "erreichbar".
Also nur wenn auch kommuniziert werden kann.

martin-st-81, 2024-09-13:
Ich werde heute am Abend, wenn die Opendtu keine Werte mehr empfängt, die Werte nochmal kontrollieren/vergleichen - möchte keine unnötige Arbeit verursachen. Dieses Script arbeitet so hervorragend dass ev. ein Fehler in Opendtu nicht hier zu Arbeitsaufwand führen sollte

martin-st-81, 2024-09-13:
ich lass es mal laufen und werde am Abend die Daten vergleichen, Danke für deine rasche Hilfe !

[image]

reserve85, 2024-09-13:
Ich glaube das geht aber so nicht bei Akku-Inverterm, da wird ja der Inverter (optional) komplett ausgeschaltet (Turn Off) -> dann dürfte auch das Flag "producing" auf "false" stehen... 
Da würde der Inverter dann nie wieder aktiviert werden?

martin-st-81, 2024-09-13:
Meine Vermutung hat sich bestätigt. 
Reachable geht erst mit der Abenddämmerung auf false
Producing geht auf false sobald der Wechselrichter keine Daten mehr liefert und somit auch keine Empfangen kann... ein paar Minuten davor
Ob das jetzt bei OpenDTU gewünscht ist oder ein Bug, kann ich nicht beurteilen 
Aufgrund deiner Änderung im Dev. Script V1.104, sendet das Script keine "ActivePowerLimit" commands mehr an die OpenDTU welche dann in der Früh die Queue blockieren da diese nicht an den Wechselrichter weitergereicht werden konnten

Ich hoffe meine Interpretierung ist korrekt und ich konnte bei der Problematik behilflich sein.

[image]
[image]

reserve85, 2024-09-13:
Ja ich glaube trotzdem es ist nicht die Lösung… könntest du morgen mal probieren was passiert, wenn du den inverter aus schaltest? Also über OpenDTU. Dann müsste producing auf false gehen und für mich wäre dann der inverter nicht erreichbar. 

Normal müsste die queue einfach intern in der dtu gelöscht werden wenn nix übertragen werden kann. Und reachable auf false gehen…

martin-st-81, 2024-09-15:
Wenn ich zB Hoymiles1 manuell via OpenDTU abschalte kann dein Script selbst nach einem Neustart des Scriptes diesen nicht wieder einschalten. Für mich würde das Script - da ich keinen Speicher verwende - jetzt funktionieren, in der Früh hatte ich keine Ausfälle mehr. Da dieses Script laut Doku auf AhoyDTU konzipiert ist und ich den Fehler auch bei OpenDTU sehe, werde ich vermutlich auf AhoyDTU (wenn es meine HW zulässt) umsteigen. 

[image]

[image]

[image]

reserve85, 2024-09-15:
Ja, das hab ich auch so befürchtet. Das Flag reachable ist für mich das Problem, das dürfte nicht true sein wenn der inverter nicht erreichbar ist. 

Ich nehme die Änderung in der dev wieder zurück.

martin-st-81, 2024-09-15:
Welche AhoyDTU Version kannst du empfehlen oder verwendest du ?

reserve85, 2024-09-15:
Ich hab die 0.8.143

reserve85, 2024-09-15:
Vielleicht könntest du das reachable Problem mal bei OpenDTU als bug melden. Wäre super!

reserve85, 2024-09-15:
Aja, da gibt’s schon einige Meldungen.

https://github.com/tbnobody/OpenDTU/issues/2237

closed 2024-09-16 by reserve85