reserve85/HoymilesZeroExport #243: 800 Watt max für 3 Wechselrichter einstellen

fatzo, 2024-09-16:
Moin zusammen
Wo kann ich das lmit für alle wechselrichter zusammen einstellen?

# if your powermeter exceeds POWERMETER_MAX_POINT: immediatelly set the limit to predefined percent of HOY_MAX_WATT (if you have more than one inverter it´s the sum of all HOY_MAX_WATT)
# value = 0 disables the feature. Values are possible from [0 to 100]
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 100

bei 2 wechselrichter 800 und 600 - 1400 watt
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 55 ( cirka)

ist meine Überlegung richtig?

gruß fatzo

RealNBB, 2024-09-17:
Man kann nicht das Limit aller Wechselrichter einstellen, sondern das ergibt sich automatisch aus der Summe aller Werte von HOY_MAX_WATT.

ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT sagt ja nur aus, dass das Skript - sobald der Wert von POWERMETER_MAX_POINT erreicht oder überschritten wurde - es auf den fest gelegten Prozentwert direkt springt.

In deinem Fall also z.B.:

POWERMETER_MAX_POINT = 600
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 50

Hast du nun einen Verbrauch von 800W (also über 600W max Point), springt das Skript direkt auf 700W (50% von 1400W) und regelt dann weiter hoch auf 800W.

Hättest du POWERMETER_MAX_POINT = 1000 und es liegen 800W an, regelt das Skript gemütlich hoch und ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT wird nicht beachtet.

fatzo, 2024-09-17:
danke

reserve85, 2026-11-11:
Die Frage ist oben beantwortet, deshalb schließe ich das Issue. Falls noch etwas unklar ist, gerne ein neues aufmachen.

closed 2026-11-11 by reserve85