reserve85/HoymilesZeroExport #245: Nulleinspeisung

NewSmokie, 2024-09-22:
Hallo ich habe es geschafft dieses Script für die nulleinspeisung über opendtu und hoymiles. Alles super.

nun möchte ich aber die überschüssige Energie die ich noch habe auf Akkus zurückführen.

Meine Frage ist nun wie ist die Verkabelung und was muss ich im Skript ändern.

meine Hardware ist:
Hoymiles 2000 4T
Opendtu
Homessistant 
PI3 darauf läuft der Zero Script ( habe es unter Home Assistant nicht hinbekommen)
200ah -24V Akkus

RealNBB, 2024-09-23:
na im skript die zeile "HOY_BATTERY_MODE = false" auf true ändern und die Spannungen gemäß deiner Batterie/Vorstellungen anpassen.

Wie meinst du das, wie die Verkabelung ist? Wie man generell eine Batterie anschließt?

NewSmokie, 2024-09-23:
Ich meine von wemm werden die Batterien eingespeist von dem wechselrichter? Panels? Laderegler?

robson1311, 2024-09-24:
Die Akkus werden bei mir vom Überschuss über den Laderegler (+Panels) gespeist.

NewSmokie, 2024-09-26:
Moin genau das meine ich wie ist die Verkabelung und welchen Laderegler brauche ich

RealNBB, 2024-09-26:
Deine Fragen sind freundlich ausgedrückt mehr als lustig.

Welchen Laderegler du brauchst? Einen, der zu deiner Batterie und installierte PV-Anlage passt.
Wie alles verkabelt wird? Ernsthaft jetzt?

Möchtest du dich nicht vllt selbst mal ein wenig in das Thema einlesen, anstatt hier Grundlage-Fragen zu stellen, die im Netz im übrigen schon tausendfach beantwortet sind?
Ich finde sowas nur frech.

Also…bisschen Selbstinitiative hat noch keinem geschadet. Insofern viel Spaß beim Lesen:

https://github.com/helgeerbe/OpenDTU-OnBattery/wiki

https://meintechblog.de/2023/08/07/operation-hausspeicher-das-kosten-die-benoetigten-werkzeuge/
https://meintechblog.de/2022/12/13/operation-hausspeicher-lifepo4-zellen-und-die-richtigen-spannungseinstellungen/
https://meintechblog.de/2022/09/13/operation-hausspeicher-stueckliste-und-bezugsquellen/

NewSmokie, 2024-09-26:
Sorry wollte keinen auf die Füße treten. Gerne lese ich mir was durch habe mir auch schon Gedanken gemacht. Leider habe keine Beispiele gefunden. Die so ähnlich sind wie ich bei mir umsetzen soll.

mi-hol, 2024-11-11:
@

> Möchtest du dich nicht vllt selbst mal ein wenig in das Thema einlesen, anstatt hier Grundlage-Fragen zu stellen, die im Netz im übrigen schon tausendfach beantwortet sind?
> Ich finde sowas nur frech.

nun ja, für einen Experten auf dem Gebiet ist das alles klar, aber eine kleine Hilfestellung im ReadMe des Projekts wäre da eine gewaltige Verbesserung für Neueinsteiger.

@reserve85 wären die im [Kommentar genannten URLs](https://github.com/reserve85/HoymilesZeroExport/issues/245#issuecomment-2376270151) ein guter Startpunkt für Neueinsteiger oder gibt es noch was besseres?

RealNBB, 2024-11-11:
wie soll man deinen Kommentar verstehen? das wir/ich direkt als Zero Export Experte geboren wurde und ich es daher nicht nötig hatte, mich in die Materie einzulesen?

ich kann dir versichern, dass das nicht so war. ;)

mi-hol, 2024-11-11:
> Eine Bastelempfehlung geb ich definitiv nicht ab,
> ich stelle nur das Script zur Verfügung, das sollte in diesem Projekt hier hoffentlich reichen :-)

Das kann man dann ja auch im ReadMe klar aussprechen. Momentan wird das halt nicht ausgesprochen und lässt damit Raum für (unbeabsichtigte) Auslegung und Anforderungen in alle Richtungen.

Diese Konversation zeigt mMn ein Mißvertändnis zwischen Unerfahrenen und Experten. 
Ich war beruflich ein "Brückenbauer"/"Übersetzer" zwischen diesen beiden Gruppen. 
Ich stelle einen PR fürs ReadMe, weitere Tipps sind gern willkomen.

tomquist, 2024-11-11:
Ich versteh ehrlich gesagt nicht so richtig an welcher Stelle du überhaupt Hilfe benötigst. Ich glaube der Grund warum manche hier so reagieren ist, dass deine Frage nach der Verkabelung den Eindruck erweckt, dass es noch an den absoluten Grundlagen von Heimspeichern fehlt.

Die Frage nach der Verkabelung ist ungefähr so als würdest du beim Windows Support anrufen und erzählen, dass du ein jetzt Mainboard und ein Netzteil gekauft hast und gerne wissen möchtest, wie du die beiden Dinge verbindest, damit du Windows installieren kannst.

Um ein bisschen auf deine Frage einzugehen, deine Hardwareliste scheint mir unvollständig zu sein. Zumindest fehlt dort meines Erachtens nach das BMS oder, ein Laderegler oder ein Ladegerät. Wie kommst du denn an die Akkus? Du solltest wissen, dass die Teile echt gefählich sind, wenn du nicht weißt wie man damit richtig umgeht.

mi-hol, 2024-11-11:
> Akkus? Du solltest wissen, dass die Teile echt gefählich sind, wenn du nicht weißt wie man damit richtig umgeht.

@reserve85 @tomquist deshalb würde ich empfehlen diesen issue nicht weiter zu kommentieren und ihn stattdessen zu schließen

mi-hol, 2024-11-11:
please see https://github.com/reserve85/HoymilesZeroExport/issues/259

NewSmokie, 2024-11-11:
Hi,

Ich wollte nur ein paar Fragen stellen. Das das hier so ausartet wollte ich nicht.

Es muss jedem klar sein wenn man Akkus ob 12v 24v oder 48v nimmt und die verkabelt das es da eine großes Risiko gibt wenn man sich nicht auskennt. 

Explosionsgefahr oder das man mit der Power von DC Gleichspannung sehr schnell sich verletzen kann.

closed 2024-11-15 by reserve85