reserve85/HoymilesZeroExport #247: WR (HM1200) startet auch bei Unterspannung

robson1311, 2024-10-06:
Hallo Leute!
Hab da seit kurzem ein Problem und zwar produziert der WR teils auch wenn er nicht soll.
Meine Komponenten:
1x Akku (max. 49,3 Volt), 2x Victron LR 100/20, 1x HM1200, OpenDtu 24.10.2 und das neueste ZeroExport Script.

Hab im Script eingestellt, dass wenn der Akku unter 39 Volt fällt, der WR ausschalten soll und erst wenn er wieder auf 42,5 Volt ist, er wieder produzieren soll.
Jetzt dürfte vor ein paar Tagen das so nicht funktioniert haben, der WR hat vermutlich zu früh gestartet, die Spannung ist runter unter 36 V und das BMS hat ausgelöst. Die Batterie sich wieder erfangen, BMS wieder freigegeben und das selbe Spiel von vorne.
Das dürfte dann so irgendwie, ohne das ich es zunächst merkte, öfters passiert sein und einer der beiden Victron hat das dann nicht überlebt.
Heute in der Früh um ca. 7.30 Uhr, startete lt. Ereignisanzeige der OpenDtu, obwoh er ja nicht dürfte, wieder der WR und produzierte 1200 Watt bei gerade mal 38 Volt und riss die Spannung der Batterie dann logischerweise wieder weit nach unten.
Der noch intakte Victron fiel aus, er produzierte nicht mehr, kein Bluetooth Zugriff möglich, und erst nachdem ich dem LR spannunglos machte, lief dieser wieder, und war Gottseidank nicht defekt.
Allerdings schaltete sich der WR trotz Unterspannung schon wieder ein, den ich dann aber per OpenDTU ausschaltete.
Momentan läuft es wie es soll, der eine LR lädt den Akku, Spannung allerdings noch unter den eingestellten 42,5 Volt, und es wird noch nichts produziert.
Ich hoffe ich konnte das so halbwegs verständlich erklären, ich hänge noch 2 Logs und die override.ini zur Kontrolle mit an und hoffe auf eure Hilfe.
Vielen Dank!
[log1.txt](https://github.com/user-attachments/files/17268710/log1.txt)
[log2.txt](https://github.com/user-attachments/files/17268711/log2.txt)
[override.ini.txt](https://github.com/user-attachments/files/17268713/override.ini.txt)

robson1311, 2024-10-06:
Leider wurde den ganzen Tag über nichts eingespeist, obwohl die Batterie über die 42.5 Volt geladen wurde.
Es wurden lediglich nur die 120 Watt, also 10% eingespeist, im reduzierten Betrieb.
Hab das Script nochmals neugestartet und bekomme zuerts einige Fehlermeldungen, wo ich aber nicht weiß wie ich diese deuten soll.
Hänge nochmals zwei Logs mit an, vielleicht könnte mir da jemand drüber schauen.
Vielen Dank!
[log3.txt](https://github.com/user-attachments/files/17270138/log3.txt)
[log4.txt](https://github.com/user-attachments/files/17270139/log4.txt)

reserve85, 2024-10-06:
Ich bin gerade unterwegs, aber es kommen Fehler beim ein/ausschalten des inverters:

```
Oct 06 17:47:29 Nulleinspeisung python3[393]: 2024-10-06 17:47:29 INFO     OpenDTU: Inverter "HM-1200": Turn on
Oct 06 17:47:29 Nulleinspeisung python3[393]: 2024-10-06 17:47:29 ERROR    Exception at SetHoymilesPowerStatus
Oct 06 17:47:29 Nulleinspeisung python3[393]: 2024-10-06 17:47:29 ERROR    Error: SetPowerStatus error: Values are missing!
Oct 06 17:47:44 Nulleinspeisung python3[393]: 2024-10-06 17:47:44 INFO     ---Start Zero Export---
Oct 06 17:47:44 Nulleinspeisung python3[393]: 2024-10-06 17:47:44 INFO     Warning: POWERMETER_MAX_POINT < POWERMETER_TARGET_POINT + POWERMETER_TOLERANCE. Setting POWERMETER_MAX_POINT to 75
Oct 06 17:47:44 Nulleinspeisung python3[393]: 2024-10-06 17:47:44 INFO     OpenDTU: Inverter "HM-1200" reachable: True
Oct 06 17:47:44 Nulleinspeisung python3[393]: 2024-10-06 17:47:44 INFO     Average min-panel voltage, inverter "HM-1200": 43.1 Volt
Oct 06 17:47:44 Nulleinspeisung python3[393]: 2024-10-06 17:47:44 INFO     OpenDTU: Inverter "HM-1200": Turn on
Oct 06 17:47:45 Nulleinspeisung python3[393]: 2024-10-06 17:47:45 ERROR    Exception at SetHoymilesPowerStatus
Oct 06 17:47:45 Nulleinspeisung python3[393]: 2024-10-06 17:47:45 ERROR    Exception at CheckBattery, Inverter 0 not reachable
```

hast du die OpenDTU aktualisiert? Ging das vorher? Evtl. hat sich was an der API geändert…

robson1311, 2024-10-06:
OpenDTU habe ich gestern erst geupdatet, allerdings hat es das Prob mit dem anlaufen unter dem Wert bei HOY_BATTERY_THRESHOLD_OFF_LIMIT schon gegeben.
Vor einer Stunde ist ne neue FW für OpenDTU rausgekommen, habe die jetzt mal aufgespielt.
Hier der aktuelle Log:
[log5.txt](https://github.com/user-attachments/files/17270602/log5.txt)

reserve85, 2024-10-06:
Welche Version genau?

robson1311, 2024-10-06:
Zurück auf die 24.5.6

reserve85, 2024-10-07:
ja, da wurde was geändert:

[image]

Es wird nun in der Firmware auf den Übergabe-Typ string und bool geprüft (macht auch sinn). 
Gesendet wird als "bool"-Wert von mir ein Integer-Value 1 oder 0. 
Ich denke da fehlt ein cast to bool in der OpenDTU? 
Oder muss ich es auch als string in Anführungszeichen senden? @tbonobody

tbnobody, 2024-10-07:
0 und 1 ist laut json Spezifikation kein Bool. True und false sind boolean values

reserve85, 2024-10-07:
besten Dank, werde ich anpassen!

reserve85, 2024-10-07:
@robson1311 ich hab das in der dev (hoffentlich richtig) angepasst, kannst du das ggf. testen? 
Wenn möglich erst mit deiner "alten" Firmware (OpenDTU 24.5.6) und wenn das klappt auch mit der aktuellen Firmware.

Gruß

robson1311, 2024-10-07:
Es scheint auch bei der v24.5.6 nicht richtig zu funktionieren, denn es wird weiterhin, obwohl die Batterie über den eingestellten 42,5 V ist, weiterhin nur die 120 Watt eingespeist.
Die dev finde ich wo, bzw. kann ich die dann per update.sh drüber installieren?
Danke!

Hier noch ein Log:
[log6.txt](https://github.com/user-attachments/files/17277353/log6.txt)

reserve85, 2024-10-07:
`sudo ./update.sh dev`
-> damit kannst du die neue installieren.

du hast 
`HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 44.3`
Volt definiert, daher bleibt der Inverter auch bei 120W bis die 44,3V erreicht sind (erst dann schaltet er auf max. 1200W um)

robson1311, 2024-10-07:
Super, danke! Werde ich dann Abends testen und berichten
LG

robson1311, 2024-10-07:
Die dev eingespielt, doch leider wieder Fehler:
`Oct 07 19:46:26 Nulleinspeisung python3[1460]: 2024-10-07 19:46:26 INFO     Log write to file: True
Oct 07 19:46:26 Nulleinspeisung python3[1460]: 2024-10-07 19:46:26 INFO     Python Version: 3.10.7 (main, Mar 10 2023, 10:47:39) [GCC 12.2.0]
Oct 07 19:46:26 Nulleinspeisung python3[1460]: 2024-10-07 19:46:26 INFO     Author: Tobias Kraft / Script Version: 1.104
Oct 07 19:46:26 Nulleinspeisung python3[1460]: 2024-10-07 19:46:26 INFO     read config file: /root/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Oct 07 19:46:26 Nulleinspeisung python3[1460]: 2024-10-07 19:46:26 INFO     read additional config file: /root/HoymilesZeroExport/HoymilesZeroExport_Config_Override.ini
Oct 07 19:46:26 Nulleinspeisung python3[1460]: 2024-10-07 19:46:26 INFO     Config file V 1.102
Oct 07 19:46:26 Nulleinspeisung python3[1460]: 2024-10-07 19:46:26 INFO     ---Init---
Oct 07 19:46:27 Nulleinspeisung python3[1460]: 2024-10-07 19:46:27 INFO     OpenDTU: Current Version: v24.5.6
Oct 07 19:46:27 Nulleinspeisung python3[1460]: 2024-10-07 19:46:27 INFO     OpenDTU: Inverter "yet unknown" reachable: True
Oct 07 19:46:27 Nulleinspeisung python3[1460]: 2024-10-07 19:46:27 INFO     OpenDTU: Inverter "HM-1200" / serial number "116171605625" / temperature 21.3 degC
Oct 07 19:46:27 Nulleinspeisung python3[1460]: 2024-10-07 19:46:27 INFO     OpenDTU: Inverter "HM-1200": Turn on
Oct 07 19:46:27 Nulleinspeisung python3[1460]: 2024-10-07 19:46:27 ERROR    Exception at SetHoymilesPowerStatus
Oct 07 19:46:27 Nulleinspeisung python3[1460]: 2024-10-07 19:46:27 ERROR    Error: SetPowerStatus error: Failed to parse data!
Oct 07 19:46:42 Nulleinspeisung python3[1460]: 2024-10-07 19:46:42 INFO     ---Start Zero Export---
Oct 07 19:46:42 Nulleinspeisung python3[1460]: 2024-10-07 19:46:42 INFO     Warning: POWERMETER_MAX_POINT < POWERMETER_TARGET_POINT + POWERMETER_TOLERANCE. Setting POWERMETER_MAX_POINT to 75
Oct 07 19:46:42 Nulleinspeisung python3[1460]: 2024-10-07 19:46:42 INFO     OpenDTU: Inverter "HM-1200" reachable: True
Oct 07 19:46:42 Nulleinspeisung python3[1460]: 2024-10-07 19:46:42 INFO     Average min-panel voltage, inverter "HM-1200": 42.40000153 Volt
Oct 07 19:46:42 Nulleinspeisung python3[1460]: 2024-10-07 19:46:42 INFO     powermeter ShellyEM: 78 Watt
Oct 07 19:46:42 Nulleinspeisung python3[1460]: 2024-10-07 19:46:42 INFO     setting new limit to 73 Watt
Oct 07 19:46:42 Nulleinspeisung python3[1460]: 2024-10-07 19:46:42 INFO     OpenDTU: Inverter "HM-1200": setting new limit from -1 Watt to 73 Watt
Oct 07 19:46:44 Nulleinspeisung python3[1460]: 2024-10-07 19:46:44 INFO     OpenDTU: Inverter "HM-1200": Limit acknowledged
Oct 07 19:46:59 Nulleinspeisung python3[1460]: 2024-10-07 19:46:59 INFO     Warning: POWERMETER_MAX_POINT < POWERMETER_TARGET_POINT + POWERMETER_TOLERANCE. Setting POWERMETER_MAX_POINT to 75
Oct 07 19:46:59 Nulleinspeisung python3[1460]: 2024-10-07 19:46:59 INFO     OpenDTU: Inverter "HM-1200" reachable: True
Oct 07 19:47:00 Nulleinspeisung python3[1460]: 2024-10-07 19:47:00 INFO     Average min-panel voltage, inverter "HM-1200": 42.5 Volt
Oct 07 19:47:00 Nulleinspeisung python3[1460]: 2024-10-07 19:47:00 INFO     OpenDTU: Inverter "HM-1200": Turn on
Oct 07 19:47:00 Nulleinspeisung python3[1460]: 2024-10-07 19:47:00 ERROR    Exception at SetHoymilesPowerStatus
Oct 07 19:47:00 Nulleinspeisung python3[1460]: 2024-10-07 19:47:00 ERROR    Exception at CheckBattery, Inverter 0 not reachable
Oct 07 19:47:15 Nulleinspeisung python3[1460]: 2024-10-07 19:47:15 INFO     Warning: POWERMETER_MAX_POINT < POWERMETER_TARGET_POINT + POWERMETER_TOLERANCE. Setting POWERMETER_MAX_POINT to 75
Oct 07 19:47:15 Nulleinspeisung python3[1460]: 2024-10-07 19:47:15 INFO     OpenDTU: Inverter "HM-1200" reachable: True
Oct 07 19:47:15 Nulleinspeisung python3[1460]: 2024-10-07 19:47:15 INFO     Average min-panel voltage, inverter "HM-1200": 42.70000076 Volt
Oct 07 19:47:15 Nulleinspeisung python3[1460]: 2024-10-07 19:47:15 INFO     OpenDTU: Inverter "HM-1200": Turn on
Oct 07 19:47:15 Nulleinspeisung python3[1460]: 2024-10-07 19:47:15 ERROR    Exception at SetHoymilesPowerStatus
Oct 07 19:47:15 Nulleinspeisung python3[1460]: 2024-10-07 19:47:15 ERROR    Exception at CheckBattery, Inverter 0 not reachable
`

reserve85, 2024-10-07:
habe es nochmal angepasst, kannst du es nochmal testen?

robson1311, 2024-10-08:
Sorry, erst jetzt dazu gekommen.
Leider noch eine Fehlermeldung:
[log7.txt](https://github.com/user-attachments/files/17296712/log7.txt)

reserve85, 2024-10-08:
eieiei, ich hab eine Zeile vom debuggen leider übersehen und mit eingecheckt... jetzt sollte es passen.
Sorry, aber wir kriegen es schon irgendwie noch hin :-)

robson1311, 2024-10-08:
Leider nein :-)
[log8.txt](https://github.com/user-attachments/files/17298163/log8.txt)

reserve85, 2024-10-08:
Ich brech ab, ich muss morgen nach der Arbeit eine OpenDTU aufsetzen und teste es dann... sorry

die txt kann ich leider nicht anschauen? kommt ein Fehler beim öffnen...

reserve85, 2024-10-08:
gesagt getan: fehler gefunden...

bitte noch ein mal probieren.

robson1311, 2024-10-08:
Schaut sehr gut aus!
Läuft jetzt auch mit der aktuellen OpenDTU ohne Fehlermeldung.
Tipitopi! 👍 
Vielen Dank!

closed 2024-10-09 by reserve85