reserve85/HoymilesZeroExport #248: Max und Min Einspeisung Jump

filou01, 2024-10-06:
Hallo. 

Ich hab die Vermutung, dass bei Überschreitung des  POWERMETER_MAX_POINT des Jump (ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 100) sofort ausgeführt wird, und nicht erst beim nächsten LOOP_INTERVAL. Kann dies so stimmen?

Im umgekehrten Fall, wo durch zu wenig Eigenverbrauch der 
POWERMETER_MIN_POINT unterschritten wird, scheint es aber ob der Jump (ON_GRID_FEED_FAST_LIMIT_DECREASE) erst beim nächsten LOOP_INTERVAL ausgeführt wird und somit in meinem Fall eine Überschreitung der zulässigen 800W Einspeisung für 20 Sekunden möglich sind. 

Sehe ich das richtig? 

Danke und Lg

reserve85, 2024-10-07:
Ne, das ist in der gleichen Schleife und wird ebenfalls sofort ausgeführt (identisch zu on_grid_usage_jump_to_limit_percent).

Allerdings ist innerhalb der 20s (LOOP_INTERVAL_IN_SECONDS) nur eine Operation möglich.

filou01, 2024-10-19:
Danke für die Erklärung

closed 2024-10-19 by filou01