reserve85/HoymilesZeroExport #250: Feature Request - MQTT Auto-Discovery

Bend3smo, 2024-10-10:
Moin, 
es wäre super, wenn Home Assistant die MQTT Entitäten automatisch anlegen und auch die aktuellen Einstelltungen auslesen könnte.

Eine Reglung abhängig vom Tibber Strompreis wäre damit zum Beispiel sehr einfach realisierbar.

Bend3smo, 2024-10-12:
?

tomquist, 2024-10-12:
Der Feature Request macht schon Sinn. Über die OpenDTU bekommt man nur die aktuell eingestellten IST-Werte, kann aber das ZeroExport script nicht steuern. HoymilesZeroExport pushed die SOLL-Werte in den MQTT. Außerdem gibt HoymilesZeroExport noch das gesamt limit aus, was relevant ist, wenn man mehrere Wechselrichter hat. Darüber hinaus macht es auch Sinn, wenn per Auto-Discovery die Entitäten für die `/set/*` topics angelegt werden, damit man darüber das Script steuern kann.

Ich kenne mich mit HomeAssistant nicht gut aus, aber wenn jemand eine passende HomeAssistant config zur Verfügung stellt, kann ich gerne damit Auto-Discovery einbauen.

Bend3smo, 2024-10-13:
Kannst du mal eine Liste aller Topics posten?
Besonders das Topic für das Gesamte Limit würde mich interessieren.

Bis jetzt zeigt mir mein MQTT Explorer in HomeAssistant noch nichts von ZeroExport an.

Ich hab in einem anderen Thread gelesen, dass das daran liegt, das ZeroExport nur liest ist das richtig?
Könnte man das nicht ändern, das alle Topics an den MQTT Broker veröffentlicht werden?

Kenne mich leider mit MQTT nicht so gut aus.

RealNBB, 2024-10-15:
Mach doch einfach "deintopic/#". Dann zeigt MQTT alles an, was rein kommt.

Ich hab es übrigen aber auch noch nicht geschafft, MQTT von ZeroExport zu empfangen. In meinem Fall denke ich aber, dass es eher daran liegt, dass mein HAOS als Container läuft (genauso wie ZeroExport) und es deshalb nicht wirklich mit dem im HAOS hinterlegten Broker klappt.

Bend3smo, 2024-10-15:
Funktioniert leider auch nicht. Weder wenn das ZeroExport Script als docker Container auf einem NAS läuft, noch wenn es per portainer als Docker auf einem NUC mit HomeAssistantOS läuft.

Bend3smo, 2024-10-17:
Ich muss mich entschuldigen mit nem vernünftigen Editor sieht man den Fehler sofort. 
[image]
Jetzt sehe ich die Topics im MQTT Explorer

reserve85, 2026-11-11:
Das ist jetzt auf dem dev-Branch umgesetzt (V 1.107). Mit

`MQTT_HOMEASSISTANT_DISCOVERY = true` (Abschnitt [MQTT_CONFIG])

legt Home Assistant ein Gerät "HoymilesZeroExport" an, mit Entitäten für die Einstellungen (Zielwert, Max-/Min-Punkt, Toleranz usw.), das aktuelle Limit gesamt und je Wechselrichter sowie einen Ein/Aus-Schalter je Wechselrichter. Der Präfix lässt sich mit MQTT_HOMEASSISTANT_DISCOVERY_PREFIX ändern (Standard: homeassistant).

Zur Frage von damals: Das gesamte Limit steht in `zeropower/state/limit`.

Test-Image: ghcr.io/reserve85/hoymileszeroexport:dev. Rückmeldung, ob alle Entitäten sauber erscheinen, wäre hilfreich.