reserve85/HoymilesZeroExport #251: How to run script for newbies ?

toutou311, 2024-10-16:
Hello,

Sorry for being such a noob but I am struggleing to launch the script

I want to use this script in parrallel of M'sunPV boiler router
The goal is of course to avoid export

The msun pv is my grid power clamp
I have an UpenDTU linked to my hoymiles inverters

I can launch Virtual machine
Mainly using Home Assistant for Everything

Someone can tell me how to run the script ?

Thanks in advance, and sorry for being such a rookie

mi-hol, 2024-11-08:
> how to run the script ?

Intent of readme is to explain that. In case issues arise, more details are required.
Please see https://opensource.com/life/16/10/how-ask-technical-questions

reserve85, 2026-11-11:
This question has been answered above, so I'm closing the issue. Feel free to open a new one if something is still unclear.

closed 2026-11-11 by reserve85