reserve85/HoymilesZeroExport #252: Integration in HA möglich?

Gyr0sK0mplett, 2024-10-30:
Hallo zusammen. Ja, es ist kein Fehler, entschuldigt bitte.
Hat jemand dieses Script in HA integriert? Für mich wäre das echt ein Sahneäubchen, aber da stoße ich erstmal an Grenzen.
Hat das womöglich schon jemand gemacht und könnte mir helfen?

bootix, 2024-12-19:
Bei mir läuft das in Portainer auf einem Raspi5 und HA OS stabil.
Ohne die 512M Speicher bricht es nach ein paar Stunden ab.

services:
  hoymileszeroexport:
    image: ghcr.io/reserve85/hoymileszeroexport:main
    restart: unless-stopped
    deploy:
      resources:
        limits:
          memory: 512M
        reservations:
          memory: 512M
    volumes:
      - /mnt/data/supervisor/homeassistant/HoymilesZeroExport:/data

    command: -c /data/HoymilesZeroExport_Config_Override.ini

Gyr0sK0mplett, 2024-12-26:
Danke dir. Mittlerweile regelt OpenDTU oB mehrere WR dynamisch, deshalb hat sich das hier erledigt.