reserve85/HoymilesZeroExport #253: inverter yet unknown

trombose009, 2024-11-05:
Sorry, 
eigentlich sagt die Log-Zeile ja schon recht präzise was los ist. Ich weiß aber nicht WARUM der inverter unknown sein soll. Er läuft in der AHOY DTU auf online. Er ist in der Override-Datei drin und ich hab den ganzen docker container gestoppt, gelöscht, das unused image auch gelöscht und alles neu ziehen lassen. 
Was mache ich falsch?

`2024-11-05 18:04:19 INFO     MQTT: Incoming power: 393 Watt

2024-11-05 18:04:21 INFO     Ahoy: Inverter "yet unknown" Available: False

2024-11-05 18:04:39 INFO     MQTT: Incoming power: 387 Watt

2024-11-05 18:04:41 INFO     Ahoy: Inverter "yet unknown" Available: False

2024-11-05 18:04:59 INFO     MQTT: Incoming power: 388 Watt

2024-11-05 18:05:01 INFO     Ahoy: Inverter "yet unknown" Available: False

2024-11-05 18:05:19 INFO     MQTT: Incoming power: 393 Watt

2024-11-05 18:05:21 INFO     Ahoy: Inverter "yet unknown" Available: False`

trombose009, 2024-11-05:
Ok, ich habe jetzt herausgefunden, dass es funktioniert, wenn ich die Reihenfolge der Inverter in der DTU umdrehe. Also Irgendwo in der Override Datei muss wohl noch definiert werden, welcher der zwei inverter gemeint ist. die seriennummer reicht wohl allein nicht aus. naja, egal für mich. hab jetzt ja erstmal durch das verdrehen der reihenfolge gelöst. 

liebe grüße

reserve85, 2024-11-06:
ja, wenn du in der override nur einen Inverter angibst, dann holt er sich auch nur den "ersten" Inverter von der AhoyDTU. Das ist auch Seriennummern unabhängig, die wird von AHOY abgefragt. Lediglich bei OpenDTU ist die notwendig.

Besser du trägst daher beide Inverter in der Override ein und setzt den Ersten auf "ENABLED = false".
Reihenfolge ist die aus der AHOY Inverterliste.

reserve85, 2026-11-11:
Die Frage ist oben beantwortet, deshalb schließe ich das Issue. Falls noch etwas unklar ist, gerne ein neues aufmachen.