reserve85/HoymilesZeroExport #260: Fehler OpenDtu - Inverter yet unkown

SK-hoy, 2024-12-15:
Nach Eingabe aller aus meiner Sicht notwendigen Angaben in das Skript erfolgt ein Fehler nach Ausführung des Skripts.
OpenDTU: Inverter "yet unknown" reachable: True
Exception at CheckBattery, Inverter 0 not reachable

Es ist der Version 1.46, versuche ich es mit der neusten Version erhalte ich Fehler bei der Installation.

Es ist ein Hoymiles 1500 , Tasmota wird korrekt ausgelesen und OpenDtu ist über den Browser zu erreichen, kann mir jemand dazu Hinweise geben?

reserve85, 2024-12-16:
Hi,

du musst die Script-Version updaten. Die API der DTU hatte sich mittlerweile mehrfach geändert und wird daher mit der alten Script-Version nicht mehr funktionieren.

> versuche ich es mit der neusten Version erhalte ich Fehler bei der Installation.
diese wären wichtig um dir zu helfen.

SK-hoy, 2024-12-16:
Ich hatte ursprünglich die 102 (Script) und 104 (PY) verwendet: Hierbei erfolgt folgende Fehlermeldung:

Dez 16 10:57:48 raspberrypi systemd[1]: Started HoymilesZeroExport Service.
Dez 16 10:57:48 raspberrypi python3[865]: Traceback (most recent call last):
Dez 16 10:57:48 raspberrypi python3[865]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 32, in <module>
Dez 16 10:57:48 raspberrypi python3[865]:     from packaging import version
Dez 16 10:57:48 raspberrypi python3[865]: ModuleNotFoundError: No module named 'packaging'
Dez 16 10:57:48 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Dez 16 10:57:48 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Dez 16 10:57:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Service RestartSec=100ms expired, scheduling restart.
Dez 16 10:57:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 6.
Dez 16 10:57:49 raspberrypi systemd[1]: Stopped HoymilesZeroExport Service.
Dez 16 10:57:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly.
Dez 16 10:57:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Dez 16 10:57:49 raspberrypi systemd[1]: Failed to start HoymilesZeroExport Service.

Hab bei der Installation schon den Packaging Befehl erhalten, daher mit einer älteren Variante probiert, die lief bis auf OpenDtu

reserve85, 2024-12-16:
https://github.com/reserve85/HoymilesZeroExport/issues/137#issuecomment-1942654871

SK-hoy, 2024-12-16:
beides leider ergebnislos, leider noch Fehler:

Dez 16 12:51:48 raspberrypi systemd[1]: Started HoymilesZeroExport Service.
Dez 16 12:51:48 raspberrypi python3[1591]: Traceback (most recent call last):
Dez 16 12:51:48 raspberrypi python3[1591]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 21, in <module>
Dez 16 12:51:48 raspberrypi python3[1591]:     from requests.sessions import Session
Dez 16 12:51:48 raspberrypi python3[1591]: ModuleNotFoundError: No module named 'requests'
Dez 16 12:51:48 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Dez 16 12:51:48 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Service RestartSec=100ms expired, scheduling restart.
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 1.
Dez 16 12:51:49 raspberrypi systemd[1]: Stopped HoymilesZeroExport Service.
Dez 16 12:51:49 raspberrypi systemd[1]: Started HoymilesZeroExport Service.
Dez 16 12:51:49 raspberrypi python3[1592]: Traceback (most recent call last):
Dez 16 12:51:49 raspberrypi python3[1592]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 21, in <module>
Dez 16 12:51:49 raspberrypi python3[1592]:     from requests.sessions import Session
Dez 16 12:51:49 raspberrypi python3[1592]: ModuleNotFoundError: No module named 'requests'
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Service RestartSec=100ms expired, scheduling restart.
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 2.
Dez 16 12:51:49 raspberrypi systemd[1]: Stopped HoymilesZeroExport Service.
Dez 16 12:51:49 raspberrypi systemd[1]: Started HoymilesZeroExport Service.
Dez 16 12:51:49 raspberrypi python3[1593]: Traceback (most recent call last):
Dez 16 12:51:49 raspberrypi python3[1593]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 21, in <module>
Dez 16 12:51:49 raspberrypi python3[1593]:     from requests.sessions import Session
Dez 16 12:51:49 raspberrypi python3[1593]: ModuleNotFoundError: No module named 'requests'
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Service RestartSec=100ms expired, scheduling restart.
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 3.
Dez 16 12:51:49 raspberrypi systemd[1]: Stopped HoymilesZeroExport Service.
Dez 16 12:51:49 raspberrypi systemd[1]: Started HoymilesZeroExport Service.
Dez 16 12:51:49 raspberrypi python3[1595]: Traceback (most recent call last):
Dez 16 12:51:49 raspberrypi python3[1595]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 21, in <module>
Dez 16 12:51:49 raspberrypi python3[1595]:     from requests.sessions import Session
Dez 16 12:51:49 raspberrypi python3[1595]: ModuleNotFoundError: No module named 'requests'
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Service RestartSec=100ms expired, scheduling restart.
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 4.
Dez 16 12:51:49 raspberrypi systemd[1]: Stopped HoymilesZeroExport Service.
Dez 16 12:51:49 raspberrypi systemd[1]: Started HoymilesZeroExport Service.
Dez 16 12:51:49 raspberrypi python3[1596]: Traceback (most recent call last):
Dez 16 12:51:49 raspberrypi python3[1596]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 21, in <module>
Dez 16 12:51:49 raspberrypi python3[1596]:     from requests.sessions import Session
Dez 16 12:51:49 raspberrypi python3[1596]: ModuleNotFoundError: No module named 'requests'
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Dez 16 12:51:49 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Dez 16 12:51:50 raspberrypi systemd[1]: HoymilesZeroExport.service: Service RestartSec=100ms expired, scheduling restart.
Dez 16 12:51:50 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 5.
Dez 16 12:51:50 raspberrypi systemd[1]: Stopped HoymilesZeroExport Service.
Dez 16 12:51:50 raspberrypi systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly.
Dez 16 12:51:50 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Dez 16 12:51:50 raspberrypi systemd[1]: Failed to start HoymilesZeroExport Service.

reserve85, 2024-12-16:
was kommt denn für eine Fehlermeldung bzw. Logausgabe hier:

`pip install -r requirements.txt --break-system-packages`

SK-hoy, 2024-12-16:
Defaulting to user installation because normal site-packages is not writeable
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
ERROR: Could not open requirements file: [Errno 2] Datei oder Verzeichnis nicht gefunden: 'requirements.txt'
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
Could not fetch URL https://www.piwheels.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='www.piwheels.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping

SK-hoy, 2024-12-16:
bzw. bei 137 folgende Meldung: 
**Defaulting to user installation because normal site-packages is not writeable
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
Could not fetch URL https://pypi.org/simple/packaging/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/packaging/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
Could not fetch URL https://www.piwheels.org/simple/packaging/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='www.piwheels.org', port=443): Max retries exceeded with url: /simple/packaging/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
ERROR: Could not find a version that satisfies the requirement packaging (from versions: none)
ERROR: No matching distribution found for packaging
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
Could not fetch URL https://www.piwheels.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='www.piwheels.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
pi@raspberrypi:/usr/bin $ cd
pi@raspberrypi:~ $ pip3.9 install packaging
Defaulting to user installation because normal site-packages is not writeable
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/packaging/
[... cut ...]
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
Could not fetch URL https://www.piwheels.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='www.piwheels.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
**

reserve85, 2024-12-16:
hm, ich glaube du hast Python ohne SSL Modul installiert bzw. compiliert und kannst dadurch keine Packages / Module nachinstallieren... 

Mein ansatz wäre erstmal Python korrekt zum Laufen zu bekommen, dann sollte auch die installation des Scripts durchlaufen.

SK-hoy, 2024-12-16:
Danke für die Info, da setze ich mal an. Auf der Basis 64 oder 32bit?

SK-hoy, 2025-01-04:
Das script und die Installation laufen jetzt, habs einmal neu aufgesetzt und jetzt funzt es vielen Dank. Habe jedoch noch einen bedingten Fehler, das die Wattzahl nicht heruntergeregelt wird bzw. immer wieder zu hoch steigt - wollte es begrenzen auf 800 aber dies wird so nicht angenommen:
Entweder Syntax und Logikfehler? 
an 04 14:16:11 raspberrypi python3[1359]: 2025-01-04 14:16:11 INFO     OpenDTU: Inverter "HM-1500": setting new limit from 481 Watt to 40 Watt
Jan 04 14:16:15 raspberrypi python3[1359]: 2025-01-04 14:16:15 INFO     OpenDTU: Inverter "HM-1500": Limit acknowledged
Jan 04 14:16:15 raspberrypi python3[1359]: 2025-01-04 14:16:15 INFO     powermeter Tasmota: 120 Watt
Jan 04 14:16:15 raspberrypi python3[1359]: 2025-01-04 14:16:15 INFO     setting new limit to 800 Watt
Jan 04 14:16:15 raspberrypi python3[1359]: 2025-01-04 14:16:15 INFO     OpenDTU: Inverter "HM-1500": setting new limit from 40 Watt to 800 Watt
Jan 04 14:16:17 raspberrypi python3[1359]: 2025-01-04 14:16:17 INFO     OpenDTU: Inverter "HM-1500": Limit acknowledged
Jan 04 14:16:37 raspberrypi python3[1359]: 2025-01-04 14:16:37 INFO     OpenDTU: Inverter "HM-1500" reachable: True
Jan 04 14:16:37 raspberrypi python3[1359]: 2025-01-04 14:16:37 INFO     Average min-panel voltage, inverter "HM-1500": 51.0 Volt
Jan 04 14:16:37 raspberrypi python3[1359]: 2025-01-04 14:16:37 INFO     OpenDTU: Inverter "HM-1500": Turn on
Jan 04 14:16:48 raspberrypi python3[1359]: 2025-01-04 14:16:48 INFO     powermeter Tasmota: -1313 Watt
Jan 04 14:16:49 raspberrypi python3[1359]: 2025-01-04 14:16:49 INFO     powermeter Tasmota: -1302 Watt

Auszug aus dem Skript: 
COMMON]
# Number of Inverters
INVERTER_COUNT = 1
# max difference in percent between SetpointLimit change to approximate the power to new setpoint
SLOW_APPROX_LIMIT_IN_PERCENT = 20
# if slow approximation: additional limit based on the limit-difference to "smoot the curve": newLimitSetpoint = newLimitSetpoint + (LimitDifference * SLOW_APPROX_FACTOR_IN_PERCENT / 100)
SLOW_APPROX_FACTOR_IN_PERCENT = 20
# interval time for setting limit to Hoymiles
LOOP_INTERVAL_IN_SECONDS = 20
# Timeout time to wait for Acknowledge after sending limit to Hoymiles Inverter
SET_LIMIT_TIMEOUT_SECONDS = 10
# polling interval for powermeter (must be <= LOOP_INTERVAL_IN_SECONDS)
POLL_INTERVAL_IN_SECONDS = 1
# if your powermeter exceeds POWERMETER_MAX_POINT: immediatelly set the limit to predefined percent of HOY_MAX_WATT (if you have more than one inverter it´s the sum of all HOY_MAX_WATT)
# value = 0 disables the feature. Values are possible from [0 to 100]
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 100
# if your powermeter falls below POWERMETER_MIN_POINT: immediatelly decrease the limit
ON_GRID_FEED_FAST_LIMIT_DECREASE = false
# max difference between Limit and real output power in % of HOY_MAX_WATT (100 = disabled)
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 100
# enable logging to file
ENABLE_LOG_TO_FILE = false
# how many logfiles you wish to keep
LOG_BACKUP_COUNT = 30
# defines how often the Inverter Power Status will be set, set it to "-1" for disabled (infinite repeat)
SET_POWERSTATUS_CNT = 10
# log the inverter temperature
LOG_TEMPERATURE = false
# delay time after turning the inverter off or on
SET_POWER_STATUS_DELAY_IN_SECONDS = 10
# define if you want to set your inverter to min-limit when your powermeter can't be read out
SET_INVERTER_TO_MIN_ON_POWERMETER_ERROR = false
# Total number of retries to allow.
MAX_RETRIES = 3
# A set of integer HTTP status codes that we should force a retry on. Don´t change unless you know what you are doing
RETRY_STATUS_CODES = 500,502,503,504
# It allows you to change how long the process will sleep between failed requests. The algorithm is as follows: {backoff factor} * (2 ** ({number of total retries} - 1))
RETRY_BACKOFF_FACTOR = 0.1

[CONTROL]
# --- global defines for control behaviour ---
# POWERMETER_TARGET_POINT is the target power for powermeter in watts
POWERMETER_TARGET_POINT = -75
# POWERMETER_TOLERANCE is the tolerance (pos and neg) around the target point. in this range no adjustment will be set
POWERMETER_TOLERANCE = 25
# POWERMETER_MAX_POINT is the maximum power of your powermeter for the normal "regulation loop".
# if your powermeter jumps over this point, the limit will be increased instantly. it is like a "super high priority limit change".
# if you defined ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT > 0, then the limit will jump to the defined percent when reaching this point.
POWERMETER_MAX_POINT = 0
# POWERMETER_MIN_POINT is the minimum power of your powermeter for the normal "regulation loop".
# if your powermeter jumps under this point, the limit will be reduced instantly. it is like a "super high priority limit change".
POWERMETER_MIN_POINT = -600

# List of INVERTERS, based on COMMON/COUNT
[INVERTER_1]
[... 18 lines cut ...]
# voltage to limit the inverter to a maximum of HOY_BATTERY_NORMAL_WATT
HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 50.5
# maximum limit in watts when battery is high (above HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V)
HOY_BATTERY_NORMAL_WATT = 800
# maximum limit in watts when battery is low (below HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V)
HOY_BATTERY_REDUCE_WATT = 10
# reenable inverter at this voltage
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 51
# define which panels you want to ignore for the panel voltage. E.g. if you have input 1 + 2 battery powered and input 3 + 4 for solar panels you can disable panel 3 and 4.
[... cut ...]
# set limit of 1100W -> inverter 1 is set to 733W and inverter 2 is set to 367W
# set limit of 300W -> inverter 1 is set to 200W and inverter 2 is set to 100W
# example 2:
# inverter 1 = 1000W, priority = 1 // inverter 2 = 500W, priority = 2: 
# set limit of 1100W -> inverter 1 is set to 1000W and inverter 2 is set to 100W
# set limit of 300W -> inverter 1 is set to 300W and inverter 2 is powered off
HOY_BATTERY_PRIORITY = 1
# Number of measured values for the moving average of the min panel voltage
HOY_BATTERY_AVERAGE_CNT = 1

Irgendeine Idee?

reserve85, 2026-11-11:
Laut deiner letzten Rückmeldung läuft das Skript nach der Neuinstallation. Ich schließe das Issue. Falls die Begrenzung auf 800 W weiterhin nicht greift, mach bitte ein neues Issue mit Log und Config auf.

closed 2026-11-11 by reserve85