reserve85/HoymilesZeroExport #264: Stromverbrauch Intervall

HA-IDIPS, 2025-02-28:
Hallo, vielen Dank nochmals für das super Skript. Ich setze es für den Akkubetrieb ein.

Dabei habe ich die Erfahrung gemacht, dass es bei Geräten mit Intervall Stromverbräuchen (Herd, Backofen, Bügeleisen) leider eher schlecht regelt. Beispiel: Herd schaltet sich immer für 10-20 Sekunden mit Vollleistung ein. Dann braucht es aber bereits im Schnitt 5 Sekunden für den Regel-Loop und nochmal weitere 10 Sekunden bis der Hoymiles das neue Limit umgesetzt hat. Wenn also hoch geregelt wird, endet meist gerade der Strombezug der Geräte.

Mein Regel-Loop steht auf 10 Sekunden, tiefer kann ich auch nicht, um erstmal die Umsetzung eines neuen Limits auf dem Hoymiles abzuwarten. Ansonsten wird beim nächsten Loop nochmal höher geregelt, weil der Hoymiles noch nicht hochgeregelt hat - Wert ist beim Smartmeter noch nicht erfasst.

Mir ist klar, dass es diese physikalischen Grenzen des Hochregelns beim Hoymiles gibt. Gäbe es denn irgendwelche Strategien, diese Geräte mit Intervall Stromverbräuchen durch den Akku besser abzufedern? Wenn man zum Beispiel das Abregeln länger hinauszögert (konfigurativer Wert x Sekunden "halten"), um so zum Beispiel in das nächste Intervall mit rein zu rutschen? Also lieber dann einen höheren Akkuverbrauch mit in Kauf zu nehmen?

Hier meine relevanten Config-Werte, eventuell ist auch etwas nicht optimal eingestellt:

[COMMON]
INVERTER_COUNT = 2
SLOW_APPROX_LIMIT_IN_PERCENT = 100
SLOW_APPROX_FACTOR_IN_PERCENT = 0
LOOP_INTERVAL_IN_SECONDS = 10
SET_LIMIT_TIMEOUT_SECONDS = 3
POLL_INTERVAL_IN_SECONDS = 1
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 100
ON_GRID_FEED_FAST_LIMIT_DECREASE = true
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 100
SET_POWERSTATUS_CNT = 2
SET_POWER_STATUS_DELAY_IN_SECONDS = 3
SET_INVERTER_TO_MIN_ON_POWERMETER_ERROR = false

[CONTROL]
POWERMETER_TARGET_POINT = -15
POWERMETER_TOLERANCE = 25
POWERMETER_MAX_POINT = 400
POWERMETER_MIN_POINT = -400

reserve85, 2025-03-03:
Sorry für die späte antwort,

bei mir geht das echt Ruck-Zuck dank ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT sowie ON_GRID_FEED_FAST_LIMIT_DECREASE . Dadurch wird die Regelschleife doch unterbrochen und sofort geregelt. Du müsstest mal ein Log anhängen bei dem dein Problem deutlich wird. Ich habe auch nur einen Inverter. 10 Sekunden bis der Hoymiles das umsetzt kommt mir sehr untypisch vor, bei mir sind das max. 1-2s bis ich das am Zähler sehe. Ist natürlich auch abhängig davon, wo du den "ist-Wert" herholst, evtl gibt es da auch noch verzögerungen.

goodnez89, 2025-04-04:
Hallo, ich habe das selbe Problem.
Habe jetzt mal diese Werte in der Config hinzugefügt.

ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 55
ON_GRID_FEED_FAST_LIMIT_DECREASE = true
POWERMETER_MAX_POINT = 400
POWERMETER_MIN_POINT = -400

sehe jetzt erstmal keine Änderung, was soll der dann machen wenn der meter -400 anzeigt?

reserve85, 2026-11-11:
Ohne Log und Config kann ich das leider nicht nachvollziehen, und seit längerem kam keine Rückmeldung. Ich schließe das Issue. Falls das Problem mit der aktuellen Version noch besteht, bitte mit Log vom Start an und der Config neu öffnen.

closed 2026-11-11 by reserve85