reserve85/HoymilesZeroExport #265: HoymilesZeroExport mit mehreren OpenDTU´s

33aw33, 2025-03-03:
Hallo,

Ist es möglich mehrere OpenDTU´s in dem Script zu definieren/konfigurieren?

Meine beiden Wechselrichter können nicht mit einer OpenDTU "versorgt" werden- Reichweite des Signals.

Vielen Dank für Rückmeldungen
aw

reserve85, 2025-03-03:
sorry, ist nicht möglich. Es wird nur eine DTU unterstützt...

Gruß

33aw33, 2025-07-06:
> Ich habe das mit Docker Compose gelöst. Die beiden kommen sich zwar ab und
> zu in die Quere, aber prinzipiell funktioniert es
>
> services:
>   hoymileszeroexport_huette:
>     image: ghcr.io/reserve85/hoymileszeroexport:main
>     volumes:
>       - ./HoymilesZeroExport_Config_Override_Huette.ini:/app/config.ini
>     command: -c /app/config.ini
>     restart: always
>
>   hoymileszeroexport_garage:
>     image: ghcr.io/reserve85/hoymileszeroexport:main
>     volumes:
>       - ./HoymilesZeroExport_Config_Override_Garage.ini:/app/config.ini
>     command: -c /app/config.ini
>     restart: always

Hallo Martin,

Danke für deine Rückmeldung- sobald ich aus dem Urlaub zurück bin, werde
ich das ausprobieren.

LG,
Andreas

reserve85, 2026-11-11:
Die Frage ist oben beantwortet, deshalb schließe ich das Issue. Falls noch etwas unklar ist, gerne ein neues aufmachen.