reserve85/HoymilesZeroExport #266: WR-Limit bei Ost-West Ausrichtung ohne Power Distribution Logic

gone-for-coding, 2025-04-03:
Ich hätte da eine Frage wie das Script mit der Limit-Berechnung in den Hoymiles-WR ohne Power Distribution Logic umgeht. Ich habe zwar einen neuen HMS-1600-4T den ich bald verbaue, der hat aber noch die alte Firmware und eine originale DTU für ein Update habe ich nicht.

Ich werde zwei Module nach Osten und zwei nach Westen haben. Wenn der WR jetzt ein Limit auf z.B. 400W bekommt, dann wird jeder der vier Eingänge auf 100W limitiert. Kommt jetzt nur von Osten Sonne, wird es statt 400W nur so 220W an Leistung geben.

Die Frage ist jetzt: Wie geht das Script damit um? Wird jetzt einfach das Limit soweit erhöht, bis die Einspeisung wieder passt? In meinem Beispiel müsste das Limit ja dann auf rund 800W gesetzt werden, damit tatsächlich ca. 400W erzeugt werden.

reserve85, 2025-04-04:
ja, wird weiter erhöht bis der output passt.

gone-for-coding, 2025-04-04:
Perfekt, danke für die schnelle Rückmeldung!

closed 2025-04-04 by gone-for-coding