reserve85/HoymilesZeroExport #268: docker: no matching manifest for linux/amd64 in the manifest list entries

elmr91, 2025-05-06:
I tried to pull the docker image on my debian linux (proxmox)
I got the following error:
 
```
# docker run -d --name hoymileszeroexport \
    -v ${PWD}/HoymilesZeroExport_Config_Override.ini:/app/HoymilesZeroExport_Config_Override.ini \
    ghcr.io/reserve85/hoymileszeroexport:main -c ./HoymilesZeroExport_Config_Override.ini
Unable to find image 'ghcr.io/reserve85/hoymileszeroexport:main' locally
main: Pulling from reserve85/hoymileszeroexport
docker: no matching manifest for linux/amd64 in the manifest list entries
```

It seems the docker image has not been built for this platform.

I also checked from my Synology docker, same problem.

reserve85, 2025-05-07:
@tomquist can you support here? gpt tells me it´s not merged correctly?

tomquist, 2025-05-07:
Ja, ich schaue es mir an. Ist vermutlich irgendwas bei der Migration der upload-artifact Action kaputt gegangen. Da gab es ein Breaking Change.

closed 2025-05-10 by reserve85

reserve85, 2025-05-10:
@elmr91 Can you please try again?

@tomquist: ❤

elmr91, 2025-05-10:
Thanks @reserve85 & @tomquist, it is working fine ❤