reserve85/HoymilesZeroExport #271: Warum startet ein Batterie-Inverter nur Nachts

Fibbbel, 2025-05-22:
Ich habe 3 Inverter. 2 Inverter werden über Panels gespeist. Der 3. Inverter hängt an einer Batterie die über eine Victron Solar-Ladegerät und weiteren Panels versorgt werden. Im Prinzip läuft alles , aber ich frage mich warum der Batterie-Inverter nur nach Abschaltung der Solar-Inverter (Nachts)  in Betrieb geht. Es gibt Fälle mit starker Einstrahlung tagsüber wo die Batterie schon voll ist und hoher Stromverbrauch existiert. Da könnte der Batterie-Inverter auch schon liefern. Ich habe schon manches über die HOY_BATTERY_PRIORITY versucht zu ändern, hatte aber nichts gebracht.

[image]

reserve85, 2025-05-22:
Kannst du bitte ein logfile eines kompletten Tages hier anhängen? Ggf. mit config.

Fibbbel, 2025-05-23:
Habe ich angestossen, kommt dann morgen.

Fibbbel, 2025-05-24:
Hallo und Guten Morgen,
anbei die aufgenommenen Daten.
Ich habe selber schon mal das LOG-file durchsucht finde aber nichts relevantes.
Interessanterweise startet der WR-Battery in der Zeit wo die Solar-Panels nichts mehr liefern, und die Kommunikation 
eingestellt wird (Sonnenuntergang). Umgekehr stoppt der WR-Battery wenn die Solarpanels Strom liefern (Sonnenaufgang).
In der Ahoy-DTU habe lasse ich die Kommunikation zu den Solar-WR stoppen.
Es gibt auch kurze Aktivität des WR-Battery über Tag die dann aber sofort wieder stoppt. Ich habe schon mal den WR-Battery manuell über die Ahoy-DTU eingeschaltet, dann läuft er manchmal auch für kurze Zeit.

[Daten.zip](https://github.com/user-attachments/files/20421973/Daten.zip)

Ansonsten noch ein schönes Wochenende.

Fibbbel, 2025-05-24:
Das war es gewesen. Jetzt arbeitet der WR auch tagsüber. Aber warum er dann nachts trotz des Wertes (HOY_MIN_WATT_IN_PERCENT=0) läuft kann ich mir nicht zusammenreimen. Egal, vielen Dank für die schnelle Unterstützung.

reserve85, 2025-05-25:
Super und Dankeschön! Ja der sprang nie mehr an weil er tagsüber ständig auf 0W gestellt wurde. Abends sind die anderen inverter aus und nach einiger Zeit resettet er sich dann wieder.